mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #300: Self-consume mode not triggering

tux43, 2024-05-18:
Hi,

I am working on a Sungrow Sh10RT inverter and I am finding that even when Self-consume mode is triggered the inverter is not supplying current to the house. I am not sure if it is a problem with the integration or the inverter? To keep things simple I have created a script that sends these modbus commands directly to the inverter, it matches this integration. Is there anything else I should be doing to trigger the inverter to go into Self-Consume mode?

```
alias: "SMART MOTION: Self-consumption mode battery"
sequence:
  - service: modbus.write_register
    data:
      hub: SungrowSHx
      slave: 1
      address: 13050
      value: 204
    alias: Set Stop (default)
  - service: modbus.write_register
    data:
      hub: SungrowSHx
      slave: 1
      address: 13049
      value: 0
    alias: Set EMS Mode Self-consumption mode (default)

Gnarfoz, 2024-05-18:
Did the inverter work correctly before you started trying to control it via Modbus?
I.e. via iSolarCloud or directly via the WiNet-S interface?

tactmaster, 2024-05-18:
I have a similar issues. It was working. I must have accidenly set export limit. So it set the limit to 500w. Then I had a play, 
Now is not suppling the house with battery. It is leaving it 100% and importing 
It can be set to force dischange.

koturbash, 2024-06-19:
Having the same issue, self-consumption worked fine before Modbus/integration into HA. Now self-consumption via Modbus does not work. 

When self-consumption is triggered in HA, it says my SH10RT-V112 is in self-consumption mode and iSolarCloud (installer account) says Self-Consumption, so Modbus works. But my SH10RT-V112 does not behave as expected (does not discharge the battery to load and imports from the grid instead).
Charging Start Power and Discharging Start Power set to the minimum possible 0.01kW iSolarCloud also tried 0 in HA.

Forced charge works as expected, but requires too many conditions/workarounds to execute properly. original Self-Consumption mode would be much more suitable and desired.

Does anyone have any updates on this? Perhaps a solution? =)

Gnarfoz, 2024-06-20:
It's nothing the integration did on its own, it doesn't write anything unless you make it write something.
I'd suggest looking at the charge/discharge time windows in the advanced settings in iSolarCloud (or via the WiNet-S interface, admin / pw8888):
[image]

There are several ways to more or less accidentally disable the battery in the various advanced settings...

koturbash, 2024-06-20:
Thanks, @Gnarfoz! 
Indeed, it was the reason for self-consumption not working as expected!

I wonder why those settings got changed in the first place, as I set those the next day SH10RT-V112 was installed and have not modified them since.
Could it be connected to these options on iSolarCloud? (see below)

What do those 0.5h 1h and 72h mean anyway? do settings go back to defaults after 72h?

[image]

tactmaster, 2024-06-20:
What I found was the several setting got set to disabled. Not 100% sure what setting i changed to make this break.

It was weekend and weekday discharging. 
[image]

So setting 5 weekend discharging which mean will use the battery for self consumtion not as i though.....to discarge to the battery to the grid..

Gnarfoz, 2024-06-21:
Yeah, the parameters are grouped in very confusing ways.

@koturbash I think those are settings that control how long iSolarCloud will wait for a result of the operation. I don't know how 72 hours makes sense for anything. Who would wait 72 hours for a setting to be applied or not applied? 🤔

Gnarfoz, 2024-06-21:
Considering that this discussion asks for the settings above to be added to this integration, I don't think it is touching these values, yet:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/240

tactmaster, 2024-08-28:
Hello, 

Ok I am confused, i was going to try add the extra ones from here
https://gist.github.com/rasmusbonnedal/31bc6dad78e6dc32bfedeb15b7b897df#file-gistfile1-txt-L3

So lets say 

> 33150   Weekday Discharging Start Time 1: Hour

however that reg is already used here 
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/00c5a1e36f1ef96361d05e1de2feb5dbdde4c75f/modbus_sungrow.yaml#L1625

It is has differnet regs here

Gnarfoz, 2024-08-29:
Interesting. But you'll have to confirm what works and what doesn't for yourself, in this case. These registers are not officially documented by Sungrow, so your guess is as good as anyone's. ;-)

closed 2025-01-05 by github-actions[bot]