mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #308: 2 inverters on Homeassistant

Markusunterholzer, 2024-06-17:
I have a question,
I have 2 identical Sungrow SH10rt inverters, both with storage.
One runs in normal mode, the second in stand-alone mode. I would like to integrate both into my Homassistant instance. How do I do that? It is not enough to rename all the sensors in the modbus_sungrow.yaml (sensor.xxx to sensor.2xxx)
Can someone help me?

TCzerny, 2024-06-22:
Did you add/renamed the host_ip, slave and port variables in modbus_sungrow.yaml and secrets.yaml too ?

on the header of modbus_sungrow.yaml:
host: !secret sungrow_modbus_host_ip_2
port: !secret sungrow_modbus_port_2

on all sensors:
device_address: !secret sungrow_modbus_slave_2

and the add in secrets.yaml

sungrow_modbus_host_ip: 192.168.178.xxx # TODO update with the IP of your inverter. Check your router.
sungrow_modbus_port: 502 # TODO update with the Modbus port of your inverter. Default is '502'
sungrow_modbus_slave: 1 #TODO update with the slave address of your inverter. Default is '1'

sungrow_modbus_host_ip_2: 192.168.178.xxx # TODO update with the IP of your inverter. Check your router.
sungrow_modbus_port_2: 502 # TODO update with the Modbus port of your inverter. Default is '502'
sungrow_modbus_slave_2: 1 #TODO update with the slave address of your inverter. Default is '1'

Markusunterholzer, 2024-06-25:
Thank you
after I renamed all sensors, scripts and automation it worked immediately.

closed 2024-06-25 by Markusunterholzer