mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #313: Inverter State - unknown

argually, 2024-06-27:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**
My Inverter state under "EMS & Battery Settings" is set to "Self-consumption mode (default)". When I set this the inverter state switches to "unknwon - you should not see me".

If I change the EMS-mode-Settings to "Forced mode" then the inverter mode state changes to "forced mode". If I reset the mode to  "Self-consumption mode (default)" it goes back into "unknown mode - you should not see me." 

At "Battery status"  it shows a battery capacity of 0,0 kWh. But our BYD battery has 12.8 kWh capacity...

A clear and concise description of what the bug is.
The Inverter state can´t be changed to "Self-consumption mode (default)".

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [] WiNet-S (LAN)
   - [x] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.6.4]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2024-04-05] (see header of the file)
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]

bjornenvall, 2024-06-30:
I have the same problem after installing the last inverter firmware.

argually, 2024-06-30:
I installed the latest firmware to solve the problem…

argually, 2024-06-30:
but it didn’t solve it…

argually, 2024-07-03:
Yesterday my Installer resetted the configuration to "factory default". I changed "set max battery charge|discharge Power to 5.000W. Now everythings seems to work fine again...

Does anybody know how to increase "max battery charge|discharge Power" back to 10.600W?
[image]

argually, 2024-07-05:
@Gnarfoz: That helped a lot! Thank you very much!
[image]
[image]

closed 2024-07-05 by argually