mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #315: Missing Statistik data post latest update

apographon, 2024-07-10:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

After the Upgrade to the latest version of Home Assistant and the Sungrow Integration my Home Assistant doesn‘t show any statistics anymore. The Grafana DB still gets data.

Core: 2024.7.1
Supervisor: 2024.06.2
Operating System: 12.4
Frontend: 20240705.

Modbus_sungrow: 2024-04-05

Happy to get some advices.

**Your Sungrow inverter:**
 - Model: SH-8RT
 - The inverter is connected via (mark one)
   - X LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - X no 
  - [] I don't know what that is

**modbus_sungrow.yaml:**
 - Version/ time stamp : See above 
 - X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
- Click on the current PV production

**Expected behavior**
- To see the history of the last hours

**Screenshots**
[image]

**Additional context**
Just updated everything lately to the latest versions

apographon, 2024-07-10:
[image]
Grafana detail - so there must be some data somehow….

apographon, 2024-07-11:
I just deleted the Database... seems to be corrupted

closed 2024-07-11 by apographon

Gnarfoz, 2024-07-11:
Turns out it was actually a bug in Home Assistant that was fixed in 2024.7.2:
https://community.home-assistant.io/t/psa-2024-7-recorder-problems/746428

...and I spent the last two weeks chasing my tail and trying to reduce the size of the database by setting up filters for recorder...
Probably not a totally wasted effort, went from 10 GiB to 900 MiB. 🔽