mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #320: Tiny visual bug at temperature sensors units

Tabis83, 2024-07-27:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

Tiny visual bug at temperature sensors units. Instead of [float]  °C, it shows [float] Â°C

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT 20]
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [x] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.03.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2024-07-17] (see header of the file)
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Ceck one of the temperature sensors like "Inverter temperature" or "Battery temperature" in the latest version of the PV Dashboard.

**Expected behavior**
Get rid of the additional "Â" in front of the unit.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Gnarfoz, 2024-07-27:
That looks more like an encoding issue on your end. How did you download the file?
Have you checked if a different web browser also shows it like this?

Tabis83, 2024-07-29:
Hi,
Thanks you are right. 👍
I've created a new file in the integrations folder and copied the content from GitHub file to the new file via clipboard using an iOS device. This probably caused the encoding error. 
The sensor unit in the modbus_sungrow.yaml was "Â°C" 🙈

After changing and restart of Home Assistant everything is fine.

closed 2024-07-29 by Tabis83