mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #322: SH-8.0 yaml configuration error

hummermann87, 2024-08-12:
Hi, I have followed the instructions to integrate my SH-8.0 in HA. When checking the configuration, I get the following error:

Error loading /config/configuration.yaml: mapping values are not allowed here
  in "/config/integrations/modbus_sungrow.yaml", line 229, column 75

Any ideas what I am doing wrong?
Latest HA and freshly downloaded modbus yaml.

Thanks in advance!

lassej01, 2024-08-13:
Start with remove the file and download it again. Ensure that you dont put in any extra spaces or similar into the file when copying.

hummermann87, 2024-08-13:
Hi, thanks for the quick reply.

Retried with a fresh file, but the problem persists.
In the meantime, I have been successful with integrating the converter using NickStallmann's WiNet extractor (https://github.com/NickStallman/home-assistant-repo/tree/main/winet-extractor).

If I can be helpful regarding improvement of the present "integration" (in case others face the same issue), i will still be happy to help, though.

closed 2024-12-22 by github-actions[bot]