mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #332: New firmware issue - SV300.001.03.P012

matheuscanela, 2024-09-05:
**Describe the bug:**

-> [Raising this bug as the new update is impacting me, so it may help others. I am still investigating, and I will close once I fix.]

I have the SH08RS and my dongle was updated to the version SV300.001.03.P012, so everything stopped working, including the WINNET webpage. I connect to my dongle via WIFI and it always worked fine.

I contacted Sungrow, and they said that the new release disabled HTTP, and now it requires HTTPS, so after the update my modbus stopped working and the page is not opening.

I am still debugging and I will provide an update, but if you are suffering the same, please comment below

**Your Sungrow inverter:**
 - Model: SH08RS 
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [] WiNet-S (LAN)
   - [X] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant version:**
 - Version: 2024.9.0

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
 - [] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

GobblerGuy, 2024-09-05:
My installation stopped suddenly 28th August 2024 at a bit before 4:05pm in the afternoon (Australian easter standard time). No homeassistant updates applied at or around this time. I do not use a WIFI dongle. Connected via LAN. Is it possible this update could have created an issue for LAN modbus as well?

matheuscanela, 2024-09-05:
Hi @GobblerGuy 

My modbus returned to sync after rebooting the inverter/batt all together; however, the new update now uses https for the webpage, so I am not sure if the update somehow stopped the sync via modbus.

Try to open the inverter's ip using http, if you get an "empty response", you are running the new version, and you will need to use https to open the inverter's page. Also, try to do a full reboot by turning off your inverter

Remigius0815, 2024-11-14:
I have ht e same issue after updating the Winet-S2 to the latest firmware. Restarting the inverter did not help, any ideas?

Gnarfoz, 2024-11-14:
Ah, yes, taking legal action against people offering a service for free on their own time that a vendor should be providing themselves in the first place. Customers love this strategy.

closed 2025-03-16 by github-actions[bot]