mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #337: SH20T Inverter

cryd-s, 2024-09-12:
**Is your feature request related to a problem? Please describe.**
Hi there i just installed my SH20T inverter with the new SBH100 and had a lot of non working entities.

- Missing MPPT3
-  No Battery charging and discarging power
-  wrong entities:

[image]

**Describe the solution you'd like**
Would like to help if u can tell me how 

[image]

here i think is a problem too.

wapman1979, 2024-09-12:
maybe this helps:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/310#discussioncomment-10562073

Gnarfoz, 2024-09-16:
Being unable to read MPPT3 directly is a known issue in the firmware.
You *can* read it, if you read it in bulk with some preceding registers. That's how the WiNet-S2 does it, as well.
See https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/298 for a lengthy discussion...

In general, the SHxT modbus situation seems quite a bit worse / less-polished than on the SHxRT series.
I'd recommend going one of the alternate routes:
- Using a RS485-to-Ethernet Converter to grab the Modbus RTU data from the "logger" interface on COM2 A1/B1).
- Using the WiNet extractor from https://github.com/NickStallman/home-assistant-repo/tree/main/winet-extractor to grab the data from the WiNet-S2 REST API.

purcell-lab, 2024-09-30:
There is some details on MPPT3 available at #349 
@RafAustralia

closed 2025-02-04 by github-actions[bot]