mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #340: Since FW-Update SAPPHIRE-H_B001.V000.P008-20240430.zip "Status" (Inverter State)  has changed from "stop" to "shutdown"

Matze-aaa, 2024-09-15:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

When the inverter is sent to "shutdown" (eeg via Inverter-Mode Box) the state changed to "stop".
Since the FW-Update 20240430 the name of the state has changed from "stop" to "shutdown" (same name as the mode).
So the entity "sungrow inverter enable/ stop input selector update" is wrong now.
Because its "stop" in the yaml, the inverter starts again after shutdown because of the slider update.

correct:
action: input_select.select_option
target:
  entity_id: input_select.set_sg_inverter_run_mode
data:
  option: |
    {% if is_state('sensor.sungrow_inverter_state', "Shutdown") %}
      **Shutdown**
    {% else %}
      Enabled
    {% endif %}

**Your Sungrow inverter:**
 - Model: [SH-10.RT v11]

latest version of your yaml.

mkaiser, 2024-09-16:
tried to update my inverter, but it looks like the winet-s is having issues.... will take a while for me until I can test & merge

mkaiser, 2024-09-16:
What about setting checking here for Shutdown AND Stop? 

`{% if is_state('sensor.sungrow_inverter_state', "Shutdown") %}`

would this work for old and newer firmware versions?

Matze-aaa, 2024-09-17:
Good idea, it should be "or", not "and" 😉

mkaiser, 2024-09-21:
Just commited - can you double check?

Matze-aaa, 2024-09-26:
it looks good, thanks!

closed 2024-09-27 by mkaiser