mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #350: Code for inverter type SH25T and SH15T is confilicting with code for SH25T-V11 and SH15T-V11

RafAustralia, 2024-09-30:
I have SH25T and when your code is implemented it shows up as SH25-V11 which is not correct

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/a03c37b82d773ecf1c25ff4f4e66bbecc948b83d/modbus_sungrow.yaml#L2154

[image]

mkaiser, 2024-09-30:
Do you know the correct device code? Should be the current value of 

> Sungrow device type

Datasheet only has these values

[image]

RafAustralia, 2024-09-30:
I do not have the code for V11 inverters, but I know for sure SH25T and Sh15T is the right code. (I have these inverters)
And both of my hybrids display the right code…
If I merge your updated code my inverters assume straight away they are V11 and they are not.
I have gone through these codes with Sungrow when they visited me from China, some months back, but we do not have V11 in Australia so these models were not verified at the time. I’m just letting you know they clash

I can email Sungrow contacts and find out what codes these are. (might take a while)

I have sent you a discord invite as there are quite a few questions I have for the updated code… some I know I understand and some I do not see reason why… I can guess…

Also I do have fully operational dual hybrid with logger running on modbus using your files but modified to suit… so happy to contribute if you wish to have a look. I have written modbus for these and chint as well.

Chat soon… if you like.

Raf

image.png (view on web)<https://github.com/user-attachments/assets/d7535fb7-12a1-4ce1-b55a-bf20c589a905>

RafAustralia, 2024-09-30:
I am wondering if it simply matters, if V11 is displayed next to my SH15T and SH25T, maybe it is really a minor thing... so in order to keep this moving... I have deleted my double ups and cleaned up the code...

I would also suggest moving the 4.6RS up where it belongs... just looks out of place amongst T inverters...

Code below: Hope it makes sense...

**@Mkaiser - again same question about dropping (default=0) and leaving only |int) - wondering why - thank you.**

        state: >-
          {% if ((states('sensor.sungrow_device_type_code') | int(default=0))  == 0x0D06) %}
            SH3K6
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D07) %}
            SH4K6
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D09) %}
            SH5K-20  
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D03) %}
            SH5K-V13
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0A) %}
            SH3K6-30
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0B) %}
            SH4K6-30
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0C) %}
            SH5K-30
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D17) %}
            SH3.RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0D) %}
            SH3.6RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D18) %}
            SH4.0RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0E) %}
            SH4.6RS 
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0F) %}
            SH5.0RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D10) %}
            SH6.0RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D1A) %}
            SH8.0RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D1B) %}
            SH10RS
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E00) %}
            SH5.0RT
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E01) %}
            SH6.0RT
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E02) %}
            SH8.0RT
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E03) %}
            SH10RT
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E10) %}
            SH5.0RT-20
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E11) %}
            SH6.0RT-20
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E12) %}
            SH8.0RT-20
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E13) %}
            SH10RT-20
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0C) %}
            SH5.0RT-V112
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0D) %}
            SH6.0RT-V112
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0E) %}
            SH8.0RT-V112
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0F) %}
            SH10RT-V112
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E08) %}
            SH5.0RT-V122
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E09) %}
            SH6.0RT-V122
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0A) %}
            SH8.0RT-V122
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0B) %}
            SH10RT-V122
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E20) %}
            SH5T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E21) %}
            SH6T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E22) %}
            SH8T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E23) %}
            SH10T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E24) %}
            SH12T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E25) %}
            SH15T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E26) %}
            SH20T-V11
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E28) %}
            SH25T-V11
            Unknown device code: {{ '%0x' % (states('sensor.sungrow_device_type_code') |int)  }}
          {% endif %}

thank you

RafAustralia, 2024-09-30:
another update

Since I have a SH25T and SH15T, and not V11 models... I was getting an unknown device code e28
by adding extra models such as SH25T and SH15T corrected that.:

I am not sure if this is right or wrong but it makes the unknown device error disappear and only displays eg. SH25T-V11.

    state: >-
      {% if ((states('sensor.sungrow_device_type_code') | int(default=0))  == 0x0D06) %}
        SH3K6
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D07) %}
        SH4K6
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D09) %}
        SH5K-20  
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D03) %}
        SH5K-V13
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0A) %}
        SH3K6-30
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0B) %}
        SH4K6-30
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0C) %}
        SH5K-30
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D17) %}
        SH3.RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0D) %}
        SH3.6RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D18) %}
        SH4.0RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0E) %}
        SH4.6RS 
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D0F) %}
        SH5.0RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D10) %}
        SH6.0RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D1A) %}
        SH8.0RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0D1B) %}
        SH10RS
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E00) %}
        SH5.0RT
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E01) %}
        SH6.0RT
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E02) %}
        SH8.0RT
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E03) %}
        SH10RT
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E10) %}
        SH5.0RT-20
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E11) %}
        SH6.0RT-20
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E12) %}
        SH8.0RT-20
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E13) %}
        SH10RT-20
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0C) %}
        SH5.0RT-V112
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0D) %}
        SH6.0RT-V112
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0E) %}
        SH8.0RT-V112
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0F) %}
        SH10RT-V112
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E08) %}
        SH5.0RT-V122
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E09) %}
        SH6.0RT-V122
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0A) %}
        SH8.0RT-V122
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E0B) %}
        SH10RT-V122
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E20) %}
        SH5T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E21) %}
        SH6T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E22) %}
        SH8T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E23) %}
        SH10T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E24) %}
        SH12T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E25) %}
        SH15T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E25) %}
        SH15T
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E26) %}
        SH20T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E26) %}
        SH20T
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E28) %}
        SH25T-V11
      {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E28) %}
        SH25T
        Unknown device code: {{ '%0x' % (states('sensor.sungrow_device_type_code') |int)  }}
      {% endif %}

Gnarfoz, 2024-10-01:
If you have two `elif` lines that check for the same value (for example `0x0E28`), the second one can never be reached.

> I am wondering if it simply matters, if V11 is displayed next to my SH15T and SH25T, maybe it is really a minor thing... 
I don't think it does. It's merely a display issue. The resulting value is not used for anything else than offering a sensor that shows which model of inverter you have.

The register documentation changelog for version 1.1.2 also mentions "delete SH5-25T_V12 series". I don't think anything other than -V11 currently exists. Even if they're not labeled as such. Since yours identify as `0x0E28` and the specification in appendix 4.6 maps that to SH25T-V11. Maybe the next version will list include new / other device codes.

RafAustralia, 2024-10-02:
Hi Christian

I have done some work on this yesterday.

My inverter is not V12 it is a straight SH25T as far as I can tell.
When I deleted my line of code referring to SH25T and left SH25T-V11 it has returned the following error:

***@***.***

When I have duplicated both, and kept SH25T and SH25T-V11, it will show V11 only (like you said it gets the first it finds):

***@***.***

The only way for me to keep SH25T on display without “V11” is to delete SH25T-V11 code all together… but in the end… it doesn’t really matter much… I do not think…

Gnarfoz, 2024-10-02:
> When I deleted my line of code referring to SH25T and left SH25T-V11 it has returned the following error:

I think you forgot to include the actual error. ;-)

RafAustralia, 2024-10-02:
I did attach it but sometimes these images do not come through. Depends what device I use to attach these.

Trying again below. Let me know if u cant see them:

[image0.png][image3.png]

Message ID: ***@***.***>

Gnarfoz, 2024-10-02:
I'm afraid that didn't work, it just shows `[image0.png][image3.png]`.

RafAustralia, 2024-10-02:
Dragged myself to my laptop...

When SH25T-V11 was left but SH25T was deleted my SH25T did not know what the e28 was...

When both were put in - it works sort of.

[image]
[image]

Gnarfoz, 2024-10-02:
Not sure how you got into that state, but the first image looks like it's returning *two* strings:
- SH25T-V11
- Unknown device code: e28

RafAustralia, 2024-10-02:
Hi Christian

Explanation to above:

Initially - months back before you guys added the new inverters - I have added my inverters into my code by simply finding out the codes and types.

My inverters are straight SH25T and SH15T - no versions , no V11, V12, nothing.
Example from isolarcloud:
[image]

It worked for me. So my current  operational code looks like that:

          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E25) %}
            SH15T
          {% elif ((states('sensor.sungrow_device_type_code') | int(default=0)) == 0x0E28) %}
            SH25T
          {% else %}
            Unknown device code!
          {% endif %}

I am in a process of updating my code with new changes from mkaiser, and I was questioning how will SH25T versus SH25T-V11 behave.

**If I do keep both lines for SH25T-V11 and SH25T - my inverter understands but turns up as SH25T-V11.**

This is the screenshot:
[image]

So my logic was - **ok, if my inverter thinks it is SH25T-V11, then delete SH25T**, cos it is not needed, right?

Well then my inverter has returned the second line of code - saying I am SH25T-V11, but I have unknown e28 code as well.

Example below:
[image]

So I can tell you definitely my inverter is not happy with SH25T-V11 only...

so options are:
1. for wider public if I want my modbus to work for both SH25T and SH25T-V11, then both lines need to stay, and my inverter will just say SH25T-V11, or
2. I do not type in SH25T-V11 at all - drop that line of code, and my inverter remains as original code only - SH25T, thats it.

I find it strange they have same codes... not impossible but doesnt allow for them to work as separate units.

Hope that makes sense...?

Gnarfoz, 2024-10-03:
First off, your inverter doesn't know about these strings at all, and it does not care about them. It doesn't check them, either.

The only interaction here with your inverter is Home Assistant reading (for the sensor named "Sungrow device type code") from Modbus register 5000 (address 4999) and your inverter returning `0x0E28`.

This sensor value is then in turn used in the template sensor "Sungrow device type" to map it into a human-readable string to be used to display in dashboards etc.

The actual reason you were getting weird output is that you seem to have accidentally left out the final `{% else %}` right before the fallback case when you edited your local copy. This is visible in your comments above from 3 days and 2 days ago. (https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/350#issuecomment-2382861687 and https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/350#issuecomment-2384427802)
That's why you were getting a multi-line result.

The other bit was already cleared up: when you have two occurrences of `0x0E28` in that big list of `{% elif %}` cases, the first one will be used. Can't have one model code refer to two model names.

Which brings me to the last bit: I'm not convinced a SH25T-with-no-version actually exists. I think the version is just not prominently shown anywhere. On the previous SHxRT series, the only place it physically shows up is on a sticker on the actual box the inverter comes in.
If you want, I can look up your inverters' serial numbers on the Sungrow service portal and check what Sungrow thinks:
[image]

Gnarfoz, 2024-10-03:
I agree, it's a cosmetic issue at best. The service portal might still shed some light on this. I'd just need the serial numbers.
Or, given that you've spent quite a bit of time working with Sungrow, maybe you already have an account there yourself?

It's this place: https://gsp.sungrow.cn/
If you go to https://gsp.sungrow.cn/maintenance/guarantee, click on _create_, scroll down to _product list_ and click on _+ add new product_, you can enter your serial numbers and it will look them up instantaneously.
*Do not click on _draft_ or _submit_.* Submit would create an actual ticket. Draft just creates a draft, but I haven't figured out how to get rid of unwanted drafts, either. :D

RafAustralia, 2024-10-03:
@Gnarfoz 
I have already asked that question from our Sydney HQ guy - will see what he says.

RafAustralia, 2024-10-03:
@Gnarfoz 
this is getting really interesting... see below

[image]

closed 2025-02-04 by github-actions[bot]