mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #354: Wrong Register used for Battery Capacity

RafAustralia, 2024-09-30:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blame/a03c37b82d773ecf1c25ff4f4e66bbecc948b83d/modbus_sungrow.yaml#L1617

I believe that wrong register is being used for this entity and below is adjusted register address to fix that issue:

      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 13039 changed from reg 33048
        input_type: input # reg changed holding to input
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.1
        scan_interval: 10

In my case this has fixed all issues with battery capacity... and I have applied it to quite a few sungrow battery units. 
(they all seem to work)

RafAustralia, 2024-09-30:
@purcell-lab - thank you :)

mkaiser, 2024-09-30:
Adapted it - the capacity is still correct in my setup. 

>  address: 13038 # reg 13039 changed from reg 33048
Where did you find the new register address?

RafAustralia, 2024-10-01:
@mkaiser 

hey... I have the following version of protocol with me (which was the latest at the time of writing)
TI_20231019_Communication Protocol of Residential Hybrid Inverter_V1.1.2_EN (007)

here is the register I used:

[image]

mkaiser, 2024-10-01:
Thank you, just integrated your PR

closed 2024-10-01 by mkaiser