mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #356: Syntax error in entity definition "Sungrow inverter state"

st-schuler, 2024-09-30:
The definitions of “Fault” and “Stop” for the entity “Sungrow inverter state” contain a syntax error. 
The comparison operator “==” should be removed as the “in” operator is already used.

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/351a509b224cb915f65a1a41cbcd4861029fe26c/modbus_sungrow.yaml#L2060

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/351a509b224cb915f65a1a41cbcd4861029fe26c/modbus_sungrow.yaml#L2062

RafAustralia, 2024-09-30:
following the syntax trail I have also added my inverter states code, where somewhere across my modbus coding I came across extra running state value "0x8200" and extra stop state value "0x0002" (could have been from the days I had SH10RT)

I just compiled this list - as a merge of my states and new mkaiser added states.

**@Mkaiser - I would like to know what is the reason to remove (default=0) please so I can understand better, please**

        state: >-
          {% if ((states('sensor.system_state') | int(default=0)) in [0x0000,0x0040,0x8200]) %}
            Running
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x0410) %}
            Off-grid Charge
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x0200) %}
            Update Failed
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x0400) %}
            Maintain mode
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x0800) %}
            Forced mode
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x1000) %}
            Off-grid mode
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x1111) %}
            Un-Initialized
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x0010,0x12000]) %}
            Initial Standby
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x1300,0x0002]) %}
            Shutdown
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x1400,0x0008] ) %}
            Standby
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x1500,0x0004] ) %}
            Emergency Stop
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x1600,0x0020]) %}
            Startup
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x1700) %}
            AFCI self test shutdown
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x1800) %}
            Intelligent Station Building Status
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x1900) %}
            Safe Mode
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x2000) %}
            Open Loop
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x2501) %}
            Restarting
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x4000) %}
            External EMS mode
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x55000,0x0100]) %}
            Fault
          {% elif ((states('sensor.system_state') | int(default=0)) in [0x8000,0x0001,0x0002]) %}
            Stop
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x8100) %}
            De-rating Running
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x8200) %}
            Dispatch Run
          {% elif ((states('sensor.system_state') | int(default=0)) == 0x9100) %}
            Warn Running
          {% else %}
            Unknown - should not see me! {{ (states('sensor.system_state') |int) }}
          {% endif %}

Thank you kindly.

mkaiser, 2024-09-30:
[st-schuler](https://github.com/st-schuler): Thank you for the PR!

[RafAustralia](https://github.com/RafAustralia):
I removed the  **int(default=0)** a couple of commits ago. The line

**availability: "{{ not is_state('sensor.system_state', 'unavailable') }}"**

checks, if the sensor.system_state is valid. Only if it is valid, the **state: >-** section is executed. 

Some time before that I did not know about the **availability** and the log was flooded with "cannot cast to..." entries. The default-statements mitigated these issues (but are no longer needed)

Please continue writing / asking stuff about the code. I am pretty sure that there are many more inconsistencies :)

closed 2024-10-01 by mkaiser