mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #362: Battery won't discharge under 5% even though Min-SOC is set to 0%

TheHairforce, 2024-10-06:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

Hello everyone,
I started using the Sunggrow Modbus Integration into Home-Assistant yesterday and it worked out nice. - Everything is showing up and working!

I only have a single (weird?) issue:
My Battery won't discharge lower than 5%. When I go to Bed this night it was at 5% and it starting pulling Energy from the Grid and 'ignored' the Batterie. What's even weirder: When I looked it up tomorrowit suddenly showed "Battery is at 10%", even though it didn't load the Batterie from the Grid.

I've checked the "EMS Settings" and there the Min-SoC is set to 0% but somehow it ignores that input. Am I doing something wront or do I need to switch some other Setting as well?

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [X] LAN (internal port) (I use this for HA)
   - [X] WiNet-S (LAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant version:**
 - Version: 2024.10.1

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-10-01
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**Expected behavior**
The Battery discharges until it's empty. And won't 'suddenly' jump back to 10% Charge after it's at 5% for a while.

korttoma, 2024-10-07:
I think this is because if you bring the battery too low it can trigger "emergency charging" and force charge it to 10%

I have set the min soc to 8% but I think also 5% can be used and still avoid the "emergency charging" from grid.

RafAustralia, 2024-10-12:
Hmmm. 
i have just pushed my battery to 2% last night due to spikes in power prices here. 

There are however few things to consider. 

1. In order to avoid error e967 the reserved SOC cannot be lower then minSOC
2. So if u change them I would change reservedSOC first. Give it few sec to apply. It needs to make a change in the inverter settings.  Then drop min SOC slider. 

Hope that helps

TheHairforce, 2024-10-13:
Hey everyone,
it seems that setting it to 8% fixed it, thank you all!

Now I just have another 'Problem' and I'm not 100% sure how I can fix that.
I've build myself a Dashboard and I have four Sensors Showing up on there:

**Total Imported Energy**
Total Imported Energy Cost
**Total Exported Energy from PV**
Total Expoerted Energy Compensation

Both of the Sensors that I marked bold are Working and showing up. - But the others Sensors: Nope, nothing. Stuck at 0 €.
Or has that something to do with Home-Assistant and not the Sungrow-Integration?