mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #363: yearly value is too high

svenschedlbauer, 2024-10-07:
**Describe the bug:**
The current yearly pv generation is 4.400 kwh, but sensor.yearly_pv_generation_current is showing 12.254 kwh. 
The monthly pv generation is showing the correct value.

**Your Sungrow inverter:**
 - Model: [e.g. [SH6.0RT]
 - The inverter is connected via (mark one)
   - [] LAN (internal port)

**Home Assistant version:**
 - Version: 2024.10

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-10-01

Gnarfoz, 2024-10-07:
The yearly registers showing wrong values is sadly a known firmware issue. You can try updating, but as far as I know, there is no fixed version.
This integration should probably switch to providing a template sensor that aggregates the monthly values instead.

FJJT01, 2024-10-11:
Same for the sensor.yearly_export_current and the specific year sensors (at least sensor.yearly_pv_generation_2024 and sensor.yearly_export_2024).

closed 2025-02-11 by github-actions[bot]