mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #366: sg_battery_capacity with wrong value

jove01, 2024-10-11:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**
```

`# Home Assistant Sungrow inverter integration
# https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
# by Martin Kaiser
# last update: 2024-10-01 (2)`
```

```
`      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 13039 
        input_type: holding
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 600`

```
**Describe the bug:**
Over 1 year I use this integration (thank you). I did notuse all sensors.

Now I need Battery Capacity. The value was zero. I found here the reason miising value in the  isolar.cloud. After that, the value was extreme high.
[image]

[image]

I deleted the sensor in modbus.yaml  and HA. Restartet the system, than I activated it again.

No change!

A clear and concise description of what the bug is.

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2023.09.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware (in summer) is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Gnarfoz, 2024-10-12:
You could try the alternate register mentioned here:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/241#issuecomment-2399090555

jove01, 2024-10-12:
Thanks for the answer. I found also another thread. So I tried different 
```
`      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 13039 
#        address: 33049  # reg 33048  
#        address: 5639  # reg 5638 
        input_type: input
#        input_type: holding
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 60`
```
This combination set the value to 1
[image]

Is there a way, to change the value manual? But I am afraid to use write.register, because what is the correct register.

jove01, 2024-10-12:
In ISolar.cloud I changed the battery_capacity from 9,6 to 8,6. The Value in HA did not change,

jove01, 2024-10-12:
Al last I have done a mistake with adress and register.

That is  correct and works fine
`        address: 5638  # reg 5639`

Thank you

closed 2024-10-12 by jove01

RafAustralia, 2024-10-12:
Hi gents, since this register comes from my PR... 
I would like to let you know that resolution is in small but important detail: 
**the register should be set to input not holding**, then the battery will show correct info, here is my register version:

      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 13039 
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.1
        scan_interval: 10

Hope that helps

[image]

jove01, 2024-10-12:
Thank you, but for me works fine (See above)
address: 5638  # reg 5639 ju

Gnarfoz, 2024-10-12:
> This combination set the value to 1

Yeah, that's due to https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/pull/355 only updating the register address but not the other parameters that are different between the previous undocumented holding register and the documented input register, as @RafAustralia pointed out above.

I was about to add a PR to fix that, but it looks like @mkaiser was doing some work at the same time and has now switched to the high-precision register 5639 anyway.

RafAustralia, 2024-10-12:
Cheers Christian. In total agreement.  @Gnarfoz

jove01, 2024-10-12:
First I did not understand you, because I have only 1 battery. Now I see, the modbus_sungrow.yaml is updated 3 hours ago from mkaiser.
Tomorrow I will try that.

RafAustralia, 2024-10-13:
> First I did not understand you, because I have only 1 battery. Now I see, the modbus_sungrow.yaml is updated 3 hours ago from mkaiser. Tomorrow I will try that.

hey @jove01 
Having two inverters and two batteries makes no difference to the code principle... but just sent you a snip of both showing correct value. I do run dual setup with two hybrids and full automation so whatever I do has to work for both.

Anyhow - code is getting better lots of new additional bits and tweaks :)

all the best:)

jove01, 2024-10-13:
I installed the new version from tonight. The last corrct value is shown. Than I changed in ISolarCloud in extended Parameters the value. This new value is not shown in HA.

Than I changed the yaml

> address: 13038  # reg 13039
After that, the value "1" is shown. Than I changed again to

> address: 5638 # reg 5639
the value 9,6 is shown an not my last change 9,0.

So I decide me for this code
```
`     - name: Battery capacity # as in TI_20240924_Communication Protocol of Residential Hybrid Inverter-V1.1.5
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 5638 # reg 5639     # wird nicht geändert; wieder aktiviert und nicht der manuelle Wert ISolarCloud
#        address: 13038  # reg 13039  # zeigt Wert 1,0
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 10   # 600` 
```
In ISolarClod is another "battery_capacity"
[image]

Shows 5638 maybe this sensor? In former discussion was said also the second must exist.

Thank you for your help

RafAustralia, 2024-10-13:
hey @jove01 

If this high precision register works for you thats great.
I never had a single issue with using, address: 13038/ #reg 13039...

I have just set up a friends system tonight in HA 2024.10.2 and this register just worked ....not sure why yours is different
whatever works in your system thats good. if high precision works - use that. 

not sure what you mean by In former discussion was said the second one must exist...
I have tried high precision register and it is ok too, address: 5638/ # reg 5639

Raf

jove01, 2024-10-13:
> hey @jove01
>
> If this high precision register works for you thats great. I never had a single issue with using, address: 13038/ #reg 13039...
>
> I have just set up a friends system tonight in HA [...]
Thanks for your help
Jürgen