mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #369: Update to new action syntax

mkaiser, 2024-10-12:
in modbus_sungrow.yaml I get 28 of these ones... What have i missed?
See picture
[image]

_Originally posted by @hedlung in https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/367_

mkaiser, 2024-10-12:
will do this end of this year to avoid issues from users with older HA versions ;)

RafAustralia, 2024-10-12:
When u move to 2024.10.2 these will need to follow new convention:

but please note. Warning
They throw syntax error now but still work!
changing them in 2024.9.3 is going to break them!

[image]

I have started the code for them for 2024.10.2 and have it ready for testing for when I push my system to the updated core. (10.2)

Gnarfoz, 2024-10-12:
> will do this end of this year to avoid issues from users with older HA versions ;)

I think you already did some of this in https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/94b7fa91e5088aec2e50400b2fb2c10492816eab -- now you're in the situation https://community.home-assistant.io/t/studio-code-server-upgrade-to-v5-16-0/776689/9 is warning against:
You have:
```action:
 - action: xyz```
And the first `action` can soon be replaced with `actions` (with HA 2024.10). That's going to be mildly annoying. I guess you could use the above commit to revert back to `service` temporarily to do it.
 
It's a bit easier to replace all instances of `action` with `actions` then:
```action:
 - service: xyz```

Anyway, this is all non-mandatory, since it's all backwards compatible. No reason to rush this. ;-)

RafAustralia, 2024-10-12:
Great reply @Gnarfoz 

Again totally agree. 
in meanwhile Ive gone and changed all triggers/trigger
actions/action
And saved the code (aside as a copy)
Etc… so hopefully when time comes u can just push it through… no need for double ups and timewasting…

RafAustralia, 2024-10-22:
Ive now swapped all syntaxes both in main code and automations... so its working like a BOSS :)

closed 2025-01-23 by mkaiser