mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #377: I have added a new sensor, namely : EMS heartbeat, which looks for the state of ems pulse

RafAustralia, 2024-10-13:
@Gnarfoz @mkaiser @purcell-lab 

This feature was suggested by code writer from sungrow this week.
It helps to see ems pulse which is set to reset rever 600s to its default, if not found.

For now the idea there is to set interval to 500s so it checks before it resets. (trial and error for now - it is really fresh)
NOTE: for 2024.10.2 it requires state_class - so I have added that in, since I am now fully in 2024.10.2 :)

      - name: EMS heartbeat
        unique_id: sg_ems_heartbeat
        slave: !secret sungrow_modbus_slave
        address: 13079  # register 13080
        input_type: holding
        data_type: uint16
        scale: 1
        offset: 0
        state_class: measurement
        unit_of_measurement: "seconds"
        scan_interval: 500 #it will reset every 600s if not found

[image]

Have fun folks - hope this works for you.

Gnarfoz, 2024-10-15:
Probably this:
[image]
with some kind of external energy management system?

RafAustralia, 2024-10-15:
> Probably this: [...]

hey @Gnarfoz 

I am currently working on trying to automate my curtailment for sungrow, and this has been identified as a very important part of curtailment not failing. It is a side project I am doing with Sungrow, but as I go I share new registers which come available... Perhaps there is another use for them...

Gnarfoz, 2024-10-16:
They are part of the default dashboard included in the project:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/_DefaultDashboard_mkaiser
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/dashboards/_DefaultDashboard_mkaiser/3_EMS_Control.png

RafAustralia, 2024-10-16:
@Gnarfoz 
hey - I have that one... that is normal... but it doesnt have that external EMS slider button...

Gnarfoz, 2024-10-16:
The slider belongs to some other element hidden behind the drop-down menu. 😉

mkaiser, 2024-10-16:
Maybe I am too confused. 

The sensor 

> - name: EMS heartbeat
>     unique_id: sg_ems_heartbeat
>     slave: !secret sungrow_modbus_slave
>     address: 13079  # register 13080
>     input_type: holding
>     data_type: uint16
>     scale: 1 [...]

is an input (holding), right? So Homeassistant --> Inverter

Then you would need an additional Automation to set the holding register every XXX (<600seconds) to avoid automatic resets within the inverter?

Gnarfoz, 2024-10-16:
I assume it's a counter that shows how many seconds ago the last heartbeat was received.
Not sure why it's a holding register, maybe that's so you can specify the limit yourself?
What happens when the limit is exceeded?

RafAustralia, 2024-10-16:
@Gnarfoz  @mkaiser 

A bit of back and forth history.
I am working on trying ot get the ZERO EXPORT - CURTAILMENT work better for sungrow. It is not great at the moment, does bad things, it is all over the place. That is my side project - and it has been for a year or so. I look at it when I have time.

In between the lines... sungrow has suggested inclusion of this register as important.

It doesnt do anything other (I think) than looks for the pulse, and to answer Christian's question about the limit... I believe the inverter's pulse is set at 600s, and after 600s each inverter is it fails to find the EMS MODE PULSE , then the inverter will revert to its DEFAULT MODE. so it could change mode back to Running? Selfconsume? maybe? It is still a very new thing for me.

Below is what Sungrow have commented on... and that is me and them talking STRICTLY ABOUT CURTAILMENT
(apologies - english in this picture is a bit weird)

[image]

@Gnarfoz @mkaiser  
If you wish to give me a hand to improve this - you are most welcome, and I can demonstrate the failures of export mode quite easily - not that it has anything to do with you directly but it could be improved by a mile... (my discord raf_t_aud)

cheers guys

RafAustralia, 2024-10-16:
> The slider belongs to some other element hidden behind the drop-down menu. 😉

yes - sorry must have been half asleep - I now know what you have shown... yeah - have that... and yes it is to do with EMS mode as explained further.

Gnarfoz, 2024-10-17:
> Then you would need an additional Automation to set the holding register every XXX (<600seconds) to avoid automatic resets within the inverter?

After reading @RafAustralia's image, I come to the same conclusion. You're supposed to write to that register periodically. This *is* the heartbeat. The payload value doubles as a way to set the heartbeat deadline. If you don't write to the register within the value you set the last time you wrote to it, the inverter will _"revert to self initiated use"_?
I can only assume that's chinglish for "self-consumption mode", meaning it will reset from EMS mode 3 to EMS mode 0?

That should be easy enough to validate.

However, I'm not sure if using Home Assistant as an external EMS for these inverters is within the scope of this project.
I don't think there even is public documentation on what that would encompass.

Gnarfoz, 2024-10-17:
> That should be easy enough to validate.

It's exactly how it works:
```
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 3
Slave configuration...: address = [1]
                        start reference = 13050, count = 1
Data type.............: 16-bit register, output (holding) register table

Written 1 references.

# mbpoll 192.168.0.63 -t 4 -1 -r 13080 30
Slave configuration...: address = [1]
                        start reference = 13080, count = 1
Data type.............: 16-bit register, output (holding) register table

Written 1 references.

# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        27
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        24
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        13
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        5
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        3
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        3
[13080]:        1
# mbpoll 192.168.0.63 -t 4 -1 -r 13050 -c 31 | rg '\[(13050|13080)'
[13050]:        0
[13080]:        0
```
You set EMS mode to 3, you set EMS heartbeat to your desired timeout value (setting this before enabling EMS mode 3 will be ignored), and if you now _read_ the EMS hearbeat register, it will show the time left until the timeout is reached.
When it's reached, it will revert to EMS mode 0.

So I guess this is one of the most basic things an external EMS would need to do to manage these inverters.
I'm not sure what external EMS mode would be good for in the context of this project, though.

I think @RafAustralia has problems with the built-in export limitation in the default self-consumption mode (there are two of those), but his setup is rather uncommon (a) multiple inverters, b) using Logger1000). IMHO that is something that should work out of the box, without Home Assistant or any other external EMS. An energy meter at the feed-in point should be enough -- isn't that what the DTSU-666 is mostly needed for?

RafAustralia, 2024-10-17:
@Gnarfoz 

Appreciated the thorough summary, but the issue is that the meter doesnt exist in the location where everyone else has them.
In my setup - which will be more common from here on as sungrow is changing their direction away from master/slave... logger will assume the role of EMS control device and it will be issuing commands to inverters instead of inverters themselves...

Meter is attached to the logger... so the inverter cannot see the register reads no more.
It is a continuing work in progress... but I do have a full modbus which I wrote for chint666-20 and Logger1000A
(I believe it is the only working version in the world for now)
And they have just added some basic new registers for controlling the logger as well... so I will be updating it shortly.

But please note: with meter not being available at location it normally is - everything is much harder to detect...

closed 2025-02-16 by github-actions[bot]