mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #383: Unknown inverter state 128

djranm, 2024-10-20:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

I have three Sungrow inverters which I'm controlling using this integration, 1x SG5.0RS and 2x SH10-RT. The two 10kw inverters are not wired in master/slave.

In the last couple of weeks I noticed lower PV generation overall, and that one of the 10kw inverters isn't producing much at all (<100w). When this is happening, the inverter state for that inverter is reported as 'Unknown - 128', while the other two inverters are reporting 'Running'.

Does anyone know what 128 is/maps to? In case it's relevant, the inverter with this issue is the V112 variant, whereas the other 10kw inverter is the standard model.

**Your Sungrow inverter:**
 - Model: 1x SG5.0RS, 1x SH-10RT, 1x SH-10RT V112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2024.10

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-10-15
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Query the inverter state.
2. Observe the result of 128 which is not mapped and therefore appears as 'Unknown - 128'

**Expected behavior**
I don't expect this integration to resolve whatever issue is going on, but it would be nice to know what the '128' inverter state maps to, and this could be helpful for others who encounter the same code in future.

**Screenshots**
Screenshot of the relevant part of my dashboard which includes the Inverter State.
[image]

Gnarfoz, 2024-10-20:
Interesting - sadly, 0x80 / 128 is not documented. We'll have to wait until someone figures it out some other way.
What do iSolarCloud and/or the local WiNet-S interface say about it?

> In the last couple of weeks I noticed lower PV generation overall [...]

Are you in the northern hemisphere? If yes, I have bad news for you. 😅 🍁 🍃 ☃️

djranm, 2024-10-20:
Haha! I'm in Australia.

OK, interestingly WiNet-S shows 'Derating Running', which is good to know and I assume the meaning of 0x80 / 128 ... but I can't explain why it would be in that state while the other two inverters are running OK.

[image]

Gnarfoz, 2024-10-20:
Good question... At least adding the running state to the list should be ready easy to do.

bleughb, 2024-10-20:
De-Rating running seems correct. I have two SG5.0RS here , and often one will start 'de-rating' whilst the other one won't. Check the local grid voltages, very often with high voltage, they'll start de-rating,
if one gets more sun and the temperature increases, it'll start de-rating.

Gnarfoz, 2024-10-21:
@mkaiser, I've created a pull request to treat this state the same as the existing derating state: #385.

commit d4558243 (2024-10-22): Add 0x0080 system state as "de-rating running", fixes #383 (#385)

* Add 0x0080 system state as "de-rating running", fixes #383

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/383
closed 2024-10-22 by mkaiser