mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #386: Modbus unavailable after adding line 350 swap: word

Matze-aaa, 2024-10-22:
After adding
350        swap: word
Modbus is no longer available.
Deleting of the line and everything works again.

mkaiser, 2024-10-22:
thanks for the quick reply. Reverted the commit and will check later

mkaiser, 2024-10-22:
lets continue investigating here #360

closed 2024-10-22 by mkaiser

RafAustralia, 2024-10-22:
> After adding 350 swap: word Modbus is no longer available. Deleting of the line and everything works again.

Could you please paste full register section
I have not seen it fail in last 5 HA installs Ive done and these included RS, RT and T type. 

So I suspect a typo somewhere. 

I also need to note: 
I very much doubt adding swap: word would kill modbus to the point of not being available. But wrong spacing or white spacing or typos might. 

Very keen to see your code. Thanks. 

Raf

dylan09, 2024-10-23:
> After adding
> 350 swap: word

Did you also change the register type to 32 bit? Without that, `swap: word` will lead to an error.
Can you show the log file with the error after changing?

RafAustralia, 2024-10-23:
> > After adding
> > 350 swap: word
> 
> Did you also change the register type to 32 bit? Without that, `swap: word` will lead to an error. Can you show the log file with the error after changing?

U r correct. Swap: word is for 32bit registers. 
and I suspect a typo. So awaiting to see where the code is wrong. Your case worked and so anyone elses.

Matze-aaa, 2024-10-23:
Hey guys, you're right, sorry. I only added the marked line swap: word without changing int16 to int32. Now the issue is gone.

RafAustralia, 2024-10-23:
> Hey guys, you're right, sorry. I only added the marked line swap: word without changing int16 to int32. Now the issue is gone.

@mkaiser @Matze-aaa 
Glad it has resolved. Cheers

Raf