mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #390: Modbus does not connect anymore

timkolloch, 2024-10-27:
**Describe the bug:**

After an update of HomeAssistant my modbus does not connect anymore. Several restarts and updating to the newest modbus_sungrow.yaml did not solve the problem.

These are the log statments containing `modbus` I found:

```
2024-10-27 17:49:43.667 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 104] Connect call failed ('192.168.178.14', 502)
2024-10-27 17:49:43.667 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus SungrowSHx communication open
2024-10-27 17:49:48.582 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Error: device: 1 address: 13045 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.14:502]
```

Are there any ideas what could be wrong?

**Your Sungrow inverter:**
 - Model: SH10RT
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [ ] WiNet-S (LAN)
   - [ ] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [ ] yes
  - [ ] no 
  - [X] I don't know what that is

**Home Assistant version:**
Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.2
Frontend 20241002.4

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-10-24 (2)
 - [X] I ensured to use the most recent version

**Inverter Firmware Status:**
   - [ ] I made sure that the newest firmware is installed via the installers account
 
I do now know as I do not have an installers account to log in to the web access of my inverter. 
But the company that installed the device said they will take care of software upgrades.
Where do I find information about that?

**To Reproduce**
Not much to reproduce here I think.

Gnarfoz, 2024-10-28:
The issue is more likely with the inverter. The Modbus stack on the inverter is notoriously finnicky. Try unplugging its network cable for a minute or so.

Maggie70, 2024-10-28:
[image]

Maggie70, 2024-10-28:
restart modbus in developer tools

timkolloch, 2024-10-29:
> The issue is more likely with the inverter. The Modbus stack on the inverter is notoriously finnicky. Try unplugging its network cable for a minute or so.

Yes, that did solve it. I will know for the future, thank you!

timkolloch, 2024-10-29:
> restart modbus in developer tools

That sadly did not solve it but unplugging and plugging it back in helped. 
Thank you for your time!

closed 2024-10-29 by timkolloch