mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #394: String does not match the pattern of "LEGACY_SYNTAX^".

davidhoffelner83, 2024-10-31:
After installing the modbus_sungrow.yaml file today for a new integration I got 28 errors in the automation section. 
- platform: state      String does not match the pattern of "LEGACY_SYNTAX^"

 - Model: SH8.0RT-V112
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [X] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant version:**
Home Assistant
Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.2
Frontend 20241002.4

**modbus_sungrow.yaml:**
 - Version/ time stamp : last update: 2024-10-24 (2)

**Expected behavior**
File without any issues.

**Screenshots**
[image]

Gnarfoz, 2024-10-31:
This is merely a style checker issue in your text editor. The syntax is still valid. (Hence: "LEGACY_SYNTAX")

There are long term plans to move to the new syntax once there's been some time for the user base to update to an adequately recent release. The October 2024 release of Home Assistant is the first one that qualifies, so it is a bit early, yet.

See also #367 and #369.

davidhoffelner83, 2024-10-31:
thanks for the fast response

closed 2024-10-31 by davidhoffelner83

MrMontesa, 2024-11-01:
Just found this warning today as well and wanted to say "thank you" to both of you. a) for reporting this already so I was able to find it and b) for keeping this awesome integration live and well maintained.

mkaiser, 2024-11-01:
Hey, I will commit the patches addressing the legacy syntax as soon as 2024.11 is released

I am curious - I use VSCode within HA (addon) as well as on my local Desktop PC. But I don't get these syntax style stuff. Do you use an extension?

davidhoffelner83, 2024-11-01:
No, installed everything from scratch. No much add-ons

MrMontesa, 2024-11-01:
Same here. only addons are python & python-debugger, which I guess are default.
As soon as I open modbus_sungrow.yaml editor goes red including the folder modbus file is located in and spitting 9+ errors
[image]

commit ee98c931 (2024-11-02): Fix #394 String does not match the pattern of "LEGACY_SYNTAX^".