mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #397: Some entities not updating for hours

djranm, 2024-11-05:
**Describe the bug:**

Over the last couple of days I've noticed some entities are not updating. This is happening across multiple inverters. Prior to the last couple of days everything was working ok. Not all entities are affected and I can't see any pattern.

**Your Sungrow inverter:**
 - Model: SH-10.RT v112 & SH-10.RT
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2024.10.02

**modbus_sungrow.yaml:**
 - Version/ time stamp: 2024-10-XX
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Open history for a given entity, e.g. battery level.
2. Observe that the data is stale and the 'last update' time is minutes/hours in the past.

**Expected behavior**
Entities should be updated periodically (seconds to minutes).

**Screenshots**
[image]
[image]
[image]
[image]

Wolkenstuermer, 2024-11-08:
Sometimes, if you are playing with your code, Entity ID will change like sensor.battery_level to sensor.battery_level_2. There new data might be logged.

I had this plenty of times, when i started with HA and also when testing my setup with two inverters.

userofgithub023897, 2024-11-16:
I experience exactly the same issue.

## Problem

Some sensors won't update for hours.

- sensor.daily_pv_gerneration
- sensor.total_dc_power
- sensor.load_power
- etc.

Some sensors aren't available anymore.

[image]

[image]

## What I tried

- Unplugged WiNet-S (LAN) and other LAN cable then reconnected.

- updated Home Assistant to the latest version.

- tried several `modbus_sungrow.yaml` versions, as well. At the moment I use `last update: 2024-11-02`.

The result is as shown in the screenshots above.

Any ideas?

userofgithub023897, 2024-11-28:
I run 

Core: `2024.11.3`
Supervisor: `2024.11.4`
Operating System: `13.2`
Frontend: `20241106.2`

Some sensors still don't get updates. 

The sensor `sensor.battery_charging_power` and `sensor.battery_discharging_power` show always `0 W`.

[image]

[image]

The battery SoC status seems to be correct when I compare it to the original data in Sungrow App.

jumahrauf, 2024-12-19:
Same issue here. Some sensors are not working correctly after the last version.
I switched back to a version from October.

closed 2025-04-20 by github-actions[bot]