mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #399: Invalid value for input_number.set_sg_battery_max_discharge_power: 10600.0 (range 100.0 - 5000.0)

JoJoJo2211, 2024-11-09:
**Describe the bug:**

In total I have three error messages, but in my opinion they describe the same problem? See screenshot

**Your Sungrow inverter:**
 - Model: [SH10RT-V112] [WiNet-S] [SBR096]
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [X] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant version:**
 - Version: [2024.11.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [2024-11-02]
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account
   -Unfortunately, I don't know where I can update a firmware? It doesn't show me anywhere which version of the firmware is installed

**To Reproduce**
Steps to reproduce the behavior:
The error message just appears without having done anything beforehand

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

Gnarfoz, 2024-11-09:
You probably updated the `modbus_sungrow.yaml` file and overwrote the changes to the maximum value for battery charge/discharge. Out of the box, they are set to 5000 W. You need to adjust them to fit your actual battery. If you update the `modbus_sungrow.yaml` file by simply overwriting it, you'll have to re-do the adjustment.

The limits can be found here:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/4cc83ca0e7bc6963edb767edb3eda70e1bc8c35f/modbus_sungrow.yaml#L2502
and here:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/4cc83ca0e7bc6963edb767edb3eda70e1bc8c35f/modbus_sungrow.yaml#L2508

JoJoJo2211, 2024-11-11:
Thanks for the tip, does that mean in my case the value 10600 as reported in the error?
The data sheet for the battery says something like:
Energy (usable) 9.6 kWh and 
DC rated power 5.76 kw

Gnarfoz, 2024-11-11:
For an SBR096 the relevant number would be 4500 W.
For the error to disappear, you just need to set both of the sliders / values to a number that is within the configured limits. It's possible that you first need to increase the limit so that the current value (10600) is within the limit, to be able to lower it.

JoJoJo2211, 2024-11-11:
Thanks for the contribution, why 4500 watts? The battery power is stated as 9.6 kWh?

Gnarfoz, 2024-11-11:
W and Wh are not the same thing. The installer training recommends to use (maximum continuous charging current × minimum operating voltage = charge/discharge power limit), which for SBR096 comes out to: 30 A × 150 V = 4500 W.

AFAIK, the remaining margin like the 42 A over 10 s and so on come into play when certain loads temporarily exceed the limits, like a fridge turning on. It might temporarily pull 5x or 10x its normal current.

JoJoJo2211, 2024-11-11:
Thank you very much for the tip. Now it seems to be working. Should I now set the value in yaml back to 5000?

Gnarfoz, 2024-11-11:
I'd set it to 4500, so you can never accidentally set it higher than that.

JoJoJo2211, 2024-11-11:
Good point, thank you

The data sheet says something about a nominal voltage of 192V for the SBR096. Shouldn't it then be 30A x 192V = 5760 watts?

Gnarfoz, 2024-11-11:
Yeah, I thought so as well, but the SBR installation training for technicians says otherwise. 🤷

JoJoJo2211, 2024-11-11:
Okay, then I'll leave it at 4500 watts.

I still have two template sensors from the dashboard that are showing me an error, can you help me with that too?

Gnarfoz, 2024-11-11:
Sure, one can always try. 😉
Best to open a new discussion thread for separate issues, though.

JoJoJo2211, 2024-11-11:
Okay, thank you very much for your help. Then the problem should be solved now :)