mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #401: Multiple: expected float for dictionary value @ data['value'] in log

gjsiiger, 2024-11-11:
I'm seeing a lot of errors in the log when reloading YAML config, when restarting Home Assistant or when the external EMS is updating parameters.

2024-11-11 21:19:55.145 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Error: device: 1 address: 5740 -> pymodbus returned isError True
2024-11-11 21:19:57.084 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Error: device: 1 address: 13073 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.198.174:502]
2024-11-11 21:19:57.127 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update] sungrow inverter max SoC input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.128 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update] Error while executing automation automation.sungrow_inverter_max_soc_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.133 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_min_soc_input_slider_update] sungrow inverter min SoC input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.134 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_min_soc_input_slider_update] Error while executing automation automation.sungrow_inverter_min_soc_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.142 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_export_power_limit_update] sungrow inverter export power limit update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.142 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_export_power_limit_update] Error while executing automation automation.sungrow_inverter_export_power_limit_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.159 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_reserved_backup_soc_input_slider_update] sungrow inverter reserved backup SoC input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.160 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_reserved_backup_soc_input_slider_update] Error while executing automation automation.sungrow_inverter_reserved_backup_soc_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.163 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_max_charge_power_input_slider_update] sungrow inverter update battery max charge power input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.164 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_max_charge_power_input_slider_update] Error while executing automation automation.sungrow_inverter_update_battery_max_charge_power_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.194 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_max_discharge_power_input_slider_update] sungrow inverter update battery max discharge power input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.195 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_max_discharge_power_input_slider_update] Error while executing automation automation.sungrow_inverter_update_battery_max_discharge_power_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.200 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_charging_start_power_input_slider_update] sungrow inverter update battery charging start power input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.201 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_charging_start_power_input_slider_update] Error while executing automation automation.sungrow_inverter_update_battery_charging_start_power_input_slider_update: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.216 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_discharging_start_power_input_slider_update] sungrow inverter update battery discharging start power input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
2024-11-11 21:19:57.217 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_discharging_start_power_input_slider_update] Error while executing automation automation.sungrow_inverter_update_battery_discharging_start_power_input_slider_update: expected float for dictionary value @ data['value']

**Your Sungrow inverter:**
 - Model: [e.g. SH-06.RT v112]
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.11.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2024-11-02] (see header of the file)
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

SNAPE15, 2024-11-12:
is that since updating to Core 2024.11?
My modbus is throwing errors and not working since 2024.11 Rolled back to 2024.10 and works. seems they have changed a lot of Modbus values/data that i need to clean/repair the yaml

Gnarfoz, 2024-11-13:
Looks very odd, but the second line already says that there's a communication problem between Home Assistant and your inverter. I think this probably not a problem with this project.

closed 2025-03-13 by github-actions[bot]