mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #406: sg_battery_power_raw Data Type changed in latest Firmware?

ksFPV, 2024-11-23:
i'm using the latest yaml

i updated the firmware of my SH10RT-20 yesterday.
MDSP - from  SAPPHIRE-H_03011.95.01 to SAPPHIRE-H_03011.95.03
LCD - from  SAPPHIRE-H_01011.95.01 to SAPPHIRE-H_01011.95.03

 - Model: SH-10.RT-20
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)

- Are you using a Modbus Proxy (mark one)
  - [x] yes

**Home Assistant version:**
 - Version: 2024.11.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024.11.3

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
As soon as my Battery charges it shows values like 65kw Charging power, iSolarcloud showing correct charging power values

**Expected behavior**
Showing correct values

i've changed the Data Type from uint16 to int16, wich solves the Values, but then goes negative and due to that breaks some templates i guess, haven's looked into those yet

      - name: Battery power raw
        unique_id: sg_battery_power_raw
        device_address: !secret sungrow_modbus_slave
        address: 13021 # reg 13022
        input_type: input
        data_type: int16
        precision: 0
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

So i think the Data Type changed somehow in the latest Firmware?

bjornenvall, 2024-11-23:
[image]

Meaparvitas111, 2024-11-23:
Hello, 

I have changed the data types of sg_battery_current and sg_battery_power_raw vom uint16 to int16.

The absolute values seem to be correct now but the system shows the battery discharging instead of charging.

I had inverted the sign of the charging power in a template sensor for the sunsynk power flow card. I changed the sign again and now everything seems to be correct again.

Kind regards

Alex

Korsberg, 2024-11-23:
Nice ! I changed thesis well 'sg_battery_power_raw' from uint16 to int16. and now its show correct value when charging :-)

Korsberg, 2024-11-23:
Now is negative charging .. can I change this ?

Meaparvitas111, 2024-11-23:
As I said, you have to invert the sign of your sensor in your configuration.

Korsberg, 2024-11-23:
Not sure what step I need do for invert the sign in the configuration ???

Meaparvitas111, 2024-11-23:
I do not know what configuration you have. Generally you can create a new template sensor which is inverted and use this instead of your old sensor:

`  - sensor:
    - name: "Export_Power_negativ" 
      unique_id: "Export_Power_negativ"
      device_class: power
      unit_of_measurement: "W"
      state: "{{ states('sensor.export_power')|float(0) * -1 | round (3) }}"
         `

Korsberg, 2024-11-23:
I only have this sensor 'sensor.signed_battery_power' that showed negative for import and positive for charging , but now is opposite

Meaparvitas111, 2024-11-23:
Try

 - sensor: - name: "sensor.signed_battery_power_inverted"  unique_id: "signed_battery_power_inverted" device_class: power unit_of_measurement: "W" state: "{{ states('signed_battery_power)|float(0) * -1 | round (3) }}"

Korsberg, 2024-11-23:
this will not work , I also have a few binary sensor that show wrong info and Flow Cards and other things, I hope for a permanently fix for 'sensor.signed_battery_power ' so it works like before :-)

BUT your code worked fine for just this sensor

Korsberg, 2024-11-23:
Now all seems to be working ?? strange haha

bjornenvall, 2024-11-23:
thanks for the fix!
i noticed one more thing after this update of the firmware: the inverter doesn't go into standby when it used to. (when the batteri is empty and the sun is under the horizon.)

anyone else notised this problem?

fwittenfeld, 2024-11-24:
Added my hotfix. Not sure how to integrate this upstream as the code depends on the firmware version.

Meaparvitas111, 2024-11-25:
@fwittenfeld 
Thank you, but I don't know what the blue lines mean.
I am a bit afraid that many dashboards and automations will not be working anymore if I do these changes.
@bjornenvall 
My SH did not go into Standby Mode in the past also, at least because of the backup reserve. I use an automation to send it to sleep after sunset and when the battery cannot be discharged any further.

Kind regards

Alex

Korsberg, 2024-11-26:
Noo, now my sensor.signed_battery_power show only negative values both when discharging a charging … is there an update yaml file for me to try ?

Schwenksen, 2024-11-27:
> Found a updated from https://github.com/fwittenfeld :-) works perfect

I also using [modbus_sungrow.yaml] (https://github.com/fwittenfeld/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml) from fwittenfeld.

But I had to fix dashboards and exchange sensors.

fwittenfeld, 2024-12-08:
Has anyone else had problems with the inverter going into shutdown mode after work and not restarting by itself since the new firmware?

Schwenksen, 2024-12-08:
> Has anyone else had problems with the inverter going into shutdown mode after work and not restarting by itself since the new firmware?

Yes, the sun was shining and the inverter was in shutdown mode. I tried several times to start up, but after few minutes the inverter went down again. Than I changed from ha to isolarcloud in admin mode. That boot up was successful without unmotivated shutdowns. 

But I really would like to downgrade this firmware, but the firmware github repository is closed down.

fwittenfeld, 2024-12-08:
My working hypothesis: The firmware has problems with register queries during startup.
If I trigger the start and then take Homeassistant offline for a few minutes, the start works. (only tested once)

Matze5593, 2024-12-08:
Has anyone ever tried scale: -1 after switching from uint16 to int16
add to get the value positive?

Schwenksen, 2024-12-08:
> My working hypothesis: The firmware has problems with register queries during startup. If I trigger the start and then take Homeassistant offline for a few minutes, the start works. (only tested once)

I can confirm this

Matze5593, 2024-12-08:
There is definitely a problem with parallel inverters in master and slave. This was confirmed to me by support. I also asked about the registers. But I still haven't received an answer.

fwittenfeld, 2024-12-08:
> For me it works with scale: - 1 and the conversion to int16

Since this version, the value battery_power is signed. This means that it is negative for charging and positive for discharging the battery.
Scale -1 merely causes the sign to be reversed.
I also used this trick in my PR to get the signed value according to mkaiser's definition. However, some further adjustments were necessary.

bart83-tech, 2024-12-08:
Has anyone else had problems with the inverter going into shutdown mode after work and not restarting by itself since the new firmware?

I have the same problem. The sungrow_inverter_state Entität seems to be the reason.

Matze5593, 2024-12-08:
> Hatte sonst noch jemand Probleme damit, dass der Wechselrichter nach der Arbeit in den Shutdown-Modus ging und seit der neuen Firmware nicht mehr von selbst neu startet?
> 
> Ich habe das gleiche Problem. Die sungrow_inverter_state Entität scheint der Grund zu sein.
**Ja das ist bei mir auch so, ganzschön nervig. ich würde die firmware gerne downlgaden aber das git firmware sungrow wurde geschlossen und ich habe nur eine firmware von 2023 auf dem handy**

> > Bei mir funktioniert es mit scale: - 1 und der Konvertierung in int16
> 
> Seit dieser Version ist der Wert battery_power vorzeichenbehaftet. Das heißt, er ist beim Laden negativ und beim Entladen des Akkus positiv. Scale -1 bewirkt lediglich eine Vorzeichenumkehr. Diesen Trick habe ich auch in meinem PR verwendet, um den sg_signed_battery_power-Wert nach mkaisers Definition zu erhalten. Allerdings waren noch einige weitere Anpassungen nötig.

**das ist gut möglich, leider ist nicht wirklich viel sonne da um das ganze ausreichend zu testen, deswegen habe ich gefragt ob dies schon mal jemand versucht hat. hast du ne Anleitung was du genau alles geändert hast? Zudem ist die frage ob sungrow es in der nächste Firmware wieder rückgängig macht..**

bart83-tech, 2024-12-08:
> **Ja das ist bei mir auch so, ganzschön nervig. ich würde die firmware gerne downlgaden aber das git firmware sungrow wurde geschlossen und ich habe nur eine firmware von 2023 auf dem handy** [...]

Vielleicht hilft es bis der Fehler behonben ist, die Automation **sungrow inverter state** und **sungrow inverter enable/ stop input selector update** zu deaktivieren. Ich teste dies mal und schaue ob der Wechselrichter heute Abend wieder in den Shutdown Modus geht.

MIRR23, 2024-12-09:
> There is definitely a problem with parallel inverters in master and slave. This was confirmed to me by support. I also asked about the registers. But I still haven't received an answer.

@Matze5593 Hast du schon ne Antwort von Sungrow? Habe auch zwei Wechselrichter Master/Slave und jeweils die neue Firmware aufgespielt. Seither Fehler 514 (Energiezähler Alarm Kommunikationsanomalie) beim Slave. Scheint damit zusammen zu hängen. Beim Support erreiche ich leider niemanden.

Gnarfoz, 2024-12-09:
Not to be rude, but this is a bug tracker for a Home Assistant integration, perhaps you should take this discussion to somewhere like https://www.photovoltaikforum.com/? There are probably more people there who can help.

Matze5593, 2024-12-09:
> @Matze5593 Hast du schon ne Antwort von Sungrow? Habe auch zwei Wechselrichter Master/Slave und jeweils die neue Firmware aufgespielt. Seither Fehler 514 (Energiezähler Alarm Kommunikationsanomalie) [...]

**genau das Problem habe ich, aktuell ist die Antwort:

Wir möchten Sie darüber informieren, dass wir derzeit auf weitere Informationen unserer Zentrale zu Ihrer Anfrage warten. Wir sind uns der Wichtigkeit Ihrer Anfrage bewusst und versichern Ihnen, dass wir Ihnen die erforderlichen Informationen umgehend nach Erhalt zukommen lassen.

Vielen Dank für Ihre Geduld und Ihr Verständnis. Wenn Sie in der Zwischenzeit weitere Fragen haben, zögern Sie bitte nicht, uns zu kontaktieren.

**

Auf die Frage Sie die alte Firmware einspielen wurde nicht reagiert.

CremaChris, 2024-12-15:
I gave up. The 95.03 firmware just has too many issues, creates too many side effects with HA and evcc. I downgraded to 95.01 and everything is back in order.

Alan3715, 2024-12-15:
> I gave up. The 95.03 firmware just has too many issues, creates too many side effects with HA and evcc. I downgraded to 95.01 and everything is back in order.

Hello, could you maybe share the old software or tell me where I can find it, I also want to downgrade. I would really appreciate it. Thank you.

CremaChris, 2024-12-15:
You need the FW that exactly matches your SG model. I only have the FW for SH10RT-20.I received it from a guy in Photovoltaik Forum. BRChris Gesendet mit der mobilen Mail AppAm 15.12.24 um 13:13 schrieb Alan3715
                
            
            
                
                    Von: "Alan3715" ***@***.***>Datum: 15. Dezember 2024An: "mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant" ***@***.***>Cc: "Comment" ***@***.***>,"CremaChris" ***@***.***>Betreff: Re: [mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant] sg_battery_power_raw Data Type changed in latest Firmware? (Issue #406)
                
                

I gave up. The 95.03 firmware just has too many issues, creates too many side effects with HA and evcc. I downgraded to 95.01 and everything is back in order.

Hello, could you maybe share the old software or tell me where I can find it, I also want to downgrade. I would really appreciate it. Thank you.

hboltz, 2024-12-18:
> The 95.03 firmware just has too many issues, creates too many side effects with HA and evcc. I downgraded to 95.01 and everything is back in order. Hello, could you maybe share the old software or tell me where I can find it, I also want to downgrade. I would really appreciate it. Thank you. >

I have the same issue - discharging value seems to be okay, but when the battery is charging, I see values round about 65500W. If anybody has a source for the 95.01 firmware or below, it would be great!

Schwenksen, 2024-12-18:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/406/2551628261@
> github.com>

I downloaded it from Photovoltaik Forum. After downgrade to 95.01
everything works once again.

Schwenksen, 2024-12-18:
Here the download link
https://www.photovoltaikforum.com/core/attachment/516161-sapphire-h-b001-v000-p009-20240705-zip/

But check twice if this firmware fits your device!

hboltz, 2024-12-18:
Hi Oliver,

how can I see, if it‘s the right one for my SH8.0RT-V112? I can see in Update report, that the Update to version looks like:

Originalversion:
SAPPHIRE-H_03011.95.01 (MDSP)
SAPPHIRE-H_01011.51.08 (LCD)

New Version:
SAPPHIRE-H_03011.95.01 (MDSP)
SAPPHIRE-H_01011.51.08 (LCD)

???

Schwenksen, 2024-12-19:
> Hi Oliver,
> 
> how can I see, if it‘s the right one for my SH8.0RT-V112? I can see in Update report, that the Update to version looks like:
> 
> Originalversion: SAPPHIRE-H_03011.95.01 (MDSP) SAPPHIRE-H_01011.51.08 (LCD)
> 
> New Version: SAPPHIRE-H_03011.95.01 (MDSP) SAPPHIRE-H_01011.51.08 (LCD)
> 
> ???
> 
> > […](#)

Sorry for responding late, but I saw that you got your answer.

bjornenvall, 2024-12-20:
I would also like to downgrade my SH8.0RT-V112, how do I do it? And is there really any danger in doing it?

hboltz, 2024-12-20:
You can do the downgrade on the iSolarCloud Website. Go to Firmwarethe, choose the hybrid inverter and there Local Update package, upload the Firmware zip file and klick enter. Thats it.

bjornenvall, 2024-12-20:
> You can do the downgrade on the iSolarCloud Website. Go to Firmware, choose the hybrid inverter and there Local Update package, upload the Firmware zip file and klick enter. Thats it.

Yeah, that part I know, but the part that I don’t understand is how I get/know what firmware that fits my SH8.0RT-V112?

And if I can break it if I try to downgrade with the wrong zip file?

hboltz, 2024-12-20:
> Yeah, that part I know, but the part that I don’t understand is how I get/know what firmware that fits my SH8.0RT-V112?
>
> And if I can break it if I try to downgrade with the wrong zip file?

You can use the file from the download link in this threat above. The firmware file is the right one for your inverter model. I have the same one and used the same file. It works!

stalakerob, 2024-12-23:
I'm using the fwittenfeld version now and for me everything seems to work fine right now without any changes in my configuration. I'm using the power-flow-card-plus custom component (no dashboards from this repository).

ABredhauer, 2025-03-08:
Mentioning it because I can't see it in the comments but this change did not impact all models and all firmware. My SH10RS is running the latest firmware and still shows this as an unsigned int

Gnarfoz, 2025-03-08:
There's a _signed battery power_ sensor now that should combine the various options and work for both signed and unsigned.

ABredhauer, 2025-03-09:
Sorry I hadn't worked my way down that far in the file. I was merging the latest version with my own local customisations to make it compatible with my inverter. I can see the other sensor. This change still works on my inverter but raises the question how long we keep continuing with one version of the file when there are other things that don't