mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #408: No Entities After Update

itzolino2024, 2024-11-24:
**Home Assistant version:**
Home Assistant

Core
2024.10.4
Supervisor
2024.11.4
Operating System
13.2
Frontend
20241002.4

**modbus_sungrow.yaml:**
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

When i update to 2024.11 my Sungrow modbus isnt working anymore. It worked for one year without any problems but the new update destroys it. No entities avaible anymore. What to do? Thank! I already changed to the neueste Version with actions instead of  action in the yaml

Gnarfoz, 2024-11-24:
Step 1: look at your Home Assistant logs and find an actual error message.

itzolino2024, 2024-11-28:
Hi!
Thanks! I made the update again and her are some screenshots.
[image]
[image]
[image]
[image]

Gnarfoz, 2024-12-02:
The error messages refer to a `modbus_sungrow2.yaml`. Such a file is not part of this integration. Did you leave a backup copy in the directory?

itzolino2024, 2024-12-03:
No, i have two inverter working. The First One is connected with the Battery. Both are working Perfect with older Version of Home Assistant

mkaiser, 2024-12-03:
Please show us thefirst 50 lines of your Modbus_sungrow_2.yaml and make sure that the indentation is not changed by posting.

I suspect some issues there.

itzolino2024, 2024-12-03:
`
modbus:
  - name: SungrowSHx_inv2
    type: tcp
    host: !secret sungrow_modbus_host_ip_inv2
    port: !secret sungrow_modbus_port_inv2
    retries: 10
    sensors:
      - name: Sungrow device type code 2
        unique_id: sg_dev_code_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 4999 # reg 5000
        input_type: input
        data_type: uint16
        scan_interval: 600

      - name: Daily PV generation & battery discharge 2
        unique_id: sg_daily_pv_gen_battery_discharge_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5002 # reg 5003
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Total PV generation & battery discharge 2
        unique_id: sg_total_pv_gen_battery_discharge_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5003 # reg 5004
        input_type: input
        data_type: uint32
        swap: word
        unit_of_measurement: kWh
        precision: 1
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Inverter temperature 2
        unique_id: sg_inverter_temperature_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5007 # reg 5008
        input_type: input
        data_type: int16
        precision: 1
        unit_of_measurement: °C
        device_class: Temperature
        scale: 0.1
        scan_interval: 10
`

Gnarfoz, 2024-12-03:
Also, did you update the second file when you updated the first one? 😉

itzolino2024, 2024-12-03:
No, I only updated the first one, you are right. That's an idea?

Gnarfoz, 2024-12-03:
Yep. You need to remove `retries`.

itzolino2024, 2024-12-03:
I can't find the update manual anymore? Can you help me? Sorry!

itzolino2024, 2024-12-03:
That's it!

THANK YOU SO SO MUCH!

closed 2024-12-03 by itzolino2024