mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #410: New modbus register for buffer SoC?

leo4417, 2024-11-28:
Actually it isn't possible to read/write the buffer SoC. Over ISolarCloud the new (?) name is Notlade-SoC. Probably it's a new holding register?

Sungrow SBR256
SH10RT

Gnarfoz, 2024-12-02:
Can you elaborate on what you mean by "buffer SoC"?

And what is its relation to the emergency charge SoC (SoC threshold at which the battery will charge from the grid to prevent damage), which should not be changed at all?

itzolino2024, 2024-12-03:
`
modbus:
  - name: SungrowSHx_inv2
    type: tcp
    host: !secret sungrow_modbus_host_ip_inv2
    port: !secret sungrow_modbus_port_inv2
    retries: 10
    sensors:
      - name: Sungrow device type code 2
        unique_id: sg_dev_code_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 4999 # reg 5000
        input_type: input
        data_type: uint16
        scan_interval: 600

      - name: Daily PV generation & battery discharge 2
        unique_id: sg_daily_pv_gen_battery_discharge_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5002 # reg 5003
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Total PV generation & battery discharge 2
        unique_id: sg_total_pv_gen_battery_discharge_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5003 # reg 5004
        input_type: input
        data_type: uint32
        swap: word
        unit_of_measurement: kWh
        precision: 1
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 600

      - name: Inverter temperature 2
        unique_id: sg_inverter_temperature_2
        slave: !secret sungrow_modbus_slave_inv2
        address: 5007 # reg 5008
        input_type: input
        data_type: int16
        precision: 1
        unit_of_measurement: °C
        device_class: Temperature
        scale: 0.1
        scan_interval: 10
`

closed 2025-04-03 by github-actions[bot]