mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #418: Error Message

derderneburger, 2024-12-19:
Error Message in Home Assistant Core Version 2024.12.4
Supervisor Version 14.0
HA running on Proxmox

Sungrow SH10RT-V112

I am receiving the following error in the system log:

Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:306
Integration: Modbus (Documentation, Issues)
First occurred: 08:56:00 (12 occurrences)
Last logged: 10:46:00

Pymodbus: SungrowSHx: Error: device: 1 address: 6226 -> pymodbus returned isError True

Can I ignore this? I am only using the values for the Energy Dashboard. Everything seems to be working fine.

Gnarfoz, 2024-12-19:
Register 6227 is "monthly PV energy yields" for January, and is not available via WiNet-S: https://github.com/Gnarfoz/Sungrow-Inverter/blob/main/Modbus%20Information/TI_20240924_Communication%20Protocol%20of%20Residential%20Hybrid%20Inverter-V1.1.5.pdf

You have a SH10RT which has a dedicated Ethernet port just for Modbus. It's recommended to use that one instead of going through the WiNet-S dongle.

If that's not possible, you can ignore the error or comment out all the sensors that the Modbus register documentation lists as unavailable on WiNet-S.

derderneburger, 2024-12-19:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/418/2553660473@
> github.com>

Hi,

Iam using the dedicated Ethernet Port for modbus. I have connected Both
Ports Witz my Router. The dedicated one and the WiNet-S Dongle.

Gnarfoz, 2024-12-19:
I'd double check IP addresses, just to be sure. Or unplug the dedicated Ethernet port. If your sensors still update while it's unplugged, you're accidentally using the WiNet-S.

derderneburger, 2024-12-19:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/418/2555522209@

It seems that I had the wrong IP entered all along. Now the issue appears
to be resolved. I never noticed anything wrong during normal operation.
Thanks for the push to check that.

Gnarfoz, 2024-12-19:
Excellent, glad to hear that. 🙂

closed 2025-04-20 by github-actions[bot]