mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #420: After firmware update limits for battery max charge/discharge power have changed

KESchnitzler, 2024-12-21:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug: with the old firmware these values could be set to numbers >5000, now after restart (due to firmware forced update) the read out values were set to 100W each and its no longer possible to set the former values of 5730W (max. due to datasheet , the error message returned informed about a limit <= 5000W**

A clear and concise description of what the bug is.

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: actual one]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2024-10-24] (see header of the file)
 - [] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

KESchnitzler, 2024-12-21:
sorry I've missed the actual firmware rev responsable for this behavior: 
ARM-Softwareversion | SAPPHIRE-H_01011.95.03
MDSP-Softwareversion | SAPPHIRE-H_03011.95.03
SDSP Softwareversion | SUBCTL-S_04011.01.01
K-E

Gnarfoz, 2024-12-21:
When you updated the `modbus_sungrow.yaml` file to the most recent version, did you make sure to re-apply any changes you had made before?
Namely, the maximum values for forced/charge/discharge powers:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2493-L2509

KESchnitzler, 2024-12-21:
Hi, maybe I wasn't clear enough, the probleme came up with the update of wr and battery, the yaml-code remaind unchanged and when looking in the history of battery max charge/dischare power its clearly to see, that the value changed in the moment of restart after firmware change, that wasn't initiated by me but by the solateur. so that tells me, that it must be driven by changing the limits of the firmware!??

Gnarfoz, 2024-12-21:
So, the `max_value` properties in the section I linked are set to... which value?

KESchnitzler, 2024-12-21:
you're right, they are set to 5000 but the history showed for months the 5760w, so I must have changed the value in early days and havn't changed it for a long time, so after the startup value after the firmware reset was possibly set to 100 (this was the value as I recognized the change). trying then to set the value to the wanted one, the limit check would alert me, cause i have missed to change the values while changing the rev of the integration... thanks for waking me up!! I think everything is now clear to me
regs K-E

Gnarfoz, 2024-12-21:
Excellent. Happy to help. :-)

KESchnitzler, 2024-12-26:
As I already wrote, the problems started after the firmware update on dec 12., (SH10RT now ARM-Softwareversion: SAPPHIRE-H_01011.95.03, MDSP-Softwareversion: SAPPHIRE-H_03011.95.03, SBR096: SBRBCU-S_22011.01.24). The problem is now. that beginning with the day after the update HA gives the "Battery Charge Power" in value arolund 60.000 W (60 kW) which is not very reasonable!? the months before it was, dependung on solar production between 0 and 5000 W. the only values I've changed in the modbus_sungrow.yaml the one quoted above... what may be the reason??
regs K-E

Gnarfoz, 2024-12-29:
The reason is that the data type of a particular Modbus register has changed in firmware version 95.03. Refer to https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/406.
@fwittenfeld was kind enough to propose a fix https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/pull/407/files which you can grab from their fork: https://github.com/fwittenfeld/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main
It's a conundrum: integrating this fix would break the same thing for everyone who **hasn't** updated to that particular firmware version, which most likely is the vast majority of users.
According to Sungrow, firmware updates should only be done when installing new systems, to fix a problem, or at the direction of Sungrow support. I guess they're painfully aware that they have a horribly intransparent versioning system and no public changelogs whatsoever.

closed 2025-04-29 by github-actions[bot]