mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #421: After FW update sensor.battery_charging_power is sometimes wrong

stalakerob, 2024-12-23:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**
After I updated my SH8.0RT-112 firmware to the latest version the battery charing power sometimes peaks to 45kw or higher. My assumption is that the data type of battery power raw has changed so that it can be negative.

**Your Sungrow inverter:**
 - Model: SH8.0RT-112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2024-12-5

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-11-02
 - [] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]

Gnarfoz, 2024-12-23:
See #406.

BigDi, 2025-01-08:
I have same issue. I have also exchange the modbus_sungrow.yaml from fwittenfeld  and updated the dashboard files. I checked if the sg_signed_battery_power is used. But my Battery Charging Power is not changed. Discharging shows the richt minus values but charing shows smoehting about 60kW. Do I need to set anything else?

Gnarfoz, 2025-01-08:
Sounds like the replacement didn't catch everything. I'd double-check that first.
Other thoughts:
Did you maybe make a backup copy of the previous file and left it in the same directory? That could result in unexpected issues as well, Home Assistant would try to load them both.
And: did you fully restart Home Assistant after replacing the files?

Mats1717, 2025-01-27:
I updated the inverter in the weekend and now it is problem with battery charging and discharging.
Have newest .yaml file and running on winet-s port. Running in VPP External EMS mode. Before it worked fine, tried old .yaml file from october but same issue.

Regards Mats

[image]

[image]

[image]

Gnarfoz, 2025-01-27:
What inverter do you have, and which firmware version did you update to?
The current version of this project should include a workaround for the battery register properties changing. Are you sure you replaced it correctly? Did you maybe leave the old file in place in the same directory?

Also, why are you running in external EMS mode? Do you have an external EMS?

Mats1717, 2025-01-27:
> What inverter do you have, and which firmware version did you update to? The current version of this project should include a workaround for the battery register properties changing. Are you sure [...]

I have SH8.0RT V11. Updated through the app that suggested a update. Where can i see firmware version?
The newest .yaml is correct installed, been using this projekt for 2 years now. If it is the wrong firmware is the a way to download correct one?

external EMS mode i because an outside company controlls it to use for frekvency support to the grid, common here in Sweden

Mats1717, 2025-02-04:
> > What inverter do you have, and which firmware version did you update to? The current version of this project should include a workaround for the battery register properties changing. Are you sure you replaced it correctly? Did you maybe leave the old file in place in the same directory?
> > Also, why are you running in external EMS mode? Do you have an external EMS?
> 
> I have SH8.0RT V11. Updated through the app that suggested a update. Where can i see firmware version? The newest .yaml is correct installed, been using this projekt for 2 years now. If it is the wrong firmware is the a way to download correct one?
> 
> external EMS mode i because an outside company controlls it to use for frekvency support to the grid, common here in Sweden
> 
> I think I found the version
> 
> [image]

I still have issues, sometime it right and somtime not, earlier today working fine now its charging all values says discharging.

[image]
Isolarcloud says charging

Mats1717, 2025-02-04:
> I still have issues, sometime it right and somtime not, earlier today working fine now its charging all values says discharging.
>
> [image] Isolarcloud says charging

as in picture forced charging and indicating discharging

[image]

Gnarfoz, 2025-02-04:
Using "Forced charge" and "Forced discharge" requires switching to "Forced mode" instead of "VPP".

Mats1717, 2025-02-05:
> Using "Forced charge" and "Forced discharge" requires switching to "Forced mode" instead of "VPP".
> 
> (I should have seen that in your post from last week...)

I need to have VPP because battery is managed from outside company.
I did a workaround that looks on the value of sensor.battery_forced_charge_discharge_cmd_raw and created 2 new sensors for charging and discharging. It still would be nice to have it working original file

mkaiser, 2025-02-05:
@stalakerob: If not already done: Could you update to  the current version? There were some battery fixes early January. 

@Mats1717: which firmware /yaml is installed?

Gnarfoz, 2025-02-05:
> I need to have VPP because battery is managed from outside company.

You already seem to be OK with ignoring what the outside company does, or you wouldn't be trying to use "force charge" and "force discharge". Changing the EMS mode to "forced mode" **and changing it back to VPP after you're done force charging/discharging** doesn't seem like much more of a "transgression". It's not a requirement of this project, it's just how the inverter works.

>I did a workaround that looks on the value of sensor.battery_forced_charge_discharge_cmd_raw and created 2 new sensors for charging and discharging. It still would be nice to have it working original file

That workaround is just an illusion, though. It only shows you what the forced command is set to, but nothing is actually happening because the EMS mode is wrong.

> [@Mats1717](https://github.com/Mats1717): which firmware /yaml is installed?

They're on inverter 95.03 + battery 24, as per this earlier comment: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/421#issuecomment-2616385721

Mats1717, 2025-02-05:
> > I need to have VPP because battery is managed from outside company.
> 
> You already seem to be OK with ignoring what the outside company does, or you wouldn't be trying to use "force charge" and "force discharge". Changing the EMS mode to "forced mode" **and changing it back to VPP after you're done force charging/discharging** doesn't seem like much more of a "transgression". It's not a requirement of this project, it's just how the inverter works.
> 
> > I did a workaround that looks on the value of sensor.battery_forced_charge_discharge_cmd_raw and created 2 new sensors for charging and discharging. It still would be nice to have it working original file
> 
> That workaround is just an illusion, though. It only shows you what the forced command is set to, but nothing is actually happening because the EMS mode is wrong.
> 
> > [@Mats1717](https://github.com/Mats1717): which firmware /yaml is installed?
> 
> They're on inverter 95.03 + battery 24, as per this earlier comment: [#421 (comment)](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/421#issuecomment-2616385721)

Is that the latest firmware i showed above? cant find any newer? I dont set anything on the inverter outside company do.
I think this sensors work and is not an illusion.
template:
  - sensor:
      - name: "Battery Charging Signed"
        unique_id: battery_charging_signed
        unit_of_measurement: "W"
        icon: mdi:flash
        device_class: power
        state_class: measurement
        state: >
          {% set cmd = states('sensor.battery_forced_charge_discharge_cmd_raw') | int(default=-1) %}
          {% if cmd == 170 %}
            {{ states('sensor.signed_battery_power') | float(default=0) | abs | int }}
          {% elif cmd == 204 %}
            {{ states('sensor.battery_charging_power') | float(default=0) | abs | int }}
          {% else %}
            0
          {% endif %}

      - name: "Battery Discharging Signed"
        unique_id: battery_discharging_signed
        unit_of_measurement: "W"
        icon: mdi:flash
        device_class: power
        state_class: measurement
        state: >
          {% set cmd = states('sensor.battery_forced_charge_discharge_cmd_raw') | int(default=-1) %}
          {% if cmd == 187 %}
            {{ states('sensor.signed_battery_power') | float(default=0) | abs | int }}
          {% elif cmd == 204 %}
            {{ states('sensor.battery_discharging_power') | float(default=0) | abs | int }}
          {% else %}
            0
          {% endif %}

Mats1717, 2025-02-05:
> [@stalakerob](https://github.com/stalakerob): If not already done: Could you update to the current version? There were some battery fixes early January.
> 
> [@Mats1717](https://github.com/Mats1717): which firmware /yaml is installed?

Latest yaml and latest firmware according to what is avalible in the Isolarcloud app, picture in earlir comment

stalakerob, 2025-02-06:
I've updated and for me everything now works fine. Thank you @mkaiser