mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #422: "Platform automation does not generate unique IDs"

ulfben, 2024-12-27:
[The github automation which produces the yaml for the second inverter](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/.github/workflows/auto-generate-second-config.yml) fails to modify one of the identifiers because the ID doesn't follow the pattern used troughout the rest of the file. 

Ergo: can we please change this line:
`[- id: "automation_sungrow_export_power_limit_mode"](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/4cc83ca0e7bc6963edb767edb3eda70e1bc8c35f/modbus_sungrow.yaml#L2832)`

To instead read: 
`- id: "automation_sungrow_inverter_export_power_limit_mode"`

mkaiser, 2024-12-27:
hey, 

good catch! I created a branch for more testing before merging your fix into main:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/unique_id_second_inverter

Can you double check if it is working? (I have only one inverter and therefore cannot test...)

ulfben, 2024-12-30:
Yep! Tried the new config on my setup and the name is fixed. Unfortunately I realized we have another issue too. 

YAML comments, it seems, must be preceded by a whitespace. There are two names ("name: Import power" and "name: Export power") in this file that are annotated with a comment on the same line, and [my regexp](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/4cc83ca0e7bc6963edb767edb3eda70e1bc8c35f/.github/workflows/auto-generate-second-config.yml#L33) breaks these by appending " 2" without moving the start of the comment. 

Here's what we get after regexp runs:
[image]

Inserting a space solves the problem:
[image]

Instead of messing with the regexp, perhaps we can just move the comment to its own line before the name is defined?

Gnarfoz, 2024-12-30:
Take note that that's just your editor's (or rather, its linter's) opinion on what coding style you should follow. It's not an actual YAML requirement.

ulfben, 2024-12-30:
Thank god :D I know YAML is _particular_, but this seemed more than usually stupid. 

However, I suspect [the VSCode addon](https://github.com/hassio-addons/addon-vscode) is very common so it might be worth adressing this to avoid unecessary user confusion.

mkaiser, 2025-01-02:
Moved the comments like you suggested - can you check again, if everything is fixed now?

ulfben, 2025-01-03:
Confirmed! No complaints from HA nor VScode.

closed 2025-01-03 by mkaiser