mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #423: Many errors when querying SG series inverter?

NigelPearson, 2024-12-27:
**Describe the bug:**

Many Home Assistant log errors from battery/SoC automation

**Your Sungrow inverter:**
 - Model: SG3.0RS
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [X] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2023.09.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-11-02
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
1. Install latest YAML integration
2. Set secrets for IP address of non-battery-hybrid inverter
3. Reload all YAML configuration
4. tail -f /homeassistant/home-assistant.log

**Expected behavior**
A few less of these errors:

`2024-12-28 06:46:01.745 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_min_soc] sungrow in
verter update min SoC: Error executing script. Unexpected error for call_service at pos 1: 'SungrowSHx'
`
(about 900 log lines of similar error repeated)
`2024-12-28 06:46:03.102 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({% if states('sensor.running_state')|is_number and states('sensor.running_state')|float > 0 %}`
...
`ValueError: Template error: int got invalid input 'unknown' when rendering template '{% if states('sensor.running_state')|is_number and states('sensor.running_state')|float > 0 %}
  {# use available sensor running_state #} 
  {{ states('sensor.running_state')|int |bitwise_and(0x1) }}
{% else %} 
  {# workaround for SH*RS inverters without working running_state #}
  {% if states('sensor.total_dc_power')|int > 0 %}
    1
  {% else %} 
    0
  {% endif %}
{% endif %}' but no default was specified`
...
`2024-12-28 06:46:03.742 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.phase_b_voltage') | float * states('sensor.phase_b_current') | float) |int }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.phase_b_power'
2024-12-28 06:46:03.760 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (states('sensor.phase_c_voltage') | float * states('sensor.phase_c_current') | float) |int }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2352, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'`
...
`ValueError: Sensor sensor.meter_phase_c_active_power has device class 'power', state class 'measurement' unit 'W' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unknown' (<class 'str'>)`

**Screenshots**
Log file attached

**Additional context**
A lot of these errors are caused by sensors having 'Unknown' for a few seconds, until the modbus read occurs. They can easily be fixed by using this Jinja filter:
|float(0)
or |int(0)

The SungrowSHx automation error though, seems more difficult. Maybe a separate version of your integration for inverters without batteries?
[home-assistant.log](https://github.com/user-attachments/files/18263615/home-assistant.log)

ulfben, 2024-12-30:
I believe I'm having the same problem?

[image]

There's a lot of output but I believe what I'm seeing is: 

1. Value formatting errors in these automations:
```yaml
automation_sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update
automation_sungrow_inverter_max_soc_input_slider_update
automation_sungrow_inverter_min_soc_input_slider_update
```
The automations expect float values but are receiving 'unknown' or non-numeric values.

2. Value range violations in:
```yaml
automation_sungrow_inverter_export_power_limit_update
automation_sungrow_inverter_update_battery_max_charge_power_input_slider_update
automation_sungrow_inverter_update_battery_max_discharge_power_input_slider_update
```

Current values exceed defined limits:
- Export power limit: receiving 11000W vs max 10500W
- Battery charge/discharge power: receiving 10600W vs max 5000W

I guess this is on me to tweak a setting somewhere. Is 11000w what we're reading from the inverter? It matches the capacity of my main fuse so it would make sense. There's a comment in the config <ins>that says 10500W is the max export for the SH10RT, but I wonder if that limit makes sense for master-slave installs?</ins> I've got two inverters and thus could potentially export a combined total far higher than 10kw.

Gnarfoz, 2024-12-30:
Battery charge / discharge power limits are set here (a few lines above the 10500 W comment you mentioned):
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2493-L2509

I'd raise the limits in both cases to be able to adjust the value back down to something reasonable. Altough I can't say what's reasonable for multiple-inverter setups.

Either way, not the same problem as the issue creator has: they're using an unsupported inverter model that uses a different Modbus register layout than the hybrid inverters.

JohnFLoki, 2024-12-30:
The SG series doesn't have the registers for a battery because it is not a hybrid inverter.
Here is the YAML Code for my SG-20 inverter connected over the WiNetS. 
`
# https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
# by Martin Kaiser
# last update: 2024-03-26
#
# PDF: https://github.com/bohdan-s/Sungrow-Inverter/blob/main/Modbus%20Information/Communication%20Protocol%20of%20PV%20Grid-Connected%20String%20Inverters_V1.1.37_EN.pdf
- name: SG-20-WiNet
  type: tcp
  host: IP_HERE
  port: 502
  sensors:
    - name: SG-20 WiNet Inverter temperature
      unique_id: sg_20_winet_inverter_temperature
      device_address: 1
      address: 5007 # reg 5008
      input_type: input
      data_type: int16
      precision: 1
      unit_of_measurement: °C
      device_class: Temperature
      state_class: measurement
      scale: 0.1
      scan_interval: 30

    - name: SG-20 WiNet apparent power
      unique_id: sg_20_winet_apparent_power
      device_address: 1
      address: 5008 # reg 5009 - 5010
      input_type: input
      data_type: uint32
      swap: word
      precision: 0
      unit_of_measurement: VA
      device_class: power
      state_class: measurement
      scale: 1
      scan_interval: 30
      
    - name: SG-20 WiNet MPPT1 voltage
      unique_id: sg_20_winet_mppt1_voltage
      device_address: 1
      address: 5010 # reg 5011
      input_type: input
      data_type: uint16
      precision: 1
      unit_of_measurement: V
      device_class: Voltage
      state_class: measurement
      scale: 0.1
      scan_interval: 10

    - name: SG-20 WiNet MPPT1 current
      unique_id: sg_20_winet_mppt1_current
      device_address: 1
      address: 5011 # reg 5012
      input_type: input
      data_type: uint16
      precision: 2
      unit_of_measurement: A
      device_class: Current
      state_class: measurement
      scale: 0.1
      scan_interval: 10

    - name: SG-20 WiNet MPPT2 voltage
      unique_id: sg_20_winet_mppt2_voltage
      device_address: 1
      address: 5012 # reg 5013
[... 372 lines cut ...]
        {% elif ((states('sensor.sg_20_winet_power_limitation_switch_raw') | int(default=0))  == 0x0055) %}
          Disabled
        {% else %}
          Unknown - should not see me! {{states('sensor.sg_20_winet_power_limitation_switch_raw')}}
        {% endif %}

    - name: SG-20 Reactive power adjustment mode
      unique_id: sg_20_export_reactive_power_adjustment_mode
      availability: "{{ not is_state('sensor.sg_20_winet_reactive_power_adjustment_mode_raw', 'unavailable') }}"
      device_class: enum
      state: >-
        {% if ((states('sensor.sg_20_winet_reactive_power_adjustment_mode_raw') | int(default=0)) == 0x0055) %}
          Off, power factor returns to 1, reactive power percentage returns to 0
        {% elif ((states('sensor.sg_20_winet_reactive_power_adjustment_mode_raw') | int(default=0))  == 0x00A1) %}
          Power factor setting valid, Reactive power percentage returns to 0
        {% elif ((states('sensor.sg_20_winet_reactive_power_adjustment_mode_raw') | int(default=0))  == 0x00A2) %}
          Reactive power percentage Setting valid, power factor returns to 1
        {% elif ((states('sensor.sg_20_winet_reactive_power_adjustment_mode_raw') | int(default=0))  == 0x00A3) %}
          Enable Q(P) curve configuration
        {% elif ((states('sensor.sg_20_winet_reactive_power_adjustment_mode_raw') | int(default=0))  == 0x00A4) %}
          Enable Q(U) curve configuration
        {% else %}
          Unknown - should not see me! {{states('sensor.sg_20_winet_reactive_power_adjustment_mode')}}
        {% endif %}
`

NigelPearson, 2025-01-01:
John, thanks for posting that, however my SG30 does not support some of those addresses:

`./mbtget.pl  192.168.0.11 -u 1 -a 5145  -n 10`
values:
  1 (ad 05145):    90
  2 (ad 05146):   170
  3 (ad 05147): 65535
  4 (ad 05148): 65535
  5 (ad 05149): 65535
  6 (ad 05150): 65535
  7 (ad 05151): 65535
  8 (ad 05152): 65535
  9 (ad 05153): 65535
 10 (ad 05154): 65535

`% ./mbtget.pl  192.168.0.11 -u 1 -a 7012  -n 1`
Use of uninitialized value $rx_except_code in concatenation (.) or string at ./mbtget.pl line 334.
exception (code )

It seems this device only supports addresses 4949–5018, 5020–5207.
Missing 5019 is very odd.
It is possible that this device has encrypted firmware. See:

https://github.com/rpvelloso/Sungrow-Modbus

https://brettbeeson.com.au/hacking-the-sungrow-5-inverter

NigelPearson, 2025-01-05:
... and I was using the script wrongly, didn't specify the type of data read ...

`% ./mbtget.pl 192.168.0.11 -a 4999  -n 10 -r4`
values:
  1 (ad 04999):  9731
  2 (ad 05000):    30
  3 (ad 05001):     0
  4 (ad 05002):    17
  5 (ad 05003):   858
  6 (ad 05004):     0
  7 (ad 05005):   709
  8 (ad 05006):     0
  9 (ad 05007):   444
 10 (ad 05008):  1537

NigelPearson, 2025-01-05:
JohnFloki's examples set me on the right path.

I will _try_ to create some simplified modbus config, so that this SHx integration can also work on the SG series, but the non-modular structure of modbus config will make this difficult?
(all I/O to one IP address & port has to be in one YAML file)

closed 2025-01-05 by NigelPearson

Gnarfoz, 2025-01-05:
Yeah, it would have to be a separate file, meaning lots of code duplication.

JohnFLoki, 2025-01-06:
The modular problem is partly solvable by using 
`modbus: !include_dir_merge_list modbus`
in configuration.yaml, then make a folder "modbus" and put the yaml files there. I have one for SG, SH and the Wallbox.
The same is possible for templates, but I dont know about the automations as I manage them in the UI.

TCzerny, 2025-01-06:
I‘m using 
 # Sensors for the Inverter and Battery has been splitted and put in this folder
    sensors: !include_dir_merge_list ../modbus/modbus_sungrow_sensors/

For all the Modus Sensors and put them in seperate files for inverter and battery. 
Automation/templates are still in the original file