mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #424: Failed to connect [Errno 113] - too many  wrong reads?

liflitz, 2024-12-30:
This integration has worked for me perfectly well for over a year. Today I tried some now register reads, when suggenly HA became very slow and the integration no longer worked. At the end I decided to do a full HA restore. Afterwards all HA components worked well again BUT this integration (even other Modbus integrations worked):

2024-12-30 20:06:33.336 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('192.168.2.105', 502)
2024-12-30 20:06:35.881 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Error: device: 1 address: 4999 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.2.105:502]

Could it be that the inverter is blocking connections because of some possibly wrong register read attempts? Any help how to be able to connect again to my inverter would be highly appreciated.

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant version:**
Core 2024.12.5
Supervisor 2024.12.0
Operating System 14.1

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

Gnarfoz, 2024-12-30:
I'd suggest disconnecting the inverter from the network and reconnecting it after a little while. Same "cure" as for when trying to use multiple Modbus clients to access the inverter.

liflitz, 2024-12-30:
1 min was not sufficient, but 45 min did the job - thanks!!!

closed 2025-01-01 by liflitz