mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #425: Running State "Shutdown" after firmware update

Hoellenwesen, 2024-12-31:
**Describe the bug:**

I updated the Sungrow inverter to the latest firmware version and rebooted the whole PV system. Everything seemed fine, but after an hour or so (I noticed it the next day) the Running State for the inverter showed "Shutdown" in Home Assistant. I double checked in iSolarcloud and the parameter also showed "Shutdown". Then I restarted the system within iSolarcloud, but after a few minutes it showed "Shutdown" again.

I did a second PV restart, but the problem remained. Then I deleted this integration from Home Assistant and restarted the PV within iSolarCloud. So far it is running stable without any problems. So currently I have a running PV system, but no integration with Home Assistant. 

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [ X ] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [ X ] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2024.12.5

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-11-02
 - [ X ] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [ X ] I made sure that the newest firmware is installed via the installers account

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Enable this integration
2. Restart Home Assistant
3. Wait some time
4. Check the PV "Running State" in Home Assistant
5. Check Running State in iSolarCloud

**Additional context**
The previous firmware wokred just fine for myn months without issues. So my guess is, that something changed related to the latest firmware.

Gnarfoz, 2024-12-31:
Please see #416.

Hoellenwesen, 2025-01-01:
okay sorry, my bad. Did not see this issue. Hopefully there will be a solution. I will try to find something, too.

closed 2025-01-01 by Hoellenwesen