mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #426: How do I make "sg_forced_charge_battery_mode" actually start charging?

ulfben, 2025-01-01:
I created an automation to let me trigger the script sg_forced_charge_battery_mode manually. The script executes and the default mkaiser dashbord shows that the battery and inverter state has updated accordingly: 

[image]

However, the battery doesn´t start charging.

Are there settings I need to adjust on the isolarcloud / inverter-side of things?  
Anything else I might be missing? 

Any guidance would be much appreciated.

lassej01, 2025-01-02:
[image]
Have you changed this parameter? If Zero (as mine is) battery will force load with 0 kW

ulfben, 2025-01-02:
I haven't changed the slider, and the inverters local webpage (under Energy Management Parameters) shows 8kw max permitted charging power:
[image]

I _can_ charge from AC by using the iSolarCloud interface to "enable forced charging" during specific time slots. But it will then charge at ~9kw, not the 8kw I've set, which suggest some confusion about what register actually controls the allowed charging effect. 

I've just assumed the 0 you circled on the dashboard was a missread perhaps related to [the new firmware that's causing so much problems](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/406). 

Are you saying that "Set forced charge discharge..." affect some other register than what Sungrow exposes under `Energy Management Parameters -> Max. Permitted Charging Power from AC`?

lassej01, 2025-01-02:
Im not familiar with the adress ranges that is used. I just learned by trial and error, that if I set the value for  "Set forced charge discharge power in W" to something else than Zero, it started to force charge the battery with that amount of W. (in combination with EMS mode "Forced Mode" &  "Forced Charge"

ulfben, 2025-01-02:
I don't understand why these values are not reflected on the iSolarCloud side (or why I can't find them), but I believe you solved my problem! 

I set the sliders as I want them and ran the script again
[image]

My energy meter immedietaly shot up as the battery began charging: 
[image]

... and far too many minutes later, iSolarCloud also decided to update the dashboard to reflect reality. :) 
[image]

In the screenshot above you can see that I'm drawing >11kw from the grid. I naively assumed that "max charging power" was intended to protect the main fuse from overloading during charging. But instead the battery simply eats the full 8kw and Sungrow doesn't care that the the house needs 4kw for other things right then. 

Does anyone know if I can limit the total draw **from the grid** while charging? Eg. I want the battery to do load balancing, just like the car charger and heat pump does.

Gnarfoz, 2025-01-03:
That's the downside of forced mode, it disables all the smart stuff that "Self-consumption mode" does. Doing both would fall into "external EMS" territory (meaning continuous real-time management of parameters, replacing the inverter's built-in logic), which this project is not.

The "Set forced charge / discharge power" slider controls register 13052:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L1531-L1545
You don't need to set "max battery charging power" (33047) at the same time, it's not relevant to "forced mode" operation.

"Max permitted charging power from AC" is the "protection" parameter you thought "max charging power" to be:
[image]

There are multiple scenarios in which the battery might be charged from the grid. Emergency charging to prevent damage, SoC going below the backup SoC threshold, SoC going below the minimum SoC threshold, SoC calibration, forced mode... not all of these use configurable power, so the "max permitted from AC" parameter should serve as a limit to hopefully all of them.

ulfben, 2025-01-04:
> That's the downside of forced mode, it disables all the smart stuff that "Self-consumption mode" does. Doing both would fall into "external EMS" territory (meaning continuous real-time management of parameters, replacing the inverter's built-in logic), which this project is not.

I am not sure if understand your meaning. Are you saying that Sungrow does not provide a way to force charge the battery without overloading the main fuse? 

> The "Set forced charge / discharge power" slider controls register [...] You don't need to set "max battery charging power" (33047) at the same time, it's not relevant to "forced mode" operation.

"set forced charge / disharge power" does indeed let me limit the charging, but not the total draw from the grid, which is what I need to fix. The battery routinely trips my main fuse if, for example, we run the dishwasher while it's charging.

What is "Max Battery Charging Power" used for? 

> "Max permitted charging power from AC" is the "protection" parameter you thought "max charging power" to be [...] the "max permitted from AC" parameter should serve as a limit to hopefully all of them.

It didn't seem to limit charging speed at all for me. :/

ulfben, 2025-01-06:
Thanks for your input! I'll keep experimenting and keeping a closer eye on the numbers. We haven't blown a fuse in a while now so maybe, maybe it is in fact working and I'm just confused. :)

closed 2025-01-17 by ulfben