mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #428: Question - Battery Forced Charged Limited by Load Power

ProtekDev, 2025-01-02:
I just have noticed, that there must be a limited factor for the forged charging.
Maybe around 20kW.

If i get over 20kW my Battery doesnt even charge.
When i drop under that value i get the difference to that "invisible" limit.

Or am i missing something ?
I just checked everything from WiNet-S Webpage.

[image]

Gnarfoz, 2025-01-03:
I'm not sure what you're asking. How does load power relate to forced mode battery charging power?
As can be seen in your screenshot, you've set "forced / charge discharge power" to 5 kW, so I don't see why it would charge faster than 5 kW in the first place, let alone 20 kW.
Do you have a battery and inverter that can charge at 20 kW?

Regardless of that, there's this parameter which I don't recall changing, so its default might be 20 kW:
[image]

And there's this other one one tab further:
[image]

Gnarfoz, 2025-01-03:
> Do you have a battery and inverter that can charge at 20 kW?

Just checked for the SHxRT series: https://info-support.sungrowpower.com/application/pdf/2024/01/17/DS_20240111_SH5.0_6.0_8.0_10RT_Datasheet_V22_EN.pdf
It caps out at 10.6 kW.
The SHxT series caps out at 30 kW: https://info-support.sungrowpower.com/application/pdf/2024/09/13/DS_20231201_SH15_20_25T_Datasheet_V3_EN.pdf

ProtekDev, 2025-01-04:
Thanks for your reply.
[image]
I've set forced charging to 4kW, but my battery supports a maximum of 6.57kW charge/discharge.
I think this might be evidence that my initial guess was correct.

[image]
As you can see from the attached image, my charging power drops significantly during the day due to solar power. However, when I enable forced charging (indicated by the blue bar), the charging power seems to adjust automatically and sometimes doesn't even reach 4kW. 

Interestingly, it appears that there's a threshold around 20kW of import/power load that causes the battery charge to degrade regardless of the charging power. This phenomenon puzzles me, as I couldn't find any option to disable or bypass this issue.

Gnarfoz, 2025-01-06:
I highlighted two parameters that might be limiting the charging power ("Site import power limit" and "Max. permitted charging power from AC")  have you checked those?

I'm more confused by the fact that "import power" at the end of your graph, after battery charging power has returned to 0, seems to indicate a power draw of ~16-ish kW? Where is that power going?!

ProtekDev, 2025-01-06:
Yes, I tried to look for those parameters. Unfortunately, I couldn't find 'Site Import Power Limit'. And yes, the power draws are a bit confusing. However, it's likely due to the night storage heaters that can turn on randomly.
[image]
And i tried to adjust some of these variables, but not too heavily.

Gnarfoz, 2025-01-06:
That looks like the WiNet-S web interface. The parameters might be named differently there compared to iSolarCloud.

You already found "Max. permitted charging power from AC", as your screenshot shows.
There's also "Import Power Limit" under "Operation Parameters" which has the same 0.01 - 50 kW range as "Site Import Power Limit" in iSolarCloud.