mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #442: Integration of EV charger for switching between charge-modes

huberda, 2025-01-22:
First of all, I love your HM integration! 
I`m a little bit paranoid and therefore would like to restrict my Sungrow devices from connecting to the internet - for me the current HM integration delivers everything I need except:
Switching the register 33540/41 on the inverter to change between the different charging modes ( fast charging, green power,....).
I would love to see this feature and currently I am a bit afraid of doing it on my own without bigger knowledge in modbus.

It seems that Sungrow already delivered some information about the register
https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/issues/13#issuecomment-2105990200 

best regards,
huberda

Gnarfoz, 2025-01-22:
The thread you mention already includes this link to a guide on how to set this up. Does that work for you?
https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger

huberda, 2025-01-22:
many thanks for your response - i tought realizing it in the same integration would be useful. I try to avoid having many different integrations/addons for the same use-case. I started the installtion of the mentioned solution and it seems this adressed my issue :-)

Gnarfoz, 2025-01-22:
Great to hear it works!

This particular project isn't even an actual "integration" in the Home Assistant jargon -- it's "just" a configuration file for the Modbus integration provided by Home Assistant itself out of the box. There isn't really anything to "save" by not using it. =)

albertofralbe, 2025-01-25:
Here it's how I've done it:

      - name: Raw Status Caricatore EV
        unique_id: ev_charger_status_raw
        slave: !secret charger_modbus_slave
        address: 21316 # reg 21317
        input_type: input
        data_type: uint16
        scan_interval: 10

 # holding registers

      - name: Raw Modalità Caricatore EV
        unique_id: ev_charger_mode_raw
        slave: !secret inverter_modbus_slave
        address: 33540 # reg 33541
        input_type: holding
        data_type: uint16
        scan_interval: 10

    switches:
      - name: Ricarica EV
        slave: !secret inverter_modbus_slave
        address: 33539
        command_on: 170
        command_off: 85
        write_type: holding
        verify:

      - name: Ricarica Solare+Grid
        slave: !secret inverter_modbus_slave
        address: 33549
        command_on: 170
        command_off: 85
        write_type: holding
        verify:

template:
    sensor:
      - name: Status Caricatore EV
        unique_id: ev_charger_status
        availability:
          "{{states('sensor.raw_status_caricatore_ev')|is_number}}"
        state: >-
          {% set mode = states('sensor.raw_status_caricatore_ev') %}
          {% set mapper = {
            '1': 'Scollegato',
            '2': 'In attesa di ricarica',
            '3': 'Ricarica in corso',
            '4': 'Ricarica sospesa (Caricatore)',
            '5': 'Ricarica sospesa (Auto)',
            '6': 'Ricarica completata',
            '7': 'Riservato',
            '8': 'Disabilitato',
            '9': 'Errore',
          }%}
          {{ mapper.get(mode) }}

    select:
      - name: "Modalità Caricatore EV"
        unique_id: ev_charger_mode
        state: >
          {% set mode = states('sensor.raw_modalita_caricatore_ev') %}
          {% set mapper = {
            '160': 'Ricarica Rapida',
            '161': 'Ricarica Solare',
            '162': 'Ricarica entro orario stabilito',
            '163': 'Ricarica in intervallo stabilito',
          }%}
          {{ mapper.get(mode) }}
        options: '{{["Ricarica Rapida", "Ricarica Solare", "Ricarica entro orario stabilito", "Ricarica in intervallo stabilito"]}}'
        select_option:
          - service: modbus.write_register
            data:
              hub: sungrow
              slave: !secret inverter_modbus_slave
              address: 33540  # reg 33541
              value: >
                {% set mapper = {
                  'Ricarica Rapida' : '160',
                  'Ricarica Solare' : '161',
                  'Ricarica entro orario stabilito' : '162',
                  'Ricarica in intervallo stabilito' : '163',
                }%}
                {{ mapper.get(option) }}
        availability: "{{ states('sensor.raw_modalita_caricatore_ev') is defined }}"

Some parts are in Italian so feel free to ask if you need anything
Note: You need to set the slave addresses for inverter_modbus_slave and charger_modbus_slave in the secret.yaml file

closed 2025-02-17 by huberda