mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #444: SH10RT no standby mode after latest firmware update

mkaiser, 2025-01-24:
After updating the latest firmware in January 9th, the inverter does not go into standby mode any more: 

[image]

**Is anybody else experiencing this?**

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [x] yes
  - [] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2025.01]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2025-01-09] (see header of the file)
 - [x] I ensured to use the most recent version

**Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

Gnarfoz, 2025-01-26:
Is that 95.03 or something even newer?

mkaiser, 2025-01-26:
Yes, [image]

Maggie70, 2025-01-26:
Since I updated the inverter and battery at the beginning of December to the latest firmware, the inverter has not gone into standby anymore.

timpetersson81, 2025-01-27:
Sometimes my inverter could not find its way back to running mode, so I had to shut the power to reset it. Perhaps its a general issue, and that's why.

Gnarfoz, 2025-01-27:
@mkaiser What do the following parameters look like for you?
- backup mode
- backup SoC
- min SoC

There's at least two threads on this topic at German PV forum photovoltaikforum.com:
https://www.photovoltaikforum.com/thread/200085-sungrow-wechselrichter-geht-nicht-mehr-in-stand-by/
https://www.photovoltaikforum.com/thread/238999-sungrow-sh10rt-geht-nicht-ins-standby/

I guess I'm happy I'm still on 95.01...

mkaiser, 2025-01-28:
Thanks for the links! That was quiet a read... but in the end there are many assumptions and it is most likely sungrows fault. 

I still hate the isolarcloud and the app and do not want to ever use it again... So here is the newest feature for controlling the off-grid option:

[image]

Will commit, soon :)

After 10 Minutes without any power (battery of, AC off), the inverter went to "initial standby". This morning it woke up and after a sunny day the battery is almost loaded. Tonight I can say more :)

Gnarfoz, 2025-01-28:
Is that holding register 13075?
I think there's been an old feature request to expose that for a long time anyway, nice. 😁
Although I find the name Sungrow uses for that register is a bit confusing. It's for grid power outages, not for island/off-grid operation (which, according to Sungrow, is not supported long-term, for reasons I don't quite understand), so "backup mode" would probably more appropriate?

After entering "initial standby", did you turned battery back on right away? Or only after the the inverter had started on solar power only?

mkaiser, 2025-01-28:
> Is that holding register 13075? I think there's been an old feature request to expose that for a long time anyway, nice. 😁

yep :)

 Although I find the name Sungrow uses for that register is a bit confusing. It's for grid power outages, not for island/off-grid operation (which, according to Sungrow, is not supported long-term, for reasons I don't quite understand), so "backup mode" would probably more appropriate?

-> yes, "Backup mode" is much better. Thanks for the feedback :)

> After entering "initial standby", did you turned battery back on right away? Or only after the the inverter had started on solar power only?

I switched the battery 10s on after enabling the AC power. 

it is too hot outside (8°c).... battery is still not empty ;) Have to wait another hour or so, until I know if it will go to sleep

Maggie70, 2025-01-29:
Nice work, I'm so grateful for the effort and time you put into making everything work.

mkaiser, 2025-01-29:
The inverter still does not want to to into standby. 

Setting the inverter status manually to "Shutdown" and back to "Enabled" after 10seconds results in a proper standby...

So I will create an automation for this Sungrow-firmware-quirk...

Will post it, as soon I could test it :)

Gnarfoz, 2025-01-29:
Almost sounds like the reverse of the issue where the automation was reacting to the "stop" state that now happens during startup, shutting down the the inverter... Did you check if those are triggered at all when it's normally time to go into standby mode?

mkaiser, 2025-01-29:
No sungrow automations were triggered (except one unrelated, which controls a fan based on the inverters temperature...)

Maggie70, 2025-01-30:
[image]
If the inverter is in standby, it does not enter forced charge mode via automation.

Gnarfoz, 2025-01-30:
Your automation would need to first turn inverter on.
Terrible car analogy: you don't expect the car to move when you press the gas pedal without first turning on the motor, go you? 🤔 😅

Maggie70, 2025-01-30:
I use an automation that has worked flawlessly for a long time, but since the latest firmware on the inverter and the latest program from Mkaiser it has become like this. and I know that he is working on a solution and the post was intended as information.

Gnarfoz, 2025-01-30:
Well, "It has worked for a long time, then multiple things changed, now it doesn't work anymore" is pretty much how things go. 😉
Especially with Sungrow's firmware updates that don't provide any insight into what has actually changed with each version... 😟

On the other hand, the changes to _this_ project are very transparent and it seems unlikely that the root cause can be found here. Even the main author is affected by this issue.

One thing I noticed: your screenshot seems to show "inverter state" changing back and forth a lot from "standby" to... _something_, between ~01:30 and ~07:00. Could you check what the other state is?

Maggie70, 2025-01-30:
I think I discovered my problem, when I rebooted the inverter last night, the forced charge power input slider was reset to 0.

Maggie70, 2025-01-30:
The status changed between standby and startup

mkaiser, 2025-02-07:
I used this automation for a workaround now for one week. Standby now works (manually) again. 

... it is really sad that going to standby mode is broken in sungrow's current firmware

```alias: Sungrow_go_to_standby_quirk
description: ""
triggers:
  - trigger: time_pattern
    minutes: /10
conditions:
  - condition: state
    entity_id: sensor.sungrow_inverter_state
    state: Running
    for:
      hours: 1
      minutes: 0
      seconds: 0
  - condition: state
    entity_id: binary_sensor.battery_discharging_delay
    state: "off"
    for:
      hours: 0
      minutes: 10
      seconds: 0
  - condition: state
    entity_id: binary_sensor.pv_generating_delay
    state: "off"
    for:
      hours: 0
      minutes: 10
      seconds: 0
  - condition: numeric_state
    entity_id: sensor.mppt1_voltage
    below: 10
  - condition: numeric_state
    entity_id: sensor.mppt2_voltage
    below: 10
  - condition: state
    entity_id: sensor.ems_mode_selection
    state: Self-consumption mode (default)
  - condition: state
    entity_id: sun.sun
    state: below_horizon
  - condition: state
    entity_id: sensor.backup_mode
    state: Disabled
actions:
  - action: input_select.select_option
    metadata: {}
    data:
      option: Shutdown
    target:
      entity_id: input_select.set_sg_inverter_run_mode
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
  - action: input_select.select_option
    metadata: {}
    data:
      option: Enabled
    target:
      entity_id: input_select.set_sg_inverter_run_mode
mode: single
```

For testing adding this at the bottom was very helpful: 

```
  - action: notify.mobile_app_smartphone_martin
    metadata: {}
    data:
      message: Triggered new automation
      title: Sungrow inverter
```

Will put this as an automation example in the git, soon and close this issue as "unresolved, but worked-around"

Gnarfoz, 2025-10-07:
"Current" being which version exactly? 95.10?

mkaiser, 2025-10-30:
yes, this is the correct behavior. Going to standby at night, when the battery is empty :)

Sungrow fixed this :)

mkaiser, 2025-10-30:
okay, now I am getting your point. 

This is really strange. I expect the inverter to go out of standby if "something forced" should be done. 

I guess this is again some Sungrow issue and maybe worth a sungrow support request.

mkaiser, 2025-10-30:
Just fu***d up the inverter again (twice today with a completely broken LAN connection requireing a cold inverter reboot...) while playing with this:

```
      - name: Forced startup under low soc standby raw
        unique_id: sg_forced_startup_under_low_soc_standby_raw
        device_address: !secret sungrow_modbus_slave
        address: 13016 # reg 13017
        input_type: holding
        data_type: uint16
        state_class: measurement
        scan_interval: 10

```

I am not sure if the reason is my frequent restarting of HA, or something is really happending with this register 

more related code / the automations for testing : 

```
      # make the sensor load_adjustment_mode_enable_raw more human readable
      - name: Forced startup under low soc standby
        unique_id: forced_startup_under_low_soc_standby
        availability: "{{ not is_state('sensor.forced_startup_under_low_soc_standby_raw', 'unavailable') }}"
        device_class: enum
        state: >-
          {% if ((states('sensor.forced_startup_under_low_soc_standby_raw') |int) == 0xAA) %}
            ON
          {% else %}
            OFF
          {% endif %}
``` 

```
  # get input for forced startup under low soc standby
  set_sg_forced_startup_under_low_soc_standby:
    name: Forced startup under low SoC standby
    options:
      - "ON"
      - "OFF"
  ```

```
 - id: "automation_sungrow_inverter_forced_startup_under_low_soc_standby"
    alias: "sungrow inverter forced startup under low SOC standby"
    description: "Updates forced startup under low SOC standby and corresponding gui input"
    variables:
      forced_startup_under_low_soc_standby_on: 0xAA
      forced_startup_under_low_soc_standby_off: 0x00
    triggers:
      - platform: state
        entity_id:
          - input_select.set_sg_forced_startup_under_low_soc_standby
        id: tid_ui_forced_startup_under_low_soc_standby
      - platform: state
        entity_id:
          - sensor.forced_startup_under_low_soc_standby
        id: tid_sensor_forced_startup_under_low_soc_standby
    conditions: []
    action:
      - choose:
          - conditions:
              - condition: trigger
                id: tid_ui_forced_startup_under_low_soc_standby
            sequence:
              - action: modbus.write_register
                data_template:
                  hub: *sg_hub_name
                  slave: !secret sungrow_modbus_slave
                  address: 13010 # reg 13011
                  value: >
                    {% if is_state('input_select.set_sg_forced_startup_under_low_soc_standby', "ON") %}
                      {{forced_startup_under_low_soc_standby_on}}
                    {% else %}
                      {{forced_startup_under_low_soc_standby_off}}
                    {% endif %}
              - delay: "00:00:01" # delay one second
              - action: homeassistant.update_entity # force update
                data:
                  entity_id:
                    - sensor.forced_startup_under_low_soc_standby_raw
          - conditions:
              - condition: trigger
                id: tid_sensor_forced_startup_under_low_soc_standby
              - condition: template
                value_template: "{{ not is_state('sensor.forced_startup_under_low_soc_standby', 'unavailable') }}"
            sequence:
              - action: input_select.select_option
                target:
                  entity_id: input_select.set_sg_forced_startup_under_low_soc_standby
                data:
                  option: >
                    {% if states('sensor.forced_startup_under_low_soc_standby_raw') | int == forced_startup_under_low_soc_standby_on %}
                      ON
                    {% else %}
                      OFF
                    {% endif %}
    mode: restart # use restart to avoid warnings, when GUI is updated too often and the delay causes this automation to not finish before the next call

```

mtrekker, 2025-11-04:
Need more testing because mine nor #d165ac5 automation didn't propagate properly. Took the commit #[d165ac5](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/d165ac54603ad1cd5162c62179c8a5b9c990670a/modbus_sungrow.yaml#L3167) and it was having side effects.

There are 2 entities linked through one automation:

1. A sensor (sensor.forced_startup_under_low_soc_mode_raw) — reads the real inverter Modbus register.
2. A switch (switch.forced_startup_under_low_soc_standby) — represents the UI control for that inverter mode.

There are 2 triggers:

1. When the sensor changes → update the switch.
2. When the switch changes → write to Modbus and then update the sensor.

Side effects:

- Duplicate Modbus writes every time the sensor refreshed.
- Sensor updates retriggering switch updates, even when nothing actually changed.

Proposal when one side (UI or sensor) is being actively synced:

- the automation sets input_boolean.force_soc_sync_lock = on.
- The other side ignores updates while the lock is active.
- Once the change settles, the lock is turned off again.

[automation_sungrow_inverter_forced_startup_under_low_soc_standby.yaml](https://github.com/user-attachments/files/23326908/automation_sungrow_inverter_forced_startup_under_low_soc_standby.yaml)

mtrekker, 2025-11-07:
3 nights of testing and here is the outcome:

**Forced Mode (Compulsory) and Standby Mode Operation**

- When the inverter is in Standby mode, and the Forced Standby switch (13017 -> 0xAA) is activated, a wait template must occur before enabling Forced Mode (Compulsory). It took 1 min 52 seconds to start up. [1]
- If the inverter is in Standby mode and both the Forced Standby switch (13017 -> 0xAA) and Forced Mode (Compulsory) are enabled sequentially, the inverter will attempt to start up three times and then revert to Standby mode. [1]
- Forced Mode (Compulsory) can only be enabled when the inverter is not in Standby mode. [2]

[1] [image]

[2] [image]

**Simplicity over complexity**

- Use switch (13017 -> 0xAA / 0x55) instead of complex automation
```
    sensors:
      - name: Forced Startup Under Low SoC Standby raw
        unique_id: sg_forced_startup_under_low_soc_standby_raw
        device_address: !secret sungrow_modbus_slave
        address: 13016 # reg 13017
        input_type: holding
        data_type: uint16
        precision: 0
        state_class: measurement
        scan_interval: 10

    switches:
      - name: Forced Startup Under Low SoC Standby
        unique_id: sg_forced_startup_under_low_soc_standby
        slave: !secret sungrow_modbus_slave
        address: 13016 # reg 13017
        command_on: 0xAA
        command_off: 0x55
        verify:
          input_type: holding
          address: 13016 # reg 13017
          state_on: 0xAA
          state_off: 0x55
          delay: 2
```

mkaiser, 2025-11-07:
Hi, 

wow, I did not know about the modbus switches: This is really cool and massivly simplifies things. Will integrate this :)

My brain is a bit fried at this moment and I am not sure if I parsed your findings correctly. 

So would the solution be a script, to "wake" the inverter first, wait some seconds and then enable the forced mode?

TCzerny, 2025-11-07:
No should it not be the other way around? 
First enable it, wait, then start. 

' By setting the "Forced Startup
Under Low SOC Standby" configuration point to "0xAA", the inverter can be manually initiated without waiting for adequate PV energy.'

BTW: works also with other Entity types. Numbers, select, etc. So many UI update automations can be avoided.

mkaiser, 2025-11-08:
I can only find some doc for Sensors, binary Sensors and switches in https://www.home-assistant.io/integrations/modbus/#example-full-tcp-configuration

Where are numbers and select located?

mkaiser, 2025-11-09:
> https://www.home-assistant.io/integrations/template/#number
> 
> https://www.home-assistant.io/integrations/template/#select

those are general Home assistant templates, which cannot be used within the modbus section. 

Supported modbus entities are

    binary_sensors:
    climates:
    covers:
    fans:
    lights:
    sensors:
    switches:

I will port the current code to switches where it makes sense.

mtrekker, 2025-11-10:
> So would the solution be a script, to "wake" the inverter first, wait some seconds and then enable the forced mode?

Yes, this is what I have done currently to resolve the latest firmware "glitch". Latest firmware caused major change in behavior when introducing Standby mode and ignoring Forced Mode at the same time.

polojoe, 2025-11-18:
If you activate backup mode and set backup soc to eg 5 %, inverter would not go to standby. Maybe a workaround.

mtrekker, 2025-11-19:
You need to wake up the inverter when in standby mode. This can be done using the new register:

forced_startup_under_low_soc_standby_raw (13017)

1. Enable forced startup-under-low-SOC if needed:
1.1 `states('sensor.forced_startup_under_low_soc_standby_raw') | int == 0x55`
1.2 `states('sensor.system_state') | int in [0x1400, 0x0008, 0x1600, 0x0020] # Standby or Startup`
2. Wait for the system to leave startup/standby mode
3. Set the inverter/battery into “forced charge” mode
4. Set the battery charge power
5. Limit maximum battery discharge power (if needed, usually you don't want to drain battery while charging)

Gnarfoz, 2025-11-20:
> That's not how it should work when in forced mode with force command being force charge.
> 
> What would be the point of that? 🤔

Just to confirm: during forced charging, there is no discharging. I was producing 7.8 kW earlier, turned on forced charging with a charge limit of 6 kW (which all went into the battery) and then turned on a 2.2 kW water kettle. As expected, the difference was covered from the grid, not from the battery:

[image]