mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #454: Readout DTSU666-20 with 2nd converter set for third-party hardware via SHT10RT

Bakari2020, 2025-02-07:
**Is your feature request related to a problem? Please describe.**
The sensor sensor.daily_pv_generation does not display a value for the added power / generated current of the 2nd transformer set. 
The values are also not available via the local Wi-Net stick by reading out real-time values.
However, the values are output correctly in the Isolarcloud for daily generation power.
I therefore do not see the 2nd value via the Modbus of the Sungrow,
the 2nd inverter does not have Modbus, only RS485, which I would like to avoid tapping into.

**Describe the solution you'd like**
Is there a solution to read the DTSU66-20 with 3rd party converter set via the inverter Modbus?

**Describe alternatives you've considered**
Integration der Gesamten Einspeisung des 3. party Inverters Gesamt erzeugung / aktuelle Leistung

Gnarfoz, 2025-02-07:
Your secondary inverter is a non-Sungrow device?
Modbus RTU is often carried via RS485, so... it technically might have Modbus?

You could use a Riemann sum helper sensor to integrate the energy values from the "Meter active power" sensor. "Meter" in this case refers to the DTSU666.

Or you could try this project to access the DTSU666 directly: https://github.com/RafAustralia/Chint-DTSU666-20-modbus/
It has a handful of sensors that measure in kWh, but I'm not sure what exactly the values refer to.
Also, you would need to figure out which Modbus slave ID the DTSU666 uses. It's not 5 like in that repository.

joshbohmer, 2025-02-26:
So I did some digging around my SH10RS and found register 5757 ( uint16, w, 1.0 scale ) holds the reading from the 2nd DTUS666 channel used for the 2nd inverter output. I would assume that's P1 and 5759 = P2 & 5761 = P3.

With that reading you can calculate house loads for the system as follows...

5757 + Phase A power ( this is calculated by Mkaiser ) - Export power = total house load including 2nd inverter.

Downside being if you proxy through WiNet these registers are unavailable. To access them you'll need to add your own Modbus TCP gateway. For inverters with a dedicated ethernet port I expect them to be fine.

Bakari2020, 2025-02-26:
I have a SHT10RT and only get the following with the following configuration. I have integrated the code from below into the sungrow yaml.

[image]

     # Thirdparty Inverter #
      - name: Second Inverter phase A power
        unique_id: sg_second_phase_a_power
        device_address: !secret sungrow_modbus_slave
        address: 5757 # reg 5757
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

      - name: Second Inverter phase B power
        unique_id: sg_second_phase_b_power
        device_address: !secret sungrow_modbus_slave
        address: 5759 # reg 5759
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

      - name: Second Inverter phase C power
        unique_id: sg_second_phase_c_power
        device_address: !secret sungrow_modbus_slave
        address: 5761 # reg 5761
        input_type: input
        data_type: uint16
        precision: 1
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

joshbohmer, 2025-02-26:
Do you have WiNet by chance to check? I suspect that address may be unavailable via ethernet also then. This is what I see, its early morning here so my 2nd inverter is only putting out 0x00BD ( 189w )... but i've checked it matches at various loads before.

[image]

joshbohmer, 2025-02-26:
actually... i just checked your code. You need to specify 1 before the register in HA Modbus as below.

` 
    address: 5756 # reg 5757

    address: 5758 # reg 5759

    address: 5760 # reg 5761`

Bakari2020, 2025-02-26:
thanks for that quick answer I tryed it and changed that address to your comments.
also this works:

[image]

but the sensor is not available actual.

I will try tomorrow because here is also night.

Bakari2020, 2025-02-26:
I only got Pymodbus: SungrowSHx: Error: device: 1 address: 5758 -> pymodbus returned isError True