mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #455: Help Support for HomeAssistant Version 2025.XX

W3isserWolf, 2025-02-07:
Hello,
Does someone know What i have to configure to make this Script working for HomeAssistant version 2025.XX like 2025.2.1?
I want to Upgrade to newer versions but also want to use this feature.

Thanks

mkaiser, 2025-02-07:
Hi, 

I am on 2025.2.0 and it still works as before :)

Can you give us some more information? Are you in stuck the installation process or did any problems occur while updating (from version ??? to 2025.*)?

Gnarfoz, 2025-02-07:
Unless you made changes to the `modbus_sungrow.yaml` file in the past, it should be enough to just replace the file.
If you're not sure, make a backup of the file before. Don't put the backup in the same directory, though - it would cause a conflict.

GobblerGuy, 2025-02-10:
I had a very old version from a couple of years ago. I wanted to upgrade to get EMS options etc.. and just overwriting the old modbus_sungrow.yaml file and it is not working. It sometimes sticks at the check configuration step for a minute before saying it will not cause any issues. Then when I click to reload all YAML it takes a long time (minutes) and sometimes crashes during the process. Clicking refresh Modbus alone hangs as well. Even if it does it 'successfully' reload the yaml many of the sensors remain unavailable. 
I am wondering if my raspberry PI 3B+ is just too underpowered.
I might try cutting out some scripts or setting register refresh to more than 10s for many of the registers. Think the latter will help?

core: 2025.1.4
Supervisor 2025.02.0
OS 14.1
frontend: 20250109.2

Gnarfoz, 2025-02-10:
Your Home Assistant is new enough to have the fix that should allow the Modbus configuration to be reloaded at runtime (https://github.com/home-assistant/core/commit/502fbe65eefa50e86b2e1deea60354601b7449c2).
However, I'd suggest trying to restart Home Assistant instead to pick up the changes to `modbus_sungrow.yaml`. Maybe that is already enough.

W3isserWolf, 2025-02-10:
Hello, and thanks for the Answers.
Im on Version 2024.12.5 but the Script was old. Then i updated it: working as expected
Upgrade HAOS to 2025.2.1 with the newest Script: did not worked
Recovered HomeAssistant(2024.12.5) from an Backup: working again.
I will try it again and will take a Look inside the Logs.

GobblerGuy, 2025-02-11:
> Your Home Assistant is new enough to have the fix that should allow the Modbus configuration to be reloaded at runtime ([home-assistant/core@502fbe6](https://github.com/home-assistant/core/commit/502fbe65eefa50e86b2e1deea60354601b7449c2)). However, I'd suggest trying to restart Home Assistant instead to pick up the changes to `modbus_sungrow.yaml`. Maybe that is already enough.

Retried to update the modbus_sungrow.yaml page again. This time just copied the modbus entries. It took a couple of tries but got the YAML to reload and all the modbus sensors. Seemed OK I guess but having to do a few times to get all sensors available made me wonder if was 100% alright.. Tried pasting the rest of the modbus_sungrow.yaml page and I could not get it to work, either by reloading YAML or restart of the whole unit a few times. tried deleting the most recent paste from modbus_sungrow.yaml and I could not get it to work again via YAML reload or restart. the PI seemed very laggy and unhappy. 

I was planning up upgrade the PI, so i'll just do that before I try again, unless anyone else makes a breakthrough. 

rolled back to prior old version of modbus_sungrow.yaml via restore of backup and all is working again.

W3isserWolf, 2025-02-11:
Okay, i have copied the .yaml from git again and now everything seems to be working. 
Thanks

closed 2025-02-11 by W3isserWolf

Gnarfoz, 2025-02-11:
Glad to hear that it worked out in the end! 🥳