mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #458: Three entities are no longer displayed in the standard dashboard

DienMan, 2025-02-10:
First of all, thank you very much for developing and providing this integration!
I happened to notice today that three entities 

[image]

are no longer displayed. Probably not for a while...

I have a default installation SH10RT+SBR096 and the unchanged latest versions of “modbus_sungrow.yaml” and the default dashboard “dashboard.yaml”.

I also use the latest versions of Home Assistant:

- Core: 2025.2.1
- Supervisor: 2025.02.0
- Operating System: 14.2 

Error message for all entities:
“This entity is no longer provided by the “template” integration. If the entity is no longer used, delete it in the settings.”

**sensor.daily_consumed_energy**
Last entry in the long-term statistics: November 3, 2024

**sensor.daily_consumed_energy_filtered**
No logbook entries

**sensor.total_consumed_energy**
Last entry in the long-term statistics: November 3, 2024

Do you have any advice on where I can look for the error?

Thanks 
Chris

DienMan, 2025-02-10:
Fantastic, in fact two of the entities were renamed with “_2”. For whatever reason. 
Thank you very much for the very quick help!

“Daily consumed energ (filtered)” has not been renamed and still shows ‘unknown’. Or is this correct in this setup?

Gnarfoz, 2025-02-10:
Does that one not show up in the search results where you found the other two? 🤔

DienMan, 2025-02-10:
Yes, but I have also just seen that this appears as a filter and not as a template. If nothing is filtered, the status may be correct with “Unknown”.

Gnarfoz, 2025-02-10:
No, that just means it's provided by the "Filter" integration: https://www.home-assistant.io/integrations/filter/
It's a moving average of the PV power: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L1684-L1699

DienMan, 2025-02-10:
Thanks for the hint, I have added the suffix "_2" to sensor.daily_consumed_energy in line 1690 of modbus_sungrow.yaml. Now the value is displayed correctly.

Gnarfoz, 2025-02-10:
Hah, I totally missed that, good catch!

closed 2025-02-11 by DienMan