mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #459: Meter voltage and Current not available

radi334, 2025-02-11:
**Describe the bug:**

The entities Meter Voltage and Current A B and C are shown as Not available. Also I can see in the error log the following messages repeating every 10 seconds

> 
> 2025-02-11 10:13:16.093 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
> 2025-02-11 10:13:16.094 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Error: device: 1 address: 5740 -> pymodbus returned isError True
> 2025-02-11 10:13:16.099 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
> 

These are flodding the logs considerably.

**Your Sungrow inverter:**
 - Model: SH10RT-V112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2025.02.2]

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2025-02-07
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Modbus Device entities and see the values of the different entities mentioned above.
2. Check the logs under Settings -> System -> Logs

**Expected behavior**
The Voltage and Current value for each phase should be displayed correctly and error messages should be cleared accordingly

Gnarfoz, 2025-02-12:
Just making sure: you do have a DTSU666 installed & connected to your inverter? Has this ever worked before?

radi334, 2025-02-12:
maybe you can educate me the URL to the internal web interface, I'm only aware about the modbus port. However iSolarCloud also shows the values, so I assume this is correctly configured on the device itself:

[image]

Gnarfoz, 2025-02-12:
You skipped over the question if you actually have a DTSU666.

The URL is simply https://<the IP address of your WiNet-S dongle>. You should be able to find it in your router's management interface or with a network scanner like https://www.advanced-ip-scanner.com/.
Its manufacturer (or rather, the manufacturer of the network chip) should be listed as _Espressif Inc._.
If you tell it to scan for HTTPS / port 443, that should show up as well.
Default credentials are `admin / pw8888` or `user / pw1111`

The phase A, B, C values in iSolarCloud are probably the inverter's internal measurements. It also offers some meter values, but not the voltages. WiNet-S clearly gathers both, and shows both:

[image]

radi334, 2025-02-13:
Yes I have a DTSU666 installed and these are the values out of the WiNet-S web interface:

[image]

Gnarfoz, 2025-02-14:
Out of good ideas here, maybe doing a full power cycle (set "inverter mode" to "shutdown" via Home Assistant, or do it via WiNet-S or iSolarCloud; turn off AC breaker, turn off DC breaker (side of the inverter); turn off battery; wait 10 minutes; turn on AC breaker; turn on battery (wait 5 minutes); turn on DC breaker) will help.

Gnarfoz, 2025-02-22:
Suspiciously, those all fall in the range for historical data (monthly, yearly), [which are not available via WiNet-S](https://github.com/Gnarfoz/Sungrow-Inverter/blob/main/Modbus%20Information/TI_20240924_Communication%20Protocol%20of%20Residential%20Hybrid%20Inverter-V1.1.5.pdf).
However, you checked the box that you're using the inverter's internal port, so... I don't know. Did you maybe connect a cable to that but you're accidentally using the WiNet-S IP address instead? **The IP address in your `secrets.yaml` should not be the same one you used to access the WiNet-S web interface a few posts ago.**

[image]

radi334, 2025-02-24:
Ok, this was the issue indeed. It took a while, to update. Thanks for having me check this again. I now put back the values and they are fetched properly. Altough the Yearly statistics values are to high, but this seems to be a known issue. Waiting for a firmware update to fix it.

closed 2025-02-24 by radi334