mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #461: Missing values

jc82517, 2025-02-12:
Is your feature request related to a problem? Please describe.

I need the following two values to create a live representation similar to the iSolar app in Home Assistant using the Power Flow Card Plus. Currently, I don’t have a value that indicates whether the battery is charging or discharging. "Battery Power Raw" does not differentiate between them. This worked in an earlier version of the YAML, but unfortunately, I can’t find it anymore.

Describe the solution you'd like

In the WiNet interface of my SH10RT-20(COM1-001), both values are displayed under "Battery Information" as "Battery charging power" in kW and "Battery discharging power" in kW. I would like to have these values accessible in Home Assistant so that I can use them in my Power Flow Card Plus configuration.

Describe alternatives you've considered

I have looked into using "Battery Power Raw," but it does not distinguish between charging and discharging. I also tried to find my old YAML configuration where this distinction was available, but I have been unable to locate it.

Additional context

[image]

The screenshot from the WiNet interface showing the values.

Gnarfoz, 2025-02-12:
Not sure why out of 4 battery power sensors, you picked the one that's _not_ suitable*... 😕 
There are three other sensors available: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2250-L2305
- _Signed battery power_ is a combined (positive+negative) sensor that should always work.
- _Battery charging power_ & _Battery discharging power_ are separate (only positive) sensors.

*) On recent firmware versions, it actually is suitable.

jc82517, 2025-02-13:
Sorry, I didn't mean to come across the wrong way. You're doing a great job, and I really appreciate your work.

In the meantime, I took another look at the YAML file and noticed that the mentioned template sensors don’t seem to be working for me. I'm currently troubleshooting the issue.

I also had to adjust the modbus_sungrow.yaml file because it wouldn’t work otherwise:

`modbus_sungrow:
  modbus:
    - name: SungrowSHx
...`

Could it be that something also needs to be adjusted in the templates?

Sorry for the seemingly "dumb" questions – I'm not a programmer, just trying to figure things out. ;)

[image]
[image]

jc82517, 2025-02-13:
I am using the latest modbus_sungrow.yaml (2025-02-07). When I use it as is, I get the error message shown.

[image]

If I modify the start as follows:

sungrow_modbus:
 modbus:
  - name: SungrowSHx
  ...

the first error message disappears, and the Modbus Connection mostly works.

What about the other messages? What am I missing?

jc82517, 2025-02-13:
After a long search, I finally found the error. I'm such an idiot. For some reason, I had used !include_dir_merge_named instead of !include_dir_named in my configuration.yaml. Thank you @Gnarfoz and @TCzerny for your input.

closed 2025-02-13 by jc82517

Gnarfoz, 2025-02-13:
Glad you got it resolved! :-)

> Sorry, I didn't mean to come across the wrong way. You're all doing a great job, and I really appreciate the effort you guys put in.

You didn't, all good. I just wanted to point out that you had bad luck picking the 1 out of 4 that wouldn't work. 😁 

> I know that the firmware version of my Sungrow system must be from sometime before October 2024. However, updating it isn't possible at the moment since it has to be done by the installer. I know I could do it myself, but I want to play it safe because I still have a warranty with them.

If that is firmware 95.01, I'd say stay on that version for now... 95.03 seems to have various problems.