mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #468: Batter Level and Discharging power

Schneeschuher, 2025-02-17:
Hello there,

i am using yaml from 2024-04-05.
It worked pretty well until the new update of my BYD HVM 11 battery.
Since then the battery level and charging power and discharging power are not working anymore.

Any idea how to fix it?

Schneeschuher, 2025-02-18:
@Gnarfoz thanks for your reply...
The inverters are SH10RT-V112, SG6.0RT and a BYD HVM 11.
The current software version is SAPPHIRE-H_03011.95.03 and BERYL-S_03011.01.77.
Yes, our provider made an update.
The battery has got the version BMU 3.24 and BMS 3.30
I hope this helps.

Schneeschuher, 2025-02-18:
Discharging and charging is working now.
Only battery level stays on 100%.

Gnarfoz, 2025-02-18:
Hard to say. Does the BYD battery show up in the WiNet-S interface as a separate device?
Like this: 

[image]

Schneeschuher, 2025-02-19:
No, there a together as a Hybrid.

Sleepy46, 2025-02-24:
I have a similar configuration SH10RT-V112 with Byd HVS 10.2. After the update the battery level is 100% 
only the inverter and the Winet module were updated

Schneeschuher, 2025-02-24:
@Sleepy46 The problem is the update of the inverter. You have to make a downgrade again. Then it is going to work again.

Sleepy46, 2025-02-24:
How i make a Firmware downgrade?

Schneeschuher, 2025-02-24:
You get all the information and the file up here: https://www.photovoltaikforum.com/thread/203177-byd-parallel-ja-oder-lieber-nein/?postID=3446744#post3446744

Sleepy46, 2025-02-24:
We are Talking about Sungrow Inverter Firmware 

OR BYD Batery Update?

Schneeschuher, 2025-02-24:
Sungrow Inverter Firmware Update

Sleepy46, 2025-02-24:
or did you mean this post
https://www.photovoltaikforum.com/thread/238994-soc-immer-100/?pageNo=1

The Post above is only for the Battery. 
at least on the first 30 pages

ahd71, 2025-04-15:
Anyone got a working solution?

What's your version of the Winet-S adapter if you are using that one. Mine is still brokem and a I have "v1" software for the adapter.
Don't have at hand which version of the batteries or inverter but vendor say "latest"

ahd71, 2025-04-15:
Thanks @Gnarfoz . Ive spoken with Sungrow Support now and they confirm I have the latest releases (unfortunately I did not ask why there is one version shown in the web UI and another one which they mention in the security bulletine.)

I have an open case with Sungrow now and let them figure out the cause to the incorrect modbus reading of SOC regression error.

I dont have an installer account (yet at least) and prefer to let my installer handle it but it has now taken so long time so I thought of taking direct contact with Sungrow to get this resolved.

Paulmx2, 2025-05-19:
Hi everyone.  Any luck as I have the same problem.  SH10RT with BYD HVS. 

Always 100% battery SOC in WiNet but correct in iSolarCloud.

Happened in April after my installer updated firmware of battery and inverter.  

I am very reluctant to revert to older firmware but I really need the Battery SOC in Home Assistant.

Does anyone have good documentation on how to revert the firmware?

ahd71, 2025-05-22:
No known solution and my supplier shows close to zero interest in trying to have it resolved and said that sungrow closed the case.

One small difference, my winet shows 0% SOC all the time, not 100% as you have, and isoloarcloud shows correct value just like for you)

closed 2025-09-26 by github-actions[bot]