mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #469: charging session at night is shown as discharging Device Model: SH5.0RS

scruysberghs, 2025-02-19:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

For some reason the sign of my nightly charging session is inverted and shown as discharging. During the day, when there is solar the sign is correct.

**Your Sungrow inverter:**
 - Model:Device Model: SH5.0RS
 - The inverter is connected via (mark one)
   - [] LAN (internal port)

- Are you using a Modbus Proxy (mark one)

  - [] no 

**Home Assistant version:**
 - Version: [e.g. 2025.02.04]

**modbus_sungrow.yaml:**
 - Version/ time stamp : # last update: 2024-11-02
 - [v] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [v] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
in iSolarCloud: 

[image]

**Additional context**
Add any other context about the problem here.

mkaiser, 2025-02-19:
> * Version/ time stamp : # last update: 2024-11-02

this is a known issue with newer firmware version and has been fixed last month.

Can you update your modbus.sungrow.yaml and give a quick feedback if it is working now?

scruysberghs, 2025-02-19:
Ow, sorry,  I copied the "last update" string from my modbus_sungrow.old file and not modbus_sungrow.yaml that I've been running since a couple of days in an attempt to fix the problem. I am indeed already running the "# last update: 2025-02-15" version.

scruysberghs, 2025-02-19:
I i look at the logic you use to go from sensor.battery_power_raw to the signed (charging or discharging) I think it is because the nightly charging session is now started from isolarcloud schedule and not using a Forced mode from Home Assistant

Gnarfoz, 2025-02-19:
> Ow, sorry, I copied the "last update" string from my modbus_sungrow.old file and not modbus_sungrow.yaml that I've been running since a couple of days in an attempt to fix the problem.

If that old copy is in the same directory as the current one, move it somewhere else. Home Assistant might trying to load both.

scruysberghs, 2025-02-19:
> 
> If that old copy is in the same directory as the current one, move it somewhere else. Home Assistant might trying to load both.

In my experience HA only loads the file if the extension is .yaml (at least that is how it works[ if you use a /package subfolder](https://www.home-assistant.io/docs/configuration/packages/)) but I have removed it now and restartted to be sure....

scruysberghs, 2025-02-20:
Running State and System State are always 0. Inverter state is always Running. The charging is now done trough a forcible charge schedule set in iSolarcloud (charge to 50% at 3am.). Discharging is blocked from midnight to 5am to have a slot for EV charging.

I'm adding a temp rule to change the sign between 3am and 5 am until I get rid of that wrong data in my sensor and I can switch to EMHASS based automcontrol using forcible charge and discharge command. Those are indeed always shown correctly.

scruysberghs, 2025-02-21:
At this moment it is charging 5000w and showing me it is discharging:

[image]

Mats1717, 2025-03-11:
> I don't understand yet:
> 
> You are in EMS mode. At night the schedule charges the battery. --> Why is it still in EMS mode?
> 
> I tried to cover some charging variants here:
> 
> [Sungrow-SHx-Inverter-Modbus-Home-Assistant/modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/e8142fce09930a8b7c879ed18160f4e094b2a433/modbus_sungrow.yaml#L1762)
> 
> Line 1762 in [e8142fc](/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/e8142fce09930a8b7c879ed18160f4e094b2a433)
> 
>              {# workaround for SH*RS inverters without working running_state #} 
> I guess yours is not covered, yet.
> 
> Can you tell me what the other states (running states etc.) are when the nightly charging is active?

I steel this tread a little, i think you can add more modes by doing this example for VPP. {# workaround for SH*RS inverters without working running_state #}
            {% if states('sensor.ems_mode_selection') in ["Forced mode", "VPP"] %}
              {# EMS forced mode or VPP #}
I think also there in somthing called compulsonary mode or something.

Gnarfoz, 2025-03-11:
Compulsory mode is chinglish for Forced Mode.

Mats1717, 2025-03-12:
> Compulsory mode is chinglish for Forced Mode.

Maybe, but VPP could be implemented anyway

Gnarfoz, 2025-03-12:
That would treat VPP mode the same as forced mode, but I don't see a relation between VPP mode and forced charge/forced discharge. Are you saying that if you set _EMS mode_ to _VPP_, and _Force command_ to _Forced charge_, your inverter starts charging your battery, and if you set _Force command_ to _Forced discharge_, it will discharge the battery?
That seems contrary to what forced mode is for.

Mats1717, 2025-03-12:
> That would treat VPP mode the same as forced mode, but I don't see a relation between VPP mode and forced charge/forced discharge. Are you saying that if you set _EMS mode_ to _VPP_, and _Force command_ to _Forced charge_, your inverter starts charging your battery, and if you set _Force command_ to _Forced discharge_, it will discharge the battery? That seems contrary to what forced mode is for.
> 
> I can't test this myself, the inverter reverts to self-consumption mode automatically a few seconds after switching to VPP mode. I assume that it requires some kind of heartbeat, similar to _External EMS_ mode.

As i tried to describe earlier an outside company in Sweden manage the battery with External EMS and VPP, Its used for frekvency control of the grid and AI funtion for self cunsumption. In External EMS mode and VPP it uses forced charge and forced discharge. Its  probarbly + 100 000 customers in Sweden and many uses mKaiser Sungrow to read the inverter. It  seems that my suggestion will fix the that disharging/charging is not correct. We are forced to use the Winet-s port because they uses the other port. I have implemented the fix i my config and it seems to solve the problem.

[image]

[image]