mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #470: Host/slave cant force charge slave

lefjomp, 2025-02-20:
I have 2 inverters with 2 batteries, i use mkaiser, its works great, i get almost all sensors from the slave inverter in a second daschboard.
the only thing i cant use is the force charge function on the slave inverter, when i start force charging the host inverter starts to force charge, i have double checked the settings in the daschboard but i cant see anything wrong.
could the problem be that that inverters are host/slave?

mkaiser, 2025-03-05:
Hey, 
it is hard to help you without more information. Could you re-open this issue and use the template, which asks you for many items like inverter type, firmware, etc...

JSoldan87, 2025-03-07:
From what i see the problem is that in the write register functions it says:
      - action: modbus.write_register
        data_template:
          hub: SungrowSHx
if you change this to:
      - action: modbus.write_register
        data_template:
          h**ub: SungrowSHx 2**
it should work.
At least that was my problem

By the way: Mr Kaiser - thank for that incredibly work!

Gnarfoz, 2025-03-07:
All the automations that use `modbus.write_register` in the file for a second inverter (https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow_second_inverter_experimental.yaml) use `slave: !secret sungrow_modbus_slave_inv2`. Meaning they should write to the slave inverter, through the master inverter.

```yaml
    actions:
      - action: modbus.write_register
        data_template:
          hub: SungrowSHx
          slave: !secret sungrow_modbus_slave_inv2
          address: 12999 # reg 13000
```

Perhaps that doesn't work for all registers? Simply changing `hub: SungrowSHx` to `hub: SungrowSHx 2` would mean all writes go directly to the slave inverter. Not sure which is better, or how Sungrow intends it to be done.
It should probably be tested before being rolled out to everyone. Did you do that for all your automations, @JSoldan87?

JSoldan87, 2025-03-07:
I would say there is a missunderstanding:
From my point of view nothing is going trough the master.
'slave' here means modbus slave id which is probably the same for master as for slave. (at least my one are the same)
The concept is the deviation in the IP Adresse. I guess by using the hub: function (which i don't know about) all of the functions write to the master instead to the slave.
I changed this for all slave automations yes. However the rollout should be check by somebody with more background knowledge.
Does this fix work for you ?

lefjomp, 2025-03-07:
thank you all for your answers, i have tried to change the hub to SungrowSHx 2.
this didnt work but i did it only to the experimental slave script, i will try it also on the host script.
if it doesnt work i will try to ad this problem to the issue section on github.
and thanks mkaiser for the great work.

Gnarfoz, 2025-03-07:
You're already in the issue section. ;-)
If you change the hub in the automations in the second file, then you also need to call *those automations*, not the ones from the primary file. Easiest way would be to set the value of `input_number.sg_set_2_xxx` directly via `input_number.set_value` in _Developer Tools > Actions_. Second easiest way would be to have a second dashboard that contains the input_number sliders for the second inverter. They should be called something like `input_number.sg_set_2_xxx` and `input_select.sg_set_2_xxx`.

> I would say there is a missunderstanding: From my point of view nothing is going trough the master. 'slave' here means modbus slave id which is probably the same for master as for slave. (at least [...]

That's kind of true.
The slave inverter's values are available on the master inverter if you query it with slave id 2.
However, Home Assistant doesn't let you have multiple `Hub`s with the same hostname/ip.
And you can't have the same Hub appear in multiple files. So beyond putting all sensors for the slave inverter into the main file and using the normal `SungrowSHx` hub, there's no good option to use this.

Which means the GitHub automation that creates the `modbus_experimental.yaml` was written with talking directly to the slave inverter in mind. Meaning, `secrets.yaml` needs to have:
`sungrow_modbus_host_ip_inv2` = slave inverter ip
`sungrow_modbus_port_inv2` = slave inverter port (502)
`!secret sungrow_modbus_slave_inv2` = `1`
And it also means the automations included in that file are incorrect and need to be adjusted to use the `SungrowSHx 2` hub, as you mentioned, @JSoldan87.

The GitHub automation should thus probably be adjusted.

Gnarfoz, 2025-03-10:
@JSoldan87 could you perhaps provide your full `secrets.yaml`, `modbus_sungrow.yaml` and `modbus_sungrow_second_inverter_experimental.yaml`?
You can just attach them here.

Can you also confirm that with your current files, you can do things like shut down the second inverter, adjust its settings (like battery charge & discharge power, backup SoC, etc)?
If so, how do you access them? Did you duplicate the default dashboard and adjust the entity names?

lefjomp, 2025-03-10:
I must ad that im connected through the winet stick, i cant seem to get connected trough  modbus, I can't get any ip from the inverter , in my howt inverter I was at first connected through the winet stick and I was able to start stop forced charging, now I'm connected with modbus instead with the host, so I should atleast be able start stop the forced charging on my slave inverter when connected with the winet stick.

Gnarfoz, 2025-03-10:
I was more interested in @JSoldan87's files because they seem to have a working setup. ;-)

The "second inverter" file as it is in the repository, combined with setting `sungrow_modbus_slave_inv2` to `1` will lead to the behaviour you described. The automations in the second inverter file target the Modbus "hub" from the main file. I assume the intention was to have `sungrow_modbus_slave_inv2` to be set to `2`, which should be the slave inverter. But that probably only works when connected to the LAN port of the inverter, not through WiNet-S.
I can only recommend switching to the inverter's own port. You can change its network settings by scanning the QR code on the WiNet-S stick and using the "local access" option in the iSolarCloud app.

You might notice a lot of shoulds, woulds, probably's -- it's hard to check some of these things without access to an actual second inverter. ;-)

Gnarfoz, 2025-03-10:
As I was trying to point out, those would be @JSoldan87's files. ;-)

JSoldan87, 2025-03-10:
Yes but not every day ;-) I will come back regarding that topic tomorrow.

lefjomp, 2025-03-11:
i finally managed to connect the slave to modbus and get an ip, so now i am actually connected and i get almost all sensors to work.
i get some strange readings in one sensor, i will ad a picture.
unfortunately i cant force charge/discharge the slave, nothing happens when i activate it, but its possible to change the force charge/discharge power in w, that slider works and sends the command to the inverter.

lefjomp, 2025-03-11:
i checked the logs and when i try to change the Max battery charge power i get this error in the log:

Logger: homeassistant.components.automation.sungrow_inverter_2_update_battery_max_charge_power
Källa: helpers/script.py:2039
integration: Automation ([dokumentation](https://www.home-assistant.io/integrations/automation), [ärenden](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+automation%22))
Inträffade först: 12:35:19 (3 händelser)
Senast loggade: 12:36:23

sungrow inverter 2 update battery max charge power: Already running

any idea whats wrong?

JSoldan87, 2025-03-11:
> [image]
>
> i finally managed to connect the slave to modbus and get an ip, so now i am actually connected and i get almost all sensors to work. i get some strange readings in one sensor, i will ad a [...]

Hello i am at the same stage: I get all the sensors however Forced Charge / Discharge is not availible. 
I will report if i can resolve that problem.

My idea rearding you wrong sensors: Your slave inverter has no meter at all - so there is some unvailid data in it.

lefjomp, 2025-03-11:
i think i solved  some of the problems, i now can change forced charge/discharge power, i checked the automation settings and i noticed that the automations for the slave inverter where pointing towards the wrong inverter, i changed them to  SungrowSHx 2
. the 2 was missing, not it works better, i cant force charge /discharge though, but the sliders are working now, im going to look more into it later in the evening

JSoldan87, 2025-03-11:
[sungrow2.txt](https://github.com/user-attachments/files/19185298/sungrow2.txt)

From what i see i can control both Inverters now. I will now work on the Automations.

Keep in mind my configuration: Both inverters are working on the Modbus LAN (not WI-NET) and NOT in Master Slave.

lefjomp, 2025-03-17:
have got almost everything to work now with the slave inverter, the only thing that i cant seem to get to work is the force charge/decharge .
nothing happens when i activate that function, 
the host inverter works flawless, also the force charge /decharge works there.
any idea how to fix the slave inverter?

lefjomp, 2025-03-17:
this is what i get in the log when i try to force charge the slave

Logger: homeassistant.components.automation.sungrow_inverter_2_update_battery_forced_charge_discharge_cmd_input_select_update_3
Källa: components/automation/__init__.py:738
integration: Automation ([dokumentation](https://www.home-assistant.io/integrations/automation), [ärenden](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+automation%22))
Inträffade först: 21:36:13 (2 händelser)
Senast loggade: 21:36:43

Error while executing automation automation.sungrow_inverter_2_update_battery_forced_charge_discharge_cmd_input_select_update_3: Invalid option: 187 (possible options: Stop (default), Forced charge, Forced discharge)
Error while executing automation automation.sungrow_inverter_2_update_battery_forced_charge_discharge_cmd_input_select_update_3: Invalid option: 204 (possible options: Stop (default), Forced charge, Forced discharge)

Gnarfoz, 2025-03-19:
Why do you have at least 3 copies of that automation? 😁  (`_3` at the end of the name)
The error message means that your automation [.sungrow_inverter_2_update_battery_forced_charge_discharge_cmd_input_select_update_3](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow_second_inverter_experimental.yaml#L2851-L2867) which updates the slider tries to use the raw Modbus value instead of the translated string that should be in `sensor.battery_forced_charge_discharge_cmd_2`.

You sohuld check what's in `sensor.battery_forced_charge_discharge_cmd_2` when that happens.

closed 2025-07-18 by github-actions[bot]