mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #473: Werte ändern sich selbstständig

Tr1n0va, 2025-02-22:
Hi,
evtl. hat ja jemand eine Idee was hier bei mir passiert.
In unregelmäßigen Abständen ändern sich bei mir ohne mein Zutun die Werte von:

Force charge, Max Batterie charge und Max Batterie discharge

Das steht dazu unter anderem im Log:

[image]

Ich kann zu 100% sagen das ich zu diesen Zeiten und auch sonst niemand, etwas irgendwo geändert hat.
Das muss irgendwie HA oder der Wechselrichter selber machen.

Für Tipps und/oder Hilfe währe ich echt Dankbar!

Vielen Dank,
Gruß
Trinova

Tr1n0va, 2025-02-22:
Sorry, diese Fehler gibt es noch dazu.
Die tauchen aber tatsächlich nicht immer in Verbindung mit dem Phänomen auf:

Logger: homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update
Quelle: components/automation/__init__.py:738
Integration: Automation (Dokumentation, Probleme)
Erstmals aufgetreten: 09:40:14 (1 Vorkommnisse)
Zuletzt protokolliert: 09:40:14

Error while executing automation automation.sungrow_inverter_max_soc_input_slider_update: Invalid value for input_number.set_sg_max_soc: 20.0 (range 50.0 - 100.0)

Logger: homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update
Quelle: helpers/script.py:2039
Integration: Automation (Dokumentation, Probleme)
Erstmals aufgetreten: 09:40:14 (1 Vorkommnisse)
Zuletzt protokolliert: 09:40:14

sungrow inverter max SoC input slider update: Error executing script. Invalid data for call_service at pos 1: Invalid value for input_number.set_sg_max_soc: 20.0 (range 50.0 - 100.0)

Logger: homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update
Quelle: components/automation/__init__.py:738
Integration: Automation (Dokumentation, Probleme)
Erstmals aufgetreten: 09:40:14 (1 Vorkommnisse)
Zuletzt protokolliert: 09:40:14

Error while executing automation automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update: Invalid option: Unknown - should not see me! code: 0 (possible options: Stop (default), Forced charge, Forced discharge)

Logger: homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update
Quelle: helpers/script.py:2039
Integration: Automation (Dokumentation, Probleme)
Erstmals aufgetreten: 09:40:14 (1 Vorkommnisse)
Zuletzt protokolliert: 09:40:14

sungrow inverter update battery forced charge discharge cmd input select update: Error executing script. Error for call_service at pos 1: Invalid option: Unknown - should not see me! code: 0 (possible options: Stop (default), Forced charge, Forced discharge)

mmierke, 2025-02-22:
Passiert bei mir und meinem Nachbarn auch in letzter Zeit. Ich hatte schon meinen Installateur oder die Chinesen im Verdacht, kann das aber ausschließen, da ich meiner PV mittlerweile den Zugang zum Internet gestrichen habe. An der Software auf dem Wechselrichter kann es auch nicht liegen, da die Anlage meines Nachbarn auf einem alten Stand ist.

Ich vermute, und das ist bisher noch nicht überprüft, dass ein seltsamer Wert im Sensor den Wert der Input Number ändert, was dann wieder dazu führt, dass die Automatisierung den Wert des Wechselrichters verstellt. Ich habe jetzt erstmal alle Automatisierungen disabled, die Werte des Wechselrichters verändern. Einen hatte ich gestern vergessen und exakt der Parameter (min SoC) war heute Morgen verändert.

Gnarfoz, 2025-02-22:
Bei @Tr1n0va sind da in den Logs nur die Automatisierungen zu sehen, die den Wert des Sliders dem tatsächlichen Parameter angleichen. Das heißt, der WR hat per Modbus einen veränderten Wert geliefert. Wer ihn geändert hat, geht daraus nicht hervor.

Außerdem stellt irgendwas den max SoC auf 20%. Das ergibt wenig Sinn und geht mit den Slidern hier gar nicht, da das außerhalb des Regelbereich des Sliders ist.

Wenn andere Systeme zu 100% ausgeschlossen werden können... Eventuell ein schlecht oder gar nicht abgeschirmtes Kabel im Einsatz, so dass Werte falsch übertragen werden?

Tr1n0va, 2025-02-22:
Ich hab noch ein paar Fehler von heute.
Habe grad mal ein Firmware Update gemacht.

Logger: pymodbus.logging
Quelle: runner.py:154
Erstmals aufgetreten: 13:39:30 (1 Vorkommnisse)
Zuletzt protokolliert: 13:39:30

Exception response 132 / 0

Logger: homeassistant.components.modbus.modbus
Quelle: components/modbus/modbus.py:310
Integration: Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 13:39:30 (1 Vorkommnisse)
Zuletzt protokolliert: 13:39:30

Pymodbus: SungrowSHx: Error: device: 1 address: 4989 -> pymodbus returned isError True

mmierke, 2025-02-22:
Das mit dem Kabel würde ich ausschließen, da das Problem nicht nur eine Anlage betrifft.

Tr1n0va, 2025-02-22:
> Bei [@Tr1n0va](https://github.com/Tr1n0va) sind da in den Logs nur die Automatisierungen zu sehen, die den Wert des Sliders dem tatsächlichen Parameter angleichen. Das heißt, der WR hat per Modbus [...]

Kabel kann ich eigentlich auch ausschließen, ist ein neues Cat7…

mmierke, 2025-02-22:
Firmware wäre seltsam, da die Anlage meines Nachbarn quasi noch nie eine neue Firmware zu Gesicht bekommen hat. Meine habe ich letzte Woche upgedated.

Tr1n0va, 2025-02-22:
Kann es was damit zu tun haben?

[image]

Das er irgendwie % erwartet und aber über den Slider 6000w bei mir eingestellt bekommt?

Gnarfoz, 2025-02-22:
Was hast du denn überhaupt für einen Wechselrichter? :-)
Beim SH10RT funktioniert es wie es ist. Was steht bei dir im Input-Register 5628? (Ist auch als Entity "BDC rated power" vorhanden.)

[image]

Ich glaube eher, dass dieser Teil der Doku sagen will:
1. Beim SH3.0-6.0RS ist der Wertebereich 0 - 5000 W.
2. Beim SH5.0-10RT ist der Wertebereich 0 - (100% von dem was in Register 5628 steht) - in Watt.
Denn die spalte direkt daneben sagt, dass die Einheit "1 W" ist, egal um welches Gerät es geht.

...Und ich frage mich, was beim SH15-25T gilt. Die werden gar nicht erwähnt. ;-)

GarfieldMZ, 2025-02-23:
Ich habe den SH5.0RT-V112. 
Nachdem ich die schon an mir gezweifelt habe und auch schon den Anlagen-Installateur in Verdaxcht hatte, habe ich jetzt diesen Thread in dem ja auch andere das Phänomen der sich selbst ändernden Werte haben. Ich habe mit in HomeAssistant schon eine Automation gebaut die mir eine Nachricht schickt, wenn sich wieder mal ein Wert plötzlich ändert. Hauptsächlich ist es bei mit der "reserved SOC for backup" der unverhofft von 0 auf 85% springt (z.b. heute wieder gegen 19:42). Hatte auch schon das der "max battery charge power" von 5000 auf 800 gesetzt wurde. 
Bin ja froh wenn ich nicht der einzige mit dem Problem bin. Die Frage ist nur, warum passiert das!?

Tr1n0va, 2025-02-23:
Ich habe den SH6.0RT und auch bei mir wurde es gestern Abend wieder umgestellt.
Auffällig ist das es immer die selben Werte sind 204 und dann 200 innerhalb kurzer Zeit.

[image]

Gnarfoz, 2025-02-23:
Und wenn du testweise mal auf Forced mode +force command (discharge oder charge ist egal) und force charge discharge power... 80? Stellst, mit wie viel Watt wird dann geladen/entladen?
Mit 80 W oder mit 0,8 × 6000 W = 4800 W? 🤔

mmierke, 2025-02-23:
> Was hast du denn überhaupt für einen Wechselrichter? :-) Beim SH10RT funktioniert es wie es ist. Was steht bei dir im Input-Register 5628? (Ist auch als Entity "BDC rated power" vorhanden.) [...]

Das mit dem SH10RT kann ich leider so nicht bestätigen, bei meinem ändern sich seit ca. 2 Wochen ständig die Werte.

mmierke, 2025-02-23:
Entfernen wir uns nicht gerade vom eigentlichen Problem? Meinem Verständnis nach ist das Problem, dass sich die Werte überhaupt verändern und nicht, wohin sie sich verändern. Hat schon jemand eine Idee, was die Werte ändert? Meine Anlage ist von 2023 und das Problem tritt bewusst seit ca. 2-3 Wochen auf. Wobei unterschiedliche Parameter bis hin zum export power limit mode (wechsel auf enabled) und export power limit (200-1000W) betroffen sind. Min backup soc von 85% ist auch sehr beliebt.

Tr1n0va, 2025-02-23:
Ich hab es als erstes so ca. vor einem halben Jahr bemerkt.
Hab es seit dem immer auf versehentliches verschieben der Regler geschoben
und dann halt auch Automationen erstellt um die Werte passend zu halten.
Aber das kann ja nicht der Sinn sein.

alexonbandit, 2025-02-23:
Ich habe leider die selben Probleme. Ich habe jetzt alle Automationen im Zusammenhang mit dieser Sungrow Integration deaktivieren. Logischerweise bleiben nun meine eingestellten Werte erhalten. Dies bedeutet aber, dass ich wenn ich was verändern will, erst die Automation aktivieren muss die Änderung vollziehen und wieder deaktivieren muss. Schön ist anders. Aber es funktioniert.
Kann es einen Zusammenhang mit HASS 2025 branch und den slidern geben?

mmierke, 2025-02-23:
Ich denke nicht, dass es an den Slidern hängt, da sich bei mir zum Beispiel auch der Export Power Limit Mode geändert hat und das ist kein Slider. Könnte sein, dass das im Zusammenhang mit der Überarbeitung der Automations hängt. Ich habe mir einen Helper Button erstellt und die Automationen angepasst(noch nicht fertig) Modbus Register werden nur noch geschrieben, wenn der Helper auf on steht. Mich hat vorher schon gestört, dass man die Werte so leicht aus Versehen ändern kann.

mkaiser, 2025-02-23:
Finde das sehr merkwürdig. Die Update Automations aktualisieren ja nur die Slider mit dem aktuellen Wert der Hardware. 

Ich würde gerne ausschließen, dass kein anderes Programm (EVCC?) darauf Zugriff hat. Bitte auch mal die Verbindung zur Isolarcloud kappen.

Gnarfoz, 2025-02-23:
Naja, wenn die Slider Automation den Slider dem echten Wert anpasst, triggert durch diese Änderung des Sliders auch die Automation, die den Wert ändern würde.
Das _sollte_ ein No-Op sein, aber evtl. sollte man da dann doch mal conditions mit drauf legen, die erst checken, ob der Wert des Sliders überhaupt vom tatsächlichen Wert abweicht. Sprich, die Actions nur ausführen, wenn es tatsächlich etwas zu ändern gibt.

Gnarfoz, 2025-02-23:
> Kann es einen Zusammenhang mit HASS 2025 branch und den slidern geben?

Was meinst du mit "HASS 2025 branch"?
Benutzt du eine andere Version als von hier im Repository? 🤔

Tr1n0va, 2025-02-23:
> Finde das sehr merkwürdig. Die Update Automations aktualisieren ja nur die Slider mit dem aktuellen Wert der Hardware.
> 
> Ich würde gerne ausschließen, dass kein anderes Programm (EVCC?) darauf Zugriff hat. Bitte auch mal die Verbindung zur Isolarcloud kappen.

EVCC hab ich zwar auch, aber erst seit ca. 2 Wochen. Das Problem bestand schon lange davor. Ansonsten hat bei mir soweit mir bekannt ist nichts Schreibenden Zugriff auf den Wechselrichter. Der Wechselrichter hat auch keine Möglichkeit das Internet zu nutzen, also auch keine Cloud.

mmierke, 2025-02-23:
EVCC habe ich auch im Einsatz. Aber wieso sollte EVCC am backup SoC rumbasteln). Die Cloud ist bei mir seit einiger Zeit gekappt.

mmierke, 2025-02-23:
Bei mir hat sich gerade Reserved Backup auf 85% geändert, ohne dass eine Automation aktiv war. Sehr Spuki

mmierke, 2025-02-23:
> Finde das sehr merkwürdig. Die Update Automations aktualisieren ja nur die Slider mit dem aktuellen Wert der Hardware.
> 
> Ich würde gerne ausschließen, dass kein anderes Programm (EVCC?) darauf Zugriff hat. Bitte auch mal die Verbindung zur Isolarcloud kappen.

Ja, aber die zweite Automation setzt einen Wert in der PV basierend auf dem Slider Wert. Was war also zuerst?

EVCC ist ein guter Punkt. Hab jetzt erstmal meine Einträge für die Sungrow von Template auf custom geändert und EVCC nur minimale Daten gegeben um zu verhindern, dass EVCC versucht die Sungrow zu steuern.

Gnarfoz, 2025-02-23:
> Bei mir hat sich gerade Reserved Backup auf 85% geändert, ohne dass eine Automation aktiv war. Sehr Spuki

85 ist für mich auch direkt verdächtig, weil das in Hexadezimal 0x55 ist und für diverse "an / aus" Einstellungen bzw. Register als "aus" verwendet wird. Schreibt da irgendwas ins falsche Register? 🤔

mmierke, 2025-02-23:
> 85 ist für mich auch direkt verdächtig, weil das in Hexadezimal 0x55 ist und für diverse "an / aus" Einstellungen bzw. Register als "aus" verwendet wird. Schreibt da irgendwas ins falsche Register? 🤔

Eigentlich sollte da gar nichts schreiben. Alles über 0% ist so nicht gewollt.
Aber mit 85 hast du da recht.

Gnarfoz, 2025-02-23:
Wenn du da eh 0% willst, hast du vermutlich den Backup Modus auch aus, oder?
Dann wäre der Wert vom Backup SoC auch Wumpe, der hat dann keine Wirkung.

Aber das ist ja eh nicht das eigentlich Problem. 😬

mmierke, 2025-02-23:
> Wenn du da eh 0% willst, hast du vermutlich den Backup Modus auch aus, oder? Dann wäre der Wert vom Backup SoC auch Wumpe, der hat dann keine Wirkung.
> 
> Aber das ist ja eh nicht das eigentlich Problem. 😬

Noe, der ist ein. Ich hätte schon gerne Strom vom Dach, wenn das Netz wegbricht. Wenn der Backup SoC also auf 85% geht, dann wird meine Batterie fleissig aus dem Grid geladen. Aber EVCC und aktive Batteriesteuerung könnte schon zumindest mit diesem Teil zu tun haben.

monsieurhannes, 2025-02-23:
Hier ist das selbe Problem. Seit einigen Tagen (ca. 16.2., nicht sicher ob davor auch schon) exportiert die Anlage keinen Überschuss mehr ins Netz.
Das PV-System besteht aus 2 Strings (ost-west) und dem Inverter, keine Batterie. 
Installiert wurde die Anlage von 1,5Grad und da ist auch noch ein heartbeat im System, der allerdings nichts steuern soll.

Habe gerade das Dashboard eingerichtet, was bisher nicht notwendig war, weil die Anlage im Prinzip "dumm" laufen sollte und von HA nur überwacht wird. Wenn ich den EMS Mode ändere, springt er kurze Zeit später wieder zurück zu "External EMS", Export Power limit ist auch an, warum auch immer, lässt sich nicht permanent ändern.

[image]

Logs für den EMS sensor:

[image]

[image]

Wie kann ich den Modbuswert händisch umstellen?

Gnarfoz, 2025-02-23:
Sieht nicht nach demselben Problem aus. Du hast ein externes EMS im Einsatz (vermutlich dieser "Heartbeat"), sonst wäre der EMS Modus nicht auf "externes EMS".

Es gibt keinen relevanten Unterschied zwischen "händisch" ins Modbus Register schreiben und dem Dropdown. Modbus Register kann man auf viele Arten bearbeiten. QModMaster, mbpoll, das Web-Interface vom WiNet-S hat auch eine Möglichkeit dafür, über Home Assistant `modbus.write_register` aufrufen...

monsieurhannes, 2025-02-23:
Bin mir nicht so ganz sicher ob das passt, werde aber mal das Heartbeat abstöpseln (lassen), bin nicht vor Ort. 
Wenn ich eine Sache ändere folgt eine kleine Kaskade an Änderungen an den anderen Werten, zumindest wirkt das so. Vielleicht hast du aber recht und Heartbeat funkt rein, warum auch immer.

Es gibt auch [diesen commit](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/193e5a802dcd12bb7601763c60297cfbf26710ae#diff-4a5f197ebed75076a2d49e4d10b18dae9a5c7cdf6fa55bd35b45c1b75e1eea18) vom 15.2., was ganz gut in die Zeit passt. Kann es was damit zu tun haben? 
(ich weiss ehrlich gesagt nicht ob die yaml-files automatisch von GitHub geladen/geupdated werden? Wenn nicht, ist die Frage natürlich hinfällig)

Gnarfoz, 2025-02-23:
Die mitgelieferten Scripts sind als bequeme Alternative gedacht, damit man die jeweilige Sache nicht per Hand machen muss. Die werden nur aufgerufen, wenn man das zusätzlich so einrichtet. Ein Beispiel, wie eine entsprechende Automatisierung aussehen könnte, ist (auskommentiert) auch dabei.

mkaiser, 2025-02-24:
> Hier ist das selbe Problem. Seit einigen Tagen (ca. 16.2., nicht sicher ob davor auch schon) exportiert die Anlage keinen Überschuss mehr ins Netz. Das PV-System besteht aus 2 Strings (ost-west) und dem Inverter, keine Batterie. Installiert wurde die Anlage von 1,5Grad und da ist auch noch ein heartbeat im System, der allerdings nichts steuern soll.
> 
> Habe gerade das Dashboard eingerichtet, was bisher nicht notwendig war, weil die Anlage im Prinzip "dumm" laufen sollte und von HA nur überwacht wird. Wenn ich den EMS Mode ändere, springt er kurze Zeit später wieder zurück zu "External EMS", Export Power limit ist auch an, warum auch immer, lässt sich nicht permanent ändern.
> 
>  (Bilder rausgenommen)
>
> Wie kann ich den Modbuswert händisch umstellen?

Hi, 

sieht aus, als wäre das Export Limit bei dir eingeschaltet und auf 0W, was das Nicht-exportieren wohl gut erklärt. 

[image]

Ich tippe darauf, dass 1,5Grad da noch eine weitere Steuerung installiert hat, die regelmäßig "ihre Einstellungen" überschreibt.

mmierke, 2025-02-26:
Ich habe jetzt ein input_boolean "sg_update_pv_setting" erstellt und eine Condition in die Automations eingebaut, die verhindert, dass die Werte aus Versehen überschrieben werden. Außerdem habe ich ein System aus Helpern für meine präferierten Werte und Binary Sensors um Abweichungen zu erkennen.

[image]

[image]

[image]

[image]

Ist aber noch WiP.

Gnarfoz, 2025-02-26:
Smart-Home-Detektiv 🕵🏻‍♀️

mmierke, 2025-02-26:
Ein Beispiel

`alias: sungrow inverter update battery discharging start power
description: Sets battery discharging start power
triggers:
  - trigger: state
    entity_id:
      - input_number.set_sg_battery_discharging_start_power
conditions: []
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_boolean.sg_update_pv_settings
            state: "on"
        sequence:
          - action: modbus.write_register
            data_template:
              hub: SungrowSHx
              slave: 1
              address: 33149
              value: >-
                {{ states('input_number.set_sg_battery_discharging_start_power')
                |float /10 | int}}
      - conditions: []
        sequence:
          - action: input_number.set_value
            target:
              entity_id: input_number.set_sg_battery_discharging_start_power
            data:
              value: "{{ states('sensor.battery_discharging_start_power') }}"
mode: single
`

Gnarfoz, 2025-02-26:
YAML kann man gut so posten, damit es lesbar bleibt:
\```yaml
Dein YAML
\```

Gnarfoz, 2025-02-26:
> [image]
> 

So eine Art Sicherheitsschalter ist natürlich ganz cool. Man könnte auch die Slider durch Eingabefelder ersetzen (`mode: box`), das ist auch mobil besser benutzbar.

maikdm, 2025-02-28:
Gibt es jetzt schon irgendeine Lösung für das Problem? Bei mir verstellen sich seit 2 Tagen auch diverse Werte von selbst

Gnarfoz, 2025-02-28:
Es gibt bisher noch nicht mal eine nachvollziehbare Ursache für das Verhalten.

Habt ihr jeweils mal geprüft, ob die automatischen Firmware-Updates in iSolarCloud an oder aus sind?
Theoretisch sollte ja das einzige andere zugreifende Gerät der WiNet-S sein, vielleicht hat eins der jüngeren Firmware-Updates das ausgelöst?

mmierke, 2025-03-03:
> > [image]
> 
> So eine Art Sicherheitsschalter ist natürlich ganz cool. Man könnte auch die Slider durch Eingabefelder ersetzen (`mode: box`), das ist auch mobil besser benutzbar.

Ich hab die Slider beibehalten und nur die Steps angepasst. Das geht auch sehr gut.

mmierke, 2025-03-03:
> > Ich habe leider die selben Probleme. Ich habe jetzt alle Automationen im Zusammenhang mit dieser Sungrow Integration deaktivieren. Logischerweise bleiben nun meine eingestellten Werte erhalten. Dies bedeutet aber, dass ich wenn ich was verändern will, erst die Automation aktivieren muss die Änderung vollziehen und wieder deaktivieren muss. Schön ist anders. Aber es funktioniert. Kann es einen Zusammenhang mit HASS 2025 branch und den slidern geben?
> 
> Genau so hab ich es jetzt auch gemacht. Nun verstellt sich auch nichts mehr und normal muss man da auch nicht mehr ran wenn alles läuft

Ich hab meinen Automations einen update Schutz in Form eines input_boolean verpasst und conditional cardsverwendet um die Slider nur anzuzeigen, wenn der Bool auf true steht, Ansonsten zeige ich dei aktuellen Werte des Wechselricheters an. Zusätzlich habe ich noch helper für prefered values eingebaut, um sofort zu erkennen, ob sich ein Wert verändert hat.

Also ungefährs so:
```yaml
type: vertical-stack
cards:
  - type: conditional
    conditions:
      - condition: state
        entity: input_boolean.sg_update_pv_settings
        state: "on"
    card:
      type: entities
      entities:
        - entity: input_number.sg_set_min_soc
        - entity: input_number.sg_set_max_soc
        - entity: input_number.sg_set_reserved_soc_for_backup
        - entity: input_number.sg_set_battery_forced_charge_discharge_power
        - entity: input_number.sg_set_battery_max_charge_power
        - entity: input_number.sg_set_battery_max_discharge_power
        - entity: input_number.sg_set_battery_charging_start_power
        - entity: input_number.sg_set_battery_discharging_start_power
      title: Battery Settings
      state_color: false
      show_header_toggle: false
  - type: conditional
    conditions:
      - condition: state
        entity: input_boolean.sg_update_pv_settings
        state_not: "on"
    card:
      type: entities
      entities:
        - entity: sensor.min_soc
          card_mod:
            style: |
              :host {
                {% if states("sensor.min_soc")|int != states("input_number.sg_pref_min_soc")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.max_soc
          card_mod:
            style: |
              :host {
                {% if states("sensor.max_soc")|int != states("input_number.sg_pref_max_soc")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.reserved_soc_for_backup
          card_mod:
            style: |
              :host {
                {% if states("sensor.reserved_soc_for_backup")|int != states("input_number.sg_pref_reserved_soc_for_backup")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.battery_forced_charge_discharge_power
          card_mod:
            style: |
              :host {
                {% if states("sensor.battery_forced_charge_discharge_power")|int != states("input_number.sg_pref_battery_forced_charge_discharge_power")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.battery_max_charge_power
          card_mod:
            style: |
              :host {
                {% if states("sensor.battery_max_charge_power")|int != states("input_number.sg_pref_battery_max_charge_power")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.battery_max_discharge_power
          card_mod:
            style: |
              :host {
                {% if states("sensor.battery_max_discharge_power")|int != states("input_number.sg_pref_battery_max_discharge_power")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.battery_charging_start_power
          card_mod:
            style: |
              :host {
                {% if states("sensor.battery_charging_start_power")|int != states("input_number.sg_pref_battery_charging_start_power")|int %}
                  color: red;
                {% endif %}
              }
        - entity: sensor.battery_discharging_start_power
          card_mod:
            style: |
              :host {
                {% if states("sensor.battery_discharging_start_power")|int != states("input_number.sg_pref_battery_discharging_start_power")|int %}
                  color: red;
                {% endif %}
              }
      title: Battery Settings
      state_color: false
      show_header_toggle: false

```

mmierke, 2025-03-03:
Und eine Automation:
```yaml
alias: sungrow inverter update battery discharging start power
description: Sets battery discharging start power
triggers:
  - trigger: state
    entity_id:
      - input_number.sg_set_battery_discharging_start_power
conditions: []
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_boolean.sg_update_pv_settings
            state: "on"
        sequence:
          - action: modbus.write_register
            data_template:
              hub: SungrowSHx
              slave: 1
              address: 33149
              value: >-
                {{ states('input_number.sg_set_battery_discharging_start_power')
                |float /10 | int}}
      - conditions: []
        sequence:
          - action: input_number.set_value
            target:
              entity_id: input_number.sg_set_battery_discharging_start_power
            data:
              value: "{{ states('sensor.battery_discharging_start_power') }}"
mode: single
```

aufi99, 2025-03-04:
Hallo, ich habe hier leider das gleiche Problem.
Aufgetreten ist es das erste mal nach dem ich ein Update vom "Operating System 14.2" (und noch ein Update im Zusammenhang mit dem Homeassistant System, sehe nur nicht wer was das für ein Update war)

Verstellt wird bei mir oft die "Export Power Limit" und die "Set max Battery Charge Power".

Auch wurde überhaupt das erste mal der Wert "Export Power Limit Mode" auf Enabled gesetzt. Leider bekomme ich den Modus "Disabled" nicht wieder richtig hin. Der Wechselrichter pendelt dann zwischen "Sungrow Inverter State" Running und De-Rating... :/

mmierke, 2025-03-04:
Derating bedeutet, dass der export power limit mode enabled ist und das export power limit greift. Bei mir stand zweiteres mal bei 1000.

mmierke, 2025-03-04:
Interessant ist der tatsächliche Wert in der PV sensor.export_power_limit und sensor.export_power_limit_mode. Der sollte sich spätestens nach 1 Minute angepasst haben. Wobei der Mode der entscheidende ist, Das Problem hängt meiner Meinung nach an HA. Bei meinem Nachbarn ist es auch aufgetaucht und der hat eine sehr alte Firmware.

mmierke, 2025-03-04:
Seit ich die Automations angepasst habe, sind meine Werte stabil. Ich habe gestern auch wieder die iSolarcloud aktiviert. Für EVCC verwende ich zur Zeit eine custom integration und nicht das Template.

aufi99, 2025-03-04:
> Interessant ist der tatsächliche Wert in der PV sensor.export_power_limit und sensor.export_power_limit_mode. Der sollte sich spätestens nach 1 Minute angepasst haben. Wobei der Mode der [...]

Hier scheint es eine Abweichung zu geben

[image]

Wobei es hier als übernommen angezeigt wird

[image]

mmierke, 2025-03-04:
Nur um sicher zu gehen. Ist das der sensor.export_power_limit_mode oder der input_select.sg_set_export_power_limit_mode?

mmierke, 2025-03-04:
Oder so ähnlich. ich hab die Namen etwas für mich angepasst

mmierke, 2025-03-05:
Also ein typische Situation

Zuerst passt sich die input number dem geänderten Modbus Wert an
[image]

In diesem Fall wird der Wert wieder zurückgeändert bevor die 2te Automation zieht
[image]

Hier wird der Wert wieder von 85 auf 0 geändert
[image]

Wenn die Reihenfolge aber zuerst den Wert schreibt und dann den Slider wieder auf den gewollten Wert ändert, setz die Automation in diesem Fall den Wert zu 85 im Inverter.

Außerdem wird die Automation, die den Wert schreibt nicht in jedem Fall ausgelöst.

mmierke, 2025-03-05:
Der letzte Call zu **sungrow inverter update reserved soc for backup** war vor der Änderung des Wertes zu 85%
[image]

Gnarfoz, 2025-03-05:
Die Condition ist so oder so eine gute Idee, genau so wie ein Check, dass `modbus.write_register` nur aufgerufen wird, wenn der neue Wert tatsächlich anders ist, als der alte Wert.
Aber die Frage bleibt: **Warum ändert sich der Modbus-Wert sich initial überhaupt?**

Was haben die betroffenen Leute und die nicht betroffenen Leute jeweils gemeinsam? 🤔

mmierke, 2025-03-05:
Der Wert der input_number wurde in beiden Fällen durch die Automation geändert und war nie unknown oder unavailable.

[image]

mmierke, 2025-03-05:
Der Auslöser für die Automation war das geändert Modbus Register

[image]

mmierke, 2025-03-05:
Die 170% verstehe ich nicht

Gnarfoz, 2025-03-05:
Die letzten Screenshots sind jetzt wieder von der "Slider update"-Automatisierung. Die andere (`sungrow inverter update reserved soc for backup`) ist die, die spannend wäre. Die hattest du oben auch als Screenshot, aber ohne die Details, was sie ausgelöst hat (Traces -> Step Details -> Changed Variables).

0xAA (170) ist der Wert für "Enable" und das Gegenstück zu 0x55 (85) für "Disable", für die meisten Holding-Register hier wo man etwas an- oder abschalten kann.
Nur sehe ich bisher nicht, wer oder was hier z.B. fälschlicherweise 0x55 oder 0xAA in Register 13100 (backup SoC) schreiben würde.

Eigentlich lässt sich damit sogar die Integration hier ausschließen, denn:
[Die Automatisierung, die in Register 13100 schreibt, nutzt als Wert **immer** den Wert von `input_number.set_sg_reserved_soc_for_backup`](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2757-L2772) und [die hat einen Maximalwert von 100](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2541-L2545).
Da sollte es gar nicht möglich sein, dass wir 170 in das Register schreiben.

*Irgendwer oder irgendwas externes macht diese Änderungen.*

Hat schon mal jemand der betroffenen folgendes Setup probiert:
- Ausschließlich HA mit dem WR verbinden.
- Direkt am WR-LAN-Port (oder falls SHxT, per RS485-Wandler am Logger-Port).
- WiNet-S(2) vom Netzwerk trennen oder besser ganz abstecken.
- Alle anderen Drittanbieter-Geräte, die mit der PV-Anlage interagieren, vom Netzwerk trennen/abschalten.
- Die _unmodifizierte_ `modbus_sungrow.yaml` von hier verwenden, ohne ergänzende Workarounds.

mmierke, 2025-03-05:
Die Update Automation setzt bei mir den Wert nur dann, wenn ich den input_boolean auf on setze (true wäre besser gewesen, aber die einfachen Lösungen fehlen hier). Das bedeutet, der Wert kommt nicht aus HA. Der Wert muss also von Hybrid Inverter kommen. 

Zuerst geht der Wert des Sensors auf 170 und die Automation schlägt fehl:
[image]

Dann geht der Wert auf 85:

[image]
Und dann auf 0:
[image]

85 und 170 sind eigentlich in diesem Fall sinnbefreit, obwohl 85 immerhin ein sinnvoller Wert sein könnte.

mmierke, 2025-03-05:
Okay, mal wild geraten. Wenn der Wert innerhalb sehr kurzer Zeit zuerst auf 85 und dann auf 0 wechselt, ist es möglich, dass die update automation nur einmal (85) ausgeführt wird, da sie den Mode single hat.

Gnarfoz, 2025-03-05:
Das ist... vielleicht möglich. Aber ist das größere Problem nicht die Quelle der Änderung ausfindig zu machen?
Hast du noch irgendein anderes System im Einsatz, das Zugriff auf den WR hat? Der Thread ist inzwischen lang -- ich glaube, du hast EVCC?
Habe da mal fix nach 0x55 und 0xAA gesucht, aber nix gefunden. Ich kenne aber auch deren Code nicht wirklich. :-D

[111 more replies]