mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #474: Batterie Power RAW

Tr1n0va, 2025-02-22:
Hi,

Ich habe heute auf die neuste Firmware geupdatet und auch die neuste yaml von hier eingespielt.

Laut den Kommentaren in der yaml, soll der Sensor Battery Power Raw positiv sein, wenn die Batterie lädt und negativ wenn sie entlädt.

Ich beobachte aber genau das Gegenteil.
Ist nur der Kommentar falsch? Oder der Sensor?

[image]

[image]

[image]

[image]

[image]

Gnarfoz, 2025-02-22:
Welche Version ist "die neuste"? 95.03? Oder gibt es was neueres?

Sieht auf den ersten Blick aus als wäre der Kommentar falsch.
Benutz einfach den universellen Sensor [_Signed battery power_](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2250-L2269). der kombiniert den Energy Flow Status (aus dem sich ergibt, ob gerade geladen/entladen wird) mit dem Raw-Wert und sorgt für das korrekte Vorzeichen.

Tr1n0va, 2025-02-23:
Hab gestern die 95.04 Installiert.

Danke!

closed 2025-02-23 by Tr1n0va