mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #475: Feature Request: Multi-Inverter Support in Default Dashboard

ulfben, 2025-02-23:
First off, thanks for the fantastic work on this integration! [The default dashboard](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/dashboards/_DefaultDashboard_mkaiser/dashboard.yaml) is incredibly useful for getting a quick overview of the plant. However, I recently realized that it only displays data for a single inverter, which can be misleading in multi-inverter setups.

This is particularly noticeable (and problematic) for metrics such as total_dc_power and daily_pv_generation, which—at least in my case—only reflect the values from the Master inverter, effectively halving the true totals since the Slave inverter's contributions aren't included.

**Potential Improvements**

- Quick Fix: Update the labels to make this explicit. For example, instead of "Total DC Power," label it "Total DC Power (Inverter 1)" to clarify that it only represents a single inverter.
- Ideal Solution: Modify the dashboard to detect multiple inverters and sum up their values where applicable, providing a more accurate total view.

I understand that the dashboard might not be the primary focus of this project, but I figured it was worth making a feature request. Worst case, this could serve as a place where others running multiple inverters can share their dashboard solutions.

Thanks again for all your work on this, and for adding the discussion feature to the repo. Having a community of other users has helped me a lot!

ulfben, 2025-02-23:
I tried whipping up a sensor combining the two I've got. 
```
template:
  - sensor:
      - name: "Total DC Power (All Inverters)"
        unique_id: total_dc_power_combined
        unit_of_measurement: "W"
        state: >
          {% set master = states('sensor.total_dc_power') | float(0) %}
          {% set slave = states('sensor.total_dc_power_2') | float(0) %}
          {{ master + slave }}

```

It's not generic enough (should enumerate all `sensor.total_dc_power_#`) but it's a start. We'll see it's correct when the sun comes back tomorrow. :)

ulfben, 2025-02-23:
Oh, the Template Editor is where it's at! 

[image]

Took me a couple of iterations but this was much easier to debug. :D 

```
- sensor:
    - name: "Total DC Power (All Inverters)"
      unique_id: sg_total_dc_power_combined
      unit_of_measurement: "W"
      state: >-
        {% set total_power = namespace(value=0) %}      
        {% for sensor in states.sensor 
            if sensor.entity_id.startswith('sensor.total_dc_power') 
            and not sensor.attributes.get('friendly_name', '').startswith('Total DC Power (All') %}
          {% set power = sensor.state | float(0) %}          
          {% set total_power.value = total_power.value + power %}
        {% endfor %}        
        {{ total_power.value | round(2) }}

- sensor:
    - name: "Daily PV Generation (All Inverters)"
      unique_id: sg_daily_pv_generation_combined
      unit_of_measurement: "kWh"
      state: >-
        {% set total_generation = namespace(value=0) %}      
        {% for sensor in states.sensor 
            if sensor.entity_id.startswith('sensor.daily_pv_generation') 
            and '_battery_discharge' not in sensor.entity_id
            and not sensor.attributes.get('friendly_name', '').startswith('Daily PV Generation (All') %}
          {% set generation = sensor.state | float(0) %}          
          {% set total_generation.value = total_generation.value + generation %}
        {% endfor %}        
        {{ total_generation.value | round(2) }}

- sensor:
    - name: "Total PV Generation (All Inverters)"
      unique_id: sg_total_pv_generation_combined
      unit_of_measurement: "kWh"
      state: >-
        {% set total_generation = namespace(value=0) %}
        {% set debug = namespace(sensors=[]) %}
        {% for sensor in states.sensor 
            if sensor.entity_id.startswith('sensor.total_pv_generation')     
            and not sensor.attributes.get('friendly_name', '').startswith('Total PV Generation (All') 
            and 'battery_' not in sensor.entity_id %}
          {% set generation = sensor.state | float(0) %}
          {% set total_generation.value = total_generation.value + generation %}
        {% endfor %}
        {{ total_generation.value | round(2) }}
```

All of these values are accurate:
[image]

ulfben, 2025-02-24:
What is the difference between export_power and meter_active_power? 

[image]

Gnarfoz, 2025-02-25:
Excellent work by the way, aggregating this stuff dynamically without knowing the actual number of inverters is great!
Just not sure if it should replace the default dashboard (the result would be the same for people with just one inverter), or if should be a separate dashboard.
Surely, people with several inverters will want to know the values of each inverter *and* the total sums?

ulfben, 2025-02-26:
Thanks! Yeah, I'm not sure how to best integrate this.

I should probably take a moment to explore if it's possible to use Jinja in the dashboard itself to dynamically adjust what to show. Of or there's perhaps a conditional card of some kind.

Then we could refactor the default dashboard a smidge, to expose all inverters if they are present. Perhaps a tab per inverter, and an aggregate view with the combined sensors.

If it's okay with you, let's leave the issue open, and maybe me or someone else will find a moment to try it out.