mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #476: Implausible total values

jpwenzel, 2025-02-24:
**Describe the bug:**

I am seeing quite unusual/implausible values for total production and consumption recently.

These values of course mess up graphs in the energy dashboard of Home Assistant.
(the spikes seem to be in the area of "total energy ever produced" mind factor 1000)

My inverter might have received an update, I am trying to figure out when this started.

I will offer any help needed to better understand what's going on.

**Your Sungrow inverter:**
 - Model: SH-10.RT v112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [x] yes (evcc)
  - [] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2025.2.5

** Inverter Firmware Status:**
   - [?] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. See energy dashboard in Home Assistant

**Expected behavior**
Spiking read values filter/omitted from sensor readings and totals

Idea: could we investigate and filter out those values?

**Screenshots**

[image]
[image]

Gnarfoz, 2025-02-24:
It's hard to tell which sensors these are when the screenshots are cropped so small. Although... They look like a custom setup anyway? Could you please include which sensors these are?

The fact that iSolarCloud seems to show the same value makes it unlikely to be a problem with this integration... If the inverter reports 17.5 MWh, that's what Home Assistant will show.

jpwenzel, 2025-02-24:
I will provide more insights shortly.

The 17.5Mwh is the total production (over approx 2 years), and iSolarCloud shows this as grand total, whereas in HA, this is a single reading of sometime this early morning.

jpwenzel, 2025-02-24:
The sensor in question seems to be `sg_total_pv_generation`:

[image]

This is what 2 days ago looks like in iSolarCloud:

[image]

jpwenzel, 2025-02-24:
So it probably boils down to "the inverter is sending fluke data". :(

jpwenzel, 2025-02-25:
Short update: I have disconnected the small "Klarsolar HERO" appliance that was additionally reading values from the inverter.

I have not seen any "awkward" values since then.

closed 2025-02-25 by jpwenzel

Gnarfoz, 2025-02-25:
How does the Klarsolar thingy know the IP address of the inverter (or WiNet-S)?
Surely there must be some kind of configuration, or does it just scan the network and tries to find it on its own?

TCzerny, 2025-02-25:
Like the efa: home I assume it scans for valid vendor Mac addresses.

Gnarfoz, 2025-02-25:
That's... the opposite of robust. 🤨 Maybe the next batch of devices comes with a different network chip?
I guess it's a good default to have, for system owners who don't even know what an IP address is.
OTOH, a way to manually specify the address would be wise to have. But... that's for that manufacturer to figure out. 😁